Koenkk/zigbee2mqtt #23285: Mumubiz TYZGTH1CH-D1RF - `temp_lower_then` property do not saved

dmitrkov, 2024-07-08:
### What happened?

[Mumubiz TYZGTH1CH-D1RF](https://www.zigbee2mqtt.io/devices/TYZGTH1CH-D1RF.html) - `temp_lower_then` property do not saved

### Zigbee2MQTT version

1.39.0-1

### Adapter firmware version

20240315

### Adapter

SLZB-06

### Setup

Add-on on HA

dzikus, 2024-09-17:
Same problem. I’m setting:

zigbee2mqtt/mumubiz_thermostat {"auto_settings":{"enabled":"on","temp_greater_then":"on","temp_greater_value":30,"temp_lower_then":"off","temp_lower_value":24},"child_lock":"UNLOCK","humidity":null,"humidity_calibration":0,"humidity_sensitivity":10,"linkquality":120,"manual_mode":"OFF","power_outage_memory":"restore","state":"ON","temperature":24.9,"temperature_calibration":0,"temperature_sensitivity":0.1}

But:
zigbee2mqtt/mumubiz_thermostat {"auto_settings":{"enabled":"on","temp_greater_then":"on","temp_greater_value":30,"temp_lower_value":24},"child_lock":"UNLOCK","humidity":null,"humidity_calibration":0,"humidity_sensitivity":10,"linkquality":120,"manual_mode":"OFF","power_outage_memory":"restore","state":"ON","temperature":24.9,"temperature_calibration":0,"temperature_sensitivity":0.1}

Is published a second later without temp_lower_then. Automation  does not working also.

Zigbee2MQTT version

1.40.1

Adapter firmware version

20240710

Adapter

cc2652p

closed 2024-12-28 by github-actions[bot]

dzikus, 2025-01-12:
Still doesn't work.

Zigbee2MQTT version 2.0.0
Coordinator type zStack3x0
Coordinator revision 20240710
Frontend version 0.9.4
zigbee-herdsman-converters version 21.12.0
zigbee-herdsman version 3.2.1