Koenkk/zigbee2mqtt #23292: [New device support]: Immax Smart knob (07768L)

Nachtom, 2024-07-08:
### Link

https://www.immax.eu/immax-neo-smart-rotary-knob-change-cct-dimming-scene-zigbee-tuya-p17656/

### Database entry

Sorry, I don't see the mentioned file but from device page I see this:   Model Zigbee: TS004F; Manufacturer Zigbee: _TZ3000_ugi8ky6u; Support: Empty (how_to_add_support); IEEE address: 0x6c5cb1fffe9a4ec2; Network address: 0x7A23; Power: Battery ?; Pairing finished: True

### Comments

I have no idea how to add devices so I didn't get far in tutorial. I don't have any similar devices to look at (and I don't know how since external definition doesn't tell me much even for already working devices).

There is nothing on exposes tab. It is supposed to recognize turning left/right, short press, long press and also turning while pressed.

Sorry if this is not very helpful, I appreciate if you get to look at it. I'm ready to send more data if needed (and if you tell me where to find it - I have HA blue with HA skyconnect). :)

### External definition

```shell
const {battery, identify, onOff, commandsOnOff, commandsLevelCtrl} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS004F'],
    model: 'TS004F',
    vendor: '_TZ3000_ugi8ky6u',
    description: 'Automatically generated definition',
    extend: [battery(), identify(), onOff({"powerOnBehavior":false}), commandsOnOff(), commandsLevelCtrl()],
    meta: {},
};

module.exports = definition;
```

sebinemeth, 2024-07-31:
Hey,
I am completely new to z2m, and I have the same problem. I will share my progress here, but I would also appreciate an expert's help.

I have found this device [Tuya ERS-10TZBVK-AA](https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html) which looks exactly the same, I am not sure what the difference is. I have started to go through each command under [Device operation](https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html#device-operation) and extend the generated definition.

This is my definition so far (only added `commandsColorCtrl` from [modernExtends.ts](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/modernExtend.ts)):

```js
// TS004F.js
const {
    battery,
    identify,
    onOff,
    commandsOnOff,
    commandsLevelCtrl,
    commandsColorCtrl,
} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS004F'],
    model: 'TS004F',
    vendor: '_TZ3000_ugi8ky6u',
    description: 'Automatically generated definition',
    extend: [
        battery(),
        identify(),
        onOff({"powerOnBehavior":false}),
        commandsOnOff(),
        commandsLevelCtrl(),
        commandsColorCtrl(),
    ],
    meta: {},
};

module.exports = definition;
```
Don't forget to add the converter to your `configuration.yaml`:
```yml
external_converters:
  - TS004F.js
```

Restarted z2m and device is appearing, and the actions in `COMMAND` mode seem to work (except `hue_stop`):

`Brightness_Step_Up`
`Brightness_Step_Down`
`Toggle`
`Hue_Move`
~`Hue_Stop`~
`color_temperature_step_up`
`color_temperature_step_down`

When I change to `EVENT` mode nothing is detected.

I will keep this thread updated

closed 2024-12-28 by github-actions[bot]

Nachtom, 2025-04-26:
Hello,
I noticed that basically the same device was added recently (ERS-10TZBVK-AA Tuya Smart knob - the one [sebinemeth](https://github.com/sebinemeth) mentioned). Is it possible to just copy the configuration so this version of the same knob works as well?