Koenkk/zigbee2mqtt #23294: Last upgrades renamed base_topic

lecourtb, 2024-07-08:
### What happened?

After upgrading Z2M from 1.36.0-1 to 1.39.0-1, all my zigbeee entities id are rediscovered and renamed with suffix "_2".
I noticed that in Z2M MQTT parameters, the base topic which was "zigbee2mqtt" before upgrade is renamed to "zigbee2mqttHA" after upgrade. 

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20230507

### Adapter

zStack3x0

### Setup

add-on

dermotos, 2024-07-11:
+1 
I get the same. It's fine after restarting the Zigbee2Mqtt Add on after updating. But after the next system restart, all Zigbee entities are renamed with an `_2` suffix. 
I've restored from backup and tried again and this consistently occurs.

Koenkk, 2024-07-11:
Do you have a `base_topic` set in the Z2M addon configuration page? Try setting that to the correct topic.

lecourtb, 2024-07-12:
I did it, but it was restored to zigbee2mqttHA each z2m restart

Koenkk, 2024-07-13:
So if you change the `base_topic` in the z2m addon page (not the `configuration.yaml`!) to `zigbee2mqtt`, start the addon and restart it, it changes back to `zigbee2mqttHA`?

Wendigogo, 2024-08-02:
I got the exact same issue (since 1.38.0-1 FYI). I had to rollback to 1.37.0-1 and remove all my "_2" devices.

> So if you change the `base_topic` in the z2m addon page (not the `configuration.yaml`!) to `zigbee2mqtt`, start the addon and restart it, it changes back to `zigbee2mqttHA`?

I did this and everything is fine now. I tried rebooting Z2M module and HA and my `base_topic` remains unchanged.
I still don't know why/how it had been changed BTW.

PS : I guess #23235 is a duplicate of this one.

Wendigogo, 2024-08-02:
> I did it, but it was restored to zigbee2mqttHA each z2m restart

Have you changed it in the z2m addon page as said by @Koenkk or in Web UI page ?
I did it from addon page. Maybe you should try from here?

Enjoy!

closed 2024-12-28 by github-actions[bot]