Koenkk/zigbee2mqtt #23296: Cannot find module 'deep-object-diff' at startup (v1.39)

muxator, 2024-07-08:
### What happened?

Zigbee2mqtt 1.39 fails to start on a clean install performed via `npm ci`.

### What did you expect to happen?

A normal startup. v1.37.x used to work.

### How to reproduce it (minimal and precise)

```
git checkout 1.39.0
npm ci
node index.js
```

zigbee2mqtt **1.39** does not start (missing `deep-object-diff` dependency?).

I performed a clean install of the dependencies via `npm ci`, which wipes out `node_nodules` and then reinstalls everything from the lockfile.
The application is started via a hardened systemd user unit (the command line is the usual `/usr/bin/node index.js`).

v1.37.x used to work. I think the problem started with 1.38. From what I see in the logs I think there might be a missing runtime dependency?

Nodejs runtime comes from the default Fedora 40 packages:

- nodejs: v20.12.2
- npm: 10.5.0

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

not relevant

### Adapter

not relevant

### Setup

source code checkout started via hardened systemd user unit

### Debug log

```
Jul 08 [...] systemd[1057]: Starting zigbee2mqtt.service - zigbee2mqtt...
Jul 08 [...] node[3035]: Starting Zigbee2MQTT without watchdog.
Jul 08 [...] node[3035]: Error: Cannot find module 'deep-object-diff'
Jul 08 [...] node[3035]: Require stack:
Jul 08 [...] node[3035]: - /home/ant/code/zigbee2mqtt/dist/util/utils.js
Jul 08 [...] node[3035]: - /home/ant/code/zigbee2mqtt/dist/util/settings.js
Jul 08 [...] node[3035]: - /home/ant/code/zigbee2mqtt/index.js
Jul 08 [...] node[3035]:     at Function.Module._resolveFilename (node:internal/modules/cjs/loader:1143:15)
Jul 08 [...] node[3035]:     at Function.Module._load (node:internal/modules/cjs/loader:984:27)
Jul 08 [...] node[3035]:     at Module.require (node:internal/modules/cjs/loader:1231:19)
Jul 08 [...] node[3035]:     at require (node:internal/modules/helpers:179:18)
Jul 08 [...] node[3035]:     at Object.<anonymous> (/home/ant/code/zigbee2mqtt/lib/util/utils.ts:7:1)
Jul 08 [...] node[3035]:     at Module._compile (node:internal/modules/cjs/loader:1369:14)
Jul 08 [...] node[3035]:     at Object.Module._extensions..js (node:internal/modules/cjs/loader:1427:10)
Jul 08 [...] node[3035]:     at Module.load (node:internal/modules/cjs/loader:1206:32)
Jul 08 [...] node[3035]:     at Function.Module._load (node:internal/modules/cjs/loader:1022:12)
Jul 08 [...] node[3035]:     at Module.require (node:internal/modules/cjs/loader:1231:19)
Jul 08 [...] systemd[1057]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
```

Koenkk, 2024-07-09:
Try doing a `npm run build` before starting.

muxator, 2024-07-09:
It worked, thank you!

closed 2024-07-09 by muxator