Koenkk/zigbee2mqtt #23297: Danfoss Ally Pairing Failed

schneili1, 2024-07-08:
### What happened?

I tried to Pair some Danfoss Ally Thermostats, bit it Failed Mutlible Times.
After checking the Model Nuber it turned Out the device is not Supportet yet (014G2420).
Is there a Soloution? 
Thanks in Advance :) 

Store Linkt to Danfoss ( https://store.danfoss.com/de/de/Climate-Solutions-W%C3%A4rmetechnik/Heizk%C3%B6rperthermostate/F%C3%BChlerelemente-f%C3%BCr-Heizk%C3%B6rper/Elektronischer-Heizk%C3%B6rperthermostat/Heizk%C3%B6rperthermostat%2C-Danfoss-Ally%E2%84%A2%2C-Adaptertyp%3A-RA%3B-M30%2C-2-4-GHz/p/014G2420 )

### Zigbee2MQTT version

1.39.0-1

### Adapter firmware version

latest

### Adapter

SkyConnect

### Setup

HomeAssitant

lux73, 2024-07-09:
even its not supported yet - it should be possible to pair

schneili1, 2024-07-09:
No, it's trying to pair but always fails

lux73, 2024-07-09:
then anything is not correct in your Setup

pairing should be possible anytime

schneili1, 2024-07-09:
I just tried another one, and it worked it's even Showing a Symbol and as Supported

closed 2024-07-09 by schneili1