Koenkk/zigbee2mqtt #23300: Sonoff presence sensor doesn't work good: Firmware on 'SNZB-06P sensor' is newer than latest firmware online could be the problem?

psummo, 2024-07-08:
### What happened?

I have bought 4  SNZB-06P sonoff presence sensor since it's one of the most used with HA and Z2M. But seems is not working well: status is always occupancy true (I tried different sensivity) and takes more than 15 seconds (really to much) to be occupancy false. Seems doesn't work as expected. Could be the problem with this firmware newer than expected? Seems 1.0.5 is the last from z2m but the sensor is at 1.0.6 (I've bought them all with same firmware)

### What did you expect to happen?

Occupancy/firmware version works good

### How to reproduce it (minimal and precise)

SNZB-06P sensor with 1.0.6 firmware

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Setup

Home Assistant on raspberry 4

psummo, 2024-07-08:
> can you see the Z2M documentation ?
> 
> https://www.zigbee2mqtt.io/devices/SNZB-06P.html
> 
> " Detection duration Select the msOccupancySensing Cluster, ultrasonicOToUDelay is the detection duration setting.
> 
> On firmware versions lower than 1.0.5, the shortest detection duration is 30 (seconds), otherwise, the sensor will stay permanently on detected status.
> 
> Firmware version 1.0.5 and above allows setting the shortest detection duration to 15 (seconds).
> 
> "

seems doesn't wrote anything (I tried to change to 30sec for wait, but from dev console doesn't work. Btw is not the same from UI? I read also this https://sonoff.tech/product-review/tutorial/snzb-06p-firmware-upgrade-and-home-assistant-operation-guide/ and they expose attributes, so they can be managed from values tab in z2m. In that case info are updated  (I double check logs) but seems is still not working properly

DieserKirsche, 2024-11-03:
I am experiencing the same behavior. Everything worked fine for months, but now the status is always "Occupied" (or over hours). Sometimes it resets when a new occupancy is detected, but most of the time it doesn't. Can't remember if there's been an update to 1.0.6 in the last few weeks, I'm actually on 1.0.6 too. I know I bought it with 1.0.3 and did a manual update - it worked fine from then until a couple weeks ago. I'm using it with the latest HAOS, ZHA and Conbee II on RPi4.

uffevedenbrant, 2024-12-05:
I can agree that it fails with a error.
For me ZHA works perfect in my test HA setup. Changing to Z2M and the devices seems to be correct discovered but after a few config fails visible in the logs all entities goes offline. ( the config is tested 3 times before )

Seems to be a few "issues" in the tracker that's related to this sensor.

My assumption would be that 1.05 of the firmware behaves slightly different than firmware 1.06 any by that the configure fails.

As it behaves well under ZHA and fails under Z2M i guess there is a quirk that might need to be updated.

tardis89, 2024-12-08:
I just got one of these in the mail today and it's running the 1.06 firmware. Would love to have a fix for it as it's currently stuck in the occupied occupancy state and "bright" illumination state. I can see the linkquality changing if I move it to different places of the house, but that's all that seems to change.

uffevedenbrant, 2024-12-09:
Issue seems to be similar to this one related to the same sensor: https://github.com/Koenkk/zigbee2mqtt/issues/20808
Might be that Sonoff added some extra features or removed some that causes the configure to fail.

uffevedenbrant, 2024-12-11:
Tested with latest Z2M Edge.. And here it seems to work without any issues.
Both Z2M and Z2M Edge are however the latest one versions could install.
My Desk SNZB-06P sensor that I use for detecting presence gives good estimated values for me being there.
( or my 4 cats )

fjumeaux, 2024-12-12:
Tested the Sonoff under Z2M with firmware 1.0.6. Incredible sensitivity, works very well… maybe too well. In my kitchen space with a central island, it could have been perfect. However, my cats’ nocturnal activity triggers the Sonoff. This product is therefore ideal if you don’t have pets or for areas where animals are not allowed.

uffevedenbrant, 2024-12-12:
For me..
Detecting cats and humans is perfect. 
My summer house will have these sensors and everything that moves inhouse will start the "ImHere" automations.. 
nothing moves inose and we are probably not there..
Main issue is however the Z2M warning's/failures.. Me trying Z1M Edge where the warnings/errors not happened..

tardis89, 2024-12-12:
> Tested with latest Z2M Edge.. And here it seems to work without any issues. Both Z2M and Z2M Edge are however the latest one versions could install. My Desk SNZB-06P sensor that I use for detecting presence gives good estimated values for me being there. ( or my 4 cats )

I ran the Z2M Edge and it immediately picked up the sensor again. This time, it showed as not detected, then as soon as it detected something it was locked on again and would not get out of detected mode again. I tried changign the sensitivities and changing the timeout settings all to no luck.

tompeel, 2025-01-06:
Same issue here on the 1.0.6 FW, not sure you can roll back a FW but seems strange this hasnt been fixed yet, seems to have been an issue for 6 months. not sure how to help.

oudoulj, 2025-01-25:
Just got 2 in the mail yestarday but have only installed one of them so far which came with firmware 1.0.6 out of the box.

Like many of you here and elsewhere, the Occupancy state remains **stuck on Occupied** in Zigbee2MQTT ("Detected" on the Home Assistant card)
I got it to be Clear once at the beginning but then it picked me and turned to Occupied to never go back to Clear although I set it to a 15s timeout.

I'm running HA 2025.1.4 on RPi4 with a SMLight SLZB-06M coordinator (Core fw 2.7.1 / Zigbee fw revision 20241127) and Z2M integration (2.0.0.2)
After reading a few threads on the internet, I am a bit concerned about the stability of the product especially with firmware >= 1.0.5 😕 

PS: I haven't found a release note on fw 1.0.6 so far...

rvoosterhout, 2025-01-28:
For some reason setting ultrasonicOToUDelay to 0 fixed this issue for me. Still gives me a 15s timeout, but it does give me a timeout like that. Running on fw 1.0.6

oudoulj, 2025-01-28:
> For some reason setting ultrasonicOToUDelay to 0 fixed this issue for me. Still gives me a 15s timeout, but it does give me a timeout like that. Running on fw 1.0.6

I have edited my comment after realizing a mistake on my end. My 2 sensors seem to work fine after all. #MyBad

tardis89, 2025-01-28:
@oudoulj @rvoosterhout 
What did you have to do to get them to work? What version of Zigbee2MQTT are you running? I still have no luck with my 1.0.6 firmware.

oudoulj, 2025-01-30:
> [@oudoulj](https://github.com/oudoulj) [@rvoosterhout](https://github.com/rvoosterhout) What did you have to do to get them to work? What version of Zigbee2MQTT are you running? I still have no luck with my 1.0.6 firmware.

I am under Z2M 2.0.0-2

tardis89, 2025-02-11:
I was using Zigbee2MQTT Edge. I switched back to the regular version of Zigbee2MQTT (version 2.1.1 ) and now the sensor is working as expected. I didn't have to rejoin the sensor to make it work properly. They must have pushed out an update in Zigbee2MQTT that can properly handle the sensor now.

Sensor: SNZB-06P (firmware 1.0.6)
Zigbe2MQTT: Version 2.1.1

closed 2025-04-21 by github-actions[bot]