Koenkk/zigbee2mqtt #23302: Many devices don't work anymore after updating to v.1.39.0-1. Too many fatal errors.

Krzysztonek, 2024-07-08:
### What happened?

Some devices disappear or they are being disconnected from the network randomly for a long time. Then only part of them appears back again. 
The most problematic is a roller driver [AM43-0.45_40-ES-EB](https://www.zigbee2mqtt.io/devices/AM43-0.45_40-ES-EB.html#moes-am43-0.45%252F40-es-eb), which are out of the network since last update.

### Zigbee2MQTT version

1.39.0-1

### Adapter firmware version

7.4.2

### Adapter

SkyConnect

### Setup

x86

### Debug log

```
[00:19:29] INFO: Preparing to start...
[00:19:30] INFO: Socat not enabled
[00:19:30] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-07-09 00:19:34] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=RESET_WATCHDOG.
[2024-07-09 00:19:34] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-09 00:19:34] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-07-09 00:19:35] warning: 	zh:ember: [EzspConfigId] Failed to SET "ADDRESS_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-07-09 00:19:35] warning: 	zh:ember: [EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-07-09 00:19:35] warning: 	zh:ember: [EzspConfigId] Failed to SET "NEIGHBOR_TABLE_SIZE" TO "26" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-07-09 00:19:35] warning: 	zh:ember: [EzspConfigId] Failed to SET "SOURCE_ROUTE_TABLE_SIZE" TO "200" with status=ERROR_INVALID_VALUE. Firmware value will be used instead.
[2024-07-09 00:19:35] warning: 	zh:ember: [EzspConfigId] Failed to SET "MULTICAST_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-07-09 00:20:07] warning: 	z2m: Failed to ping 'Roleta Iga Okno D' (attempt 1/1, ZCL command 0x048727fffe557123/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":113}))
[2024-07-09 00:20:15] warning: 	z2m: Failed to ping 'Roleta Sypialnia Okno L' (attempt 1/1, ZCL command 0x3410f4fffee3b19f/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":115}))
[2024-07-09 00:20:24] error: 	z2m: Failed to configure 'Energia AC Gab_Iga', attempt 1 (Error: ZCL command 0xa4c138bcff9500c7/1 genBasic.read(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":118})
    at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:96:31)
    at EmberAdapter.onMessageSent (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:607:34)
    at Ezsp.emit (node:events:517:28)
    at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:3953:14)
    at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:792:18)
    at Ezsp.tick (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:446:22)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7))
[2024-07-09 00:15:08] warning: 	z2m: Failed to ping 'Roleta Salon Okno PL' (attempt 1/1, ZCL command 0xcc86ecfffeb010ff/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":255}))
[2024-07-09 00:15:45] warning: 	z2m: Failed to ping 'Roleta Salon Okno LP' (attempt 1/1, ZCL command 0xcc86ecfffea1f7ca/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":31}))
[2024-07-09 00:15:52] warning: 	z2m: Failed to ping 'Roleta Salon Okno LL' (attempt 1/1, ZCL command 0x3410f4fffee9ef9f/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":41}))
```

Nerivec, 2024-07-09:
Those fatal errors happen during the startup sequence but there is a retry mechanism in place to bypass them (due to hardware flow control issues, Silabs is aware). In short, it's a problem, but it's bypassed for now, and doesn't prevent Z2M from starting; it should not be related to your troubles here.

Seems devices are failing to ping though. Any chance you have a fuller `debug` log file I can take a look at?

Did you try clearing the NVM3 on your adapter (backup/clear/restore)? Seems some configs are failing to set that really shouldn't, could be the adapter is having some troubles.

Krzysztonek, 2024-07-09:
> Did you try clearing the NVM3 on your adapter (backup/clear/restore)?

I'm not aware of any method of clearing the NVM3 on my SkyConnect adapter.
**Would you please give me any hint? commands? procedure?**

I also rebooted my HA server several times. Unfortunately, it doesn't help/change/repair anything. Still the same set of errors:
```
[00:19:29] INFO: Preparing to start...
[00:19:30] INFO: Socat not enabled
[00:19:30] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-07-10 00:19:35] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=RESET_WATCHDOG.
[2024-07-10 00:19:35] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-10 00:19:35] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-07-10 00:19:37] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-07-10 00:19:37] error: 	zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-07-10 00:19:37] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-10 00:19:37] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-07-10 00:19:37] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[2024-07-10 00:19:37] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-07-10 00:19:39] warning: 	zh:ember: [EzspConfigId] Failed to SET "ADDRESS_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-07-10 00:19:39] warning: 	zh:ember: [EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-07-10 00:19:39] warning: 	zh:ember: [EzspConfigId] Failed to SET "NEIGHBOR_TABLE_SIZE" TO "26" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-07-10 00:19:39] warning: 	zh:ember: [EzspConfigId] Failed to SET "SOURCE_ROUTE_TABLE_SIZE" TO "200" with status=ERROR_INVALID_VALUE. Firmware value will be used instead.
[2024-07-10 00:19:39] warning: 	zh:ember: [EzspConfigId] Failed to SET "MULTICAST_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
```
I'm even not able to switch to the Z2M UI:
[image]

Krzysztonek, 2024-07-09:
> Did you try clearing the NVM3 on your adapter (backup/clear/restore)?

No, I'm not aware of any method clearing the NVM on my SkyConnect adapter.
Would you please give me any hints? guidance? commands? procedures?

I rebooted my HA server several times. Unfortunately, it doesn't help/change/repair anything. Still the same set of error messages appears:

```
[21:11:24] INFO: Preparing to start...
[21:11:25] INFO: Socat not enabled
[21:11:26] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-07-09 21:13:18] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=RESET_WATCHDOG.
[2024-07-09 21:13:18] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-09 21:13:18] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-07-09 21:13:20] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-07-09 21:13:20] error: 	zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-07-09 21:13:20] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-09 21:13:20] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-07-09 21:13:20] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[2024-07-09 21:13:20] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-07-09 21:13:22] warning: 	zh:ember: [EzspConfigId] Failed to SET "ADDRESS_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-07-09 21:13:22] warning: 	zh:ember: [EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-07-09 21:13:22] warning: 	zh:ember: [EzspConfigId] Failed to SET "NEIGHBOR_TABLE_SIZE" TO "26" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-07-09 21:13:22] warning: 	zh:ember: [EzspConfigId] Failed to SET "SOURCE_ROUTE_TABLE_SIZE" TO "200" with status=ERROR_INVALID_VALUE. Firmware value will be used instead.
[2024-07-09 21:13:22] warning: 	zh:ember: [EzspConfigId] Failed to SET "MULTICAST_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-07-09 21:38:05] warning: 	zh:ember:ezsp: An ID conflict was detected for network address "59061". Corresponding devices kicked from the network.
[2024-07-09 23:50:40] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":161} messageTag=255]
[2024-07-10 00:00:43] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":170} messageTag=255]
[2024-07-10 00:17:23] error: 	z2m: Not connected to MQTT server!
[2024-07-10 00:17:23] error: 	z2m: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: '{"state":"offline"}
```
I'm even to able to get into UI:
[image]

Nerivec, 2024-07-09:
The failed pings on startup are likely preventing the frontend from showing up in a timely manner. Current dev/edge has some improvements in that domain.

For the NVM3, you can try using the new `ember-zli`: https://github.com/Nerivec/ember-zli (check the [Wiki](https://github.com/Nerivec/ember-zli/wiki/Bootloader#clear-nvm3))
or `universal-silabs-flasher` + `bellows`: [procedure details](https://github.com/darkxst/silabs-firmware-builder/issues/84#issuecomment-2094078288)
The overall process is the same in both: backup, clear the NVM3, restore.
Note: the backup/restore process is the same executed when you stop/start Z2M (when applicable).

You can DM me from zigbee2mqtt server on Discord if needed.

Krzysztonek, 2024-07-13:
I tried to do backup of my skyconnect configuration following the linked procedure.
When executing the following command:
```
➜  ~ bellows -d /dev/ttyUSB0 -b 115200 backup | tee bellow.backup.json 
```
I got the following log:
```
Traceback (most recent call last):
  File "/usr/bin/bellows", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/bellows/cli/util.py", line 39, in inner
    loop.run_until_complete(f(*args, **kwargs))
  File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/bellows/cli/backup.py", line 80, in backup
    await _backup(ezsp)
  File "/usr/lib/python3.12/site-packages/bellows/cli/backup.py", line 114, in _backup
    (status, key) = await ezsp.getKey(key_type)
                          ^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/bellows/ezsp/__init__.py", line 300, in __getattr__
    return getattr(self._protocol, name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/bellows/ezsp/protocol.py", line 148, in __getattr__
    raise AttributeError(f"{name} not found in COMMANDS")
AttributeError: getKey not found in COMMANDS
```
 So, I've followed again [this video](https://www.youtube.com/watch?v=x1QeNPi6tK8) to upgrade my SkyConnect dongle to 7.4.3.0 [from this repo](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/skyconnect/ncp-uart-hw-v7.4.3.0-skyconnect-115200.gbl).

And I failed again:
```
[14:37:59] INFO: Preparing to start...
[14:37:59] INFO: Socat not enabled
[14:37:59] INFO: Enabled Zigbee2MQTT watchdog with value 'default'
[14:38:00] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT with watchdog (60000,300000,900000,1800000,3600000).
[2024-07-13 14:38:04] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=RESET_WATCHDOG.
[2024-07-13 14:38:04] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-13 14:38:04] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-07-13 14:38:06] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-07-13 14:38:06] error: 	zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-07-13 14:38:06] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-13 14:38:06] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-07-13 14:38:06] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[2024-07-13 14:38:06] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-07-13 14:38:08] warning: 	zh:ember: [EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-07-13 14:40:34] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-07-13 14:40:34] error: 	zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-07-13 14:40:34] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-13 14:40:34] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-07-13 14:40:34] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[2024-07-13 14:40:39] warning: 	zh:ember: [EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
```

I'm stuck again, really don't understand what is going wrong in there.

Nerivec, 2024-07-13:
Try this:
- Shutdown Z2M
- Check that your `coordinator_backup.json` (in zigbee2mqtt data folder) has been updated/is recent (open with a text editor and check the `date` in it)
- Install `ember-zli` https://github.com/Nerivec/ember-zli/wiki#installation
- Launch `ember-zli bootloader` command with the adapter plugged into the machine
- Answer the prompt's questions
- Go to `Clear NVM3` menu https://github.com/Nerivec/ember-zli/wiki/Bootloader#clear-nvm3
- Confirm and wait for it to be done
- Go to `Update firmware` menu https://github.com/Nerivec/ember-zli/wiki/Bootloader#update-firmware
- Select `Pre-defined` > `Latest`
- Confirm and wait for it to be done
- Exit bootloader
- Go back to Z2M, and start it again

Since you have a proper coordinator backup created by Z2M, it should detect that the adapter was reset, and reform the network using that backup. If all goes well, this should clear the NVM3 and restore your backup from Z2M.
Note that step 2 is important, as the process described here relies on that backup file to be correct, if not you may end up having to re-pair all devices.

Krzysztonek, 2024-07-14:
Yesterday I did the backup:
```
{
  "metadata": {
    "format": "zigpy/open-coordinator-backup",
    "version": 1,
    "source": "zigbee-herdsman@0.50.1",
    "internal": {
      "date": "2024-07-13T11:00:21.474Z",
      "ezspVersion": 13
    }
  },
  "stack_specific": {
    "ezsp": {
      "hashed_tclk": "7c257da8ae16c70d6ffececad90cfaf8"
    }
  },
...
```
I already updated my SkyConnect dongle firmware to EMBER. So, my first concern is, why is this dongle still visible as `ezsp`, not as `ember`.

Today I launched  `ember-zli bootloader` command and got the following:

Nerivec, 2024-07-14:
EZSP is the name of the "middle-man" in the EmberZNet stack (EmberZNet Serial Protocol = EZSP) so it's still there in some names, just the Z2M driver itself is named `ember`, as opposed to the old driver, named `ezsp`.
Your backup appears correct, and according to your logs, you are using the `ember` driver properly.

Can you go into the [data folder for ember-zli](https://github.com/Nerivec/ember-zli/wiki#data-folder) and retrieve the log file with the failed NVM3 clearing?
Also, can you enable `debug` log for your next Z2M logs, so I can see where the error is exactly.

Krzysztonek, 2024-07-14:
Here is my ember-zli data folder content:
[logs.zip](https://github.com/user-attachments/files/16228490/logs.zip)

And this is my Z2M log folder content:
[z2m_log.zip](https://github.com/user-attachments/files/16228524/z2m_log.zip)

Nerivec, 2024-07-17:
Alright, let's try another method:

- Launch `ember-zli stack` command with the adapter plugged into the machine
- Answer the prompt's questions
- Go to Reset NVM3 tokens menu https://github.com/Nerivec/ember-zli/wiki/Stack#reset-nvm3-tokens
- Confirm
- Exit
- Go back to Z2M, and start it again

Same as before, make sure your `coordinator_backup.json` from Z2M is proper so it will restore when you start it again.

PS: you can enable `debug` in logs from the `configuration.yaml` file directly.
```yaml
advanced:
  log_level: debug
```

Krzysztonek, 2024-07-17:
I launched `ember-zli stack` command with the SkyConnect dongle plugged into my PC USB port, getting this:

```
[2024-07-17 20:21:31.136] info:         cli: Data folder: C:\Users\Krzysztof\ember-zli.
? Path: COM4, Baudrate: 115200, RTS/CTS: true. Use this config? yes
[2024-07-17 20:21:31.638] info:         zh:ember:ezsp: ======== EZSP starting ========
[2024-07-17 20:21:31.639] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-07-17 20:21:31.651] info:         zh:ember:uart:ash: Serial port opened
[2024-07-17 20:21:31.655] info:         zh:ember:uart:ash: ======== ASH starting ========
[2024-07-17 20:21:31.194] error:        zh:ember:uart:ash: ASH disconnected: ASH_ERROR_RESET_FAIL | Adapter status: NO_ERROR
[2024-07-17 20:21:31.195] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-07-17 20:21:31.197] info:         zh:ember:uart:ash: ======== ASH starting ========
[2024-07-17 20:21:31.731] error:        zh:ember:uart:ash: ASH disconnected: ASH_ERROR_RESET_FAIL | Adapter status: NO_ERROR
[2024-07-17 20:21:31.732] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-07-17 20:21:31.738] info:         zh:ember:uart:ash: ======== ASH starting ========
[2024-07-17 20:21:31.261] error:        zh:ember:uart:ash: ASH disconnected: ASH_ERROR_RESET_FAIL | Adapter status: NO_ERROR
[2024-07-17 20:21:31.262] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-07-17 20:21:31.265] info:         zh:ember:uart:ash: ======== ASH starting ========
[2024-07-17 20:21:31.802] error:        zh:ember:uart:ash: ASH disconnected: ASH_ERROR_RESET_FAIL | Adapter status: NO_ERROR
[2024-07-17 20:21:31.803] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-07-17 20:21:31.806] info:         zh:ember:uart:ash: ======== ASH starting ========
[2024-07-17 20:21:31.343] error:        zh:ember:uart:ash: ASH disconnected: ASH_ERROR_RESET_FAIL | Adapter status: NO_ERROR
    Error: Failed to start EZSP layer with status=33.
```
Since some errors appeared, I launched the below set of commands:

[image]

As you can see, during first atempt the `Clear NVM3` command failed with errors. During second atempt I was not able even to launch this command due to ` Error: Failed to start EZSP layer with status=33.` appearance, which finally stopped the whole process.

I don't understand what is wrong with this damn adapter. The only I can say, seems it's not damaged.

I would also bring your attention to one more point (please, see the below screen with `Update Firmware Menu`):
[image]

From what I see, the recommended version of the firmware is 7.4.1.0 while I already updated the adapter to version 7.4.3.0. As per your advise I didn't downgrade it to `recommended`, still keeping `the latest` 7.4.3.0.

Nerivec, 2024-07-17:
Since you flashed latest before, make sure you use these settings: `port: COM4, baudrate: 115200, rtscts: false`.
In the first screenshot, did you unplug the adapter before launching `ember-zli stack` again? It's likely stuck in bootloader since you `Force close` (instead of `Exit bootloader`).
Unplug the adapter for a few seconds, then  try using `ember-zli stack` directly after re-plugging it.

Krzysztonek, 2024-07-18:
I did **exactly** what you've suggested:

> 
    Launch ember-zli stack command with the adapter plugged into the machine
    Answer the prompt's questions
    Go to Reset NVM3 tokens menu https://github.com/Nerivec/ember-zli/wiki/Stack#reset-nvm3-tokens
    Confirm
    Exit
    Go back to Z2M, and start it again

This time `Reset NVM3 tokens` was executed successfully! Indeed, when launching `ember-zli stack` directly right after re-pluging the dongle, it works fine.

However, seems this method still doesn't bring any success, since the same errors appear:
```
[08:26:04] INFO: Preparing to start...
[08:26:04] INFO: Socat not enabled
[08:26:05] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-07-18 08:26:20] error: 	z2m: Error while starting zigbee-herdsman
[2024-07-18 08:26:20] error: 	z2m: Failed to start zigbee
[2024-07-18 08:26:20] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-07-18 08:26:20] error: 	z2m: Exiting...
[2024-07-18 08:26:20] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:889:19)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2722:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:129:29)
    at Zigbee.start (/app/lib/zigbee.ts:63:27)
    at Controller.start (/app/lib/controller.ts:139:27)
    at start (/app/index.js:154:5)
``` 

I also enabled `debug` for Z2M, but it changed nothing in the logs.

Nerivec, 2024-07-18:
I seem to remember you had some past troubles with your `configuration.yaml` file being duplicated in different folders and the file you were editing wasn't pointing to the one actually used by Z2M. Can you double-check that it didn't happen again?

Krzysztonek, 2024-07-18:
Yes, you're right, I had some troubles in the past. It was caused by having two different Z2M instances installed on the same machine with two different Zigbee dongles connected to this machine, to ballance the zigbee network traffic ( those days I was facing huge lags in my zigbee network)..

But this is a past, few months ago I installed a fresh HA system from scratch with one Z2M add-on. Since then it worked fine until this huge failure, which I'm facing now. 
Currently no other Z2M folders exist on my machine, just one. Here is my full `configuration.yaml` zip file uploaded directly from `//192.168.1.20/config/zigbee2mqtt` folder:
 
[configuration.zip](https://github.com/user-attachments/files/16288524/configuration.zip)

Nerivec, 2024-07-18:
Stop Z2M.
Change `rtscts: true` to `rtscts: false` and `log_level: warning` to `log_level: debug`.
Start Z2M.

See if you can at least get the logs to show as debug.

Note: August release will support concurrency, so you will probably want to remove `adapter_concurrent: 1` too.

Nerivec, 2024-07-19:
`rtscts` being true (with the firmware you have requiring false) is likely what's preventing Z2M from starting since I don't see anything else that would be an issue now.

@Koenkk You know the structure of the HA add-on better than me. Could something be cached and make a mess of the settings in `configuration.yaml`? Seems like something's not quite right here, it's not following what's in the file properly.

Krzysztonek, 2024-07-19:
What a lovely Friday! I powered my HA server off for 2 hours, then powered on back. And my Z2M ...started working!
Looks that even whole zigbee network has been rebuilt successfully (I still need to re-pair only few devices). Thanks to you, this huge milestone has been achieved!
Here are my logs:

[log.zip](https://github.com/user-attachments/files/16311243/log.zip)

Currently only the UI is still not reachable. When clicking `Open UI` button, I'm getting this:
 
[image]

Clicking `RERTY` doesn't change anything, still the same message appears..
Do you know how to get into IU screen?

Nerivec, 2024-07-19:
There are some improvements for the UI being slow to load on start for 1.39 in current dev branch/edge. If you want, you can try the `Edge` version of the add-on. You have to make sure to shutdown the regular add-on, and disable auto-start and watchdog for it. Also make sure to remove the line `adapter_concurrent: 1` from your `configuration.yaml`. (You can switch back to regular, once August release is available.)

I am seeing a few repeating issues in your latest logs though:
```
[2024-07-19 12:41:06] error: 	z2m: MQTT error: client disconnecting
[2024-07-19 12:41:06] error: 	z2m: MQTT error: client disconnecting
[2024-07-19 12:41:06] error: 	z2m: MQTT error: client disconnecting
[2024-07-19 12:41:11] error: 	z2m: MQTT error: client disconnecting
```
This is happening right around the time the UI should become available.

Koenkk, 2024-07-21:
You have to put `rtscts` in the Z2M addon page configuration (under `serial`), not the z2m `configuration.yaml` itself.

n-prat, 2024-08-24:
Hello,

I seemed to have a related issue (?) with the logs spammed of
```
zh:ember:uart:ash: ======== ASH starting ========
error:        zh:ember:uart:ash: ASH disconnected: ASH_ERROR_RESET_FAIL | Adapter status: NO_ERROR
info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
```
and then Z2M refusing to start.

Reflashing + doing the `Reset NVM3 tokens [all defaults/unselected]` seemed to have done the trick. Thank you for that!

[not sure reflashing was needed but can't have hurt either]

n-prat, 2024-12-11:
So some observations:
- this issue is still happening regularly; at least with <7.4.4. I have just today flashed 7.4.5 so we will see
- I am sort of seeing a pattern where it happens only b/c of reboot/restart, either of Z2M, Home Assistant, or the whole VM (QEMU). Indeed, it can work for weeks when I am not rebooting, but if I try new stuff and end up having to reboot a bunch, I can pretty much bet it will fail.

Nerivec, 2024-12-11:
Could you try this:
- (when you network is working) stop Z2M, with Ember ZLI, do a [Backup NVM3 tokens](https://github.com/Nerivec/ember-zli/wiki/Stack#backup-nvm3-tokens), keep that file with a clear name, in a safe place (so you don't overwrite it in next step).
- Next time your network has the issue, repeat the previous step.

I'll compare the files, see if something is going wrong inside your adapter. Please also include the Z2M `data` folder (`configuration.yaml` and `database.db`).
_Note: This will contain a lot of information about your network configuration, if you don't want to share it in public, you can create a private repo on Github, or find me on Discord..._

n-prat, 2025-01-28:
@Nerivec it's happened again; it's been quite a while.

I am now on z2m 2.0; I have given you access to my "debug repo".

This time: I only:
- unplugged then replugged [I wanted to avoid it; but I did it by reflex 🤦‍♂️]
- then did the `Reset NVM3 tokens [all defaults/unselected]` (I saved the backup before, and uploaded to the git repo)

NOTE: when it was failing to start, the path changed for the normal `/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20231008114538-if00` to `/dev/serial/by-id/usb-1a86_55d4-if00`. After rebooting the VM, and `Reset NVM3 tokens [all defaults/unselected]` it's back to the "good path". Not sure if that's relevant.

But again; I can live with only having to do a quick workaround every couple months or so; so I would totally understand if you don't want to waste time on this.

Nerivec, 2025-01-28:
Sounds like some path conflict, rather than an issue with the dongle itself? Some misconfiguration of the VM related to the USB/serial passthrough? The "by-id" path should remain the same. That second one looks like something went wrong with reading device info, so it reverted to a raw manuf/id path.
What's your setup exactly?

n-prat, 2025-01-28:
Yes but even when changing to the wrong path `/dev/serial/by-id/usb-1a86_55d4-if00`, I still got 

```
[17:41:39] INFO: Preparing to start...
[17:41:39] INFO: Socat not enabled
[17:41:39] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-28 17:41:42] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-28 17:41:42] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-28 17:41:42] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-28 17:41:42] info: 	zh:ember: Using default stack config.
[2025-01-28 17:41:42] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-01-28 17:41:42] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-01-28 17:41:42] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-28 17:41:42] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-01-28 17:41:52] info: 	zh:ember:uart:ash: Serial port opened
[2025-01-28 17:41:52] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-28 17:41:55] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-28 17:41:55] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-28 17:41:58] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-28 17:41:58] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-28 17:42:01] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-28 17:42:01] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-28 17:42:04] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-28 17:42:04] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-28 17:42:07] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-28 17:42:07] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-28 17:42:07] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-28 17:42:07] error: 	z2m: Exiting...
[2025-01-28 17:42:07] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
```

Only after the `Reset`(and changing back the path), it re-worked.

My setup is:
- MacOS host: old Intel Mac Mini (circa 2016?)
- Qemu VM (direct, not using UTM); capturing the adapter with 
```
-device \
usb-host,vendorid=0x1a86,productid=0x55d4 \
-device \
usb-host,vendorid=0x05ac,productid=0x8289
```

The full VM starting script is: [qemu.sh](https://github.com/user-attachments/files/18577565/qemu.txt); and that has pretty much not changed since I set up HAOS.

Nerivec, 2025-01-28:
Is it about the reset or about the unplugging though? Unplug/replug would force the host to reload the USB device info. That might recover it from a crash of some kind.
I know nothing about that setup though, I had never even seen it before for Z2M usage, so, I won't be able to help much on that front... You are going to want to check everything on the host, and same on the VM, make sure something's not going wrong with the USBs, the drivers, etc. (might want to check for updates too).

Maybe you can try an [APP clear](https://github.com/Nerivec/ember-zli/wiki/Bootloader#clear-app), that will erase the firmware completely, clear out everything (network state included). Then you re-flash a [recent firmware](https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update5), and see if it happens again (make sure you have the latest version of Ember ZLI)...

n-prat, 2025-01-29:
Both I guess? Or I don't understand. In any case, to fix `HOST_FATAL_ERROR` needs a `Reset`.

I'll keep your idea in mind, but that seems extreme, and I really don't want to redo my network so if that fails only every other month or so, I'm fine with the current situation.

It could indeed just be my setup; maybe the USB capturing is flaky when rebooting the VM? But I am 90% sure I have tried rebooting+unplugging and that was not enough.

You can close this ticket if you want.

Nerivec, 2025-01-29:
The behavior is a bit strange. But we'd need more definitive facts to figure out where the problem might be.
Clearing the zigbee tokens should have no bearing on the device's ability to be recognized properly. The more likely scenario is that a combination of unplugging the device, or the use of Ember ZLI itself, does something in that regard. Check the host's system logs around the time it stopped working, and when it started working again, something must be happening in the background.

You can try going to 8.0.2 if you haven't already: https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update5

n-prat, 2025-01-29:
My setup is even worse than I thought; I am using the `Shortcuts` app and I don't think it logs anything...

Anyway, I will get to you if I find something, but that does not look promising.

Nerivec, 2025-01-29:
You may want to migrate to a more common setup, if only to be able to gather more insights from other users.

closed 2025-04-08 by github-actions[bot]