Koenkk/zigbee2mqtt #23304: SNZB-01P  bind not effective

liangjia2019, 2024-07-09:
### What happened?

I have two devices, SNZB-01P and SNZB-01, and there was an issue while executing a bind on them and another dual channel switching device 
SNZB-01 Perfect Execution
[image]
SNZB-01P binds normally, but the device has never been triggered to execute normally
[image]
[image]
[SNZB-01P_bind_Z2M.zip](https://github.com/user-attachments/files/16140048/SNZB-01P_bind_Z2M.zip)
[image]
From the perspective of network sniffing, devices do indeed have successful responses

Does he need more configuration or operation? Please let me know. Thank you

### Zigbee2MQTT version

1.39.0 commit: 03269267

### Adapter firmware version

20240709

### Adapter

skyconnect

### Setup

 x86-64

liangjia2019, 2024-07-19:
@Koenkk can you help us

Koenkk, 2024-07-20:
After they are bound, do you see the SNZB-01 sending on/off commands to the switch? If not, then this an issue of the SNZB-01, not of z2m.

ilbarone87, 2024-09-20:
I have same issue, the pairing goes fine but no on/off triggering. 
Have force deleted many times (also force delete and restart container and then re-pair) but no luck.
 @liangjia2019 have you manage to solve this or is still an issue for you too

ilbarone87, 2024-09-26:
Any reason why this reported as failed?

z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":["genOnOff"],"failed":[],"from":"0x003c84fffece4137/1","to":"led_cucina_top/1"},"status":"ok","transaction":"sgyhr-1"}'

liangjia2019, 2024-09-29:
> After they are bound, do you see the SNZB-01 sending on/off commands to the switch? If not, then this an issue of the SNZB-01, not of z2m.

You're right. This is indeed a problem with SNZB-01P, and I will push the team to solve it

liangjia2019, 2024-09-29:
> Any reason why this reported as failed?
> 
> `z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":["genOnOff"],"failed":[],"from":"0x003c84fffece4137/1","to":"led_cucina_top/1"},"status":"ok","transaction":"sgyhr-1"}'`

The key point when setting up is that you need to wake up the device through buttons or other means.
Unfortunately, there are some issues with the binding implementation mechanism of this device, and the effect cannot meet expectations

closed 2024-09-29 by liangjia2019

ifurnadjiev, 2025-01-14:
> You're right. This is indeed a problem with SNZB-01P, and I will push the team to solve it

@liangjia2019 any progress on this? Are there any firmware update plans that address this issue?

Jerepain, 2025-02-07:
I'ml looking forward to it as well

> > You're right. This is indeed a problem with SNZB-01P, and I will push the team to solve it
> 
> [@liangjia2019](https://github.com/liangjia2019) any progress on this? Are there any firmware update plans that address this issue?

liangjia2019, 2025-02-08:
We are planning to sort out and correct the unreasonable implementation of the sensor. We can expect that it will be implemented normally

Fohdeesha, 2025-05-05:
@liangjia2019 any update on new firmware for these?

AlbertoKarakoutev, 2025-06-23:
Checking in for updates on this. It's the same behaviour for me - output clusters are there and binding completes normally, but nothing is actually controlled when pressing the button.

PetitPhoenix, 2025-07-30:
Hello, I still have issues with binding and don't see any OTA update.

@liangjia2019 do you have any update?

Fexiven, 2025-12-09:
I don't see why they wouldn't make it work. Isn't it just a tiny firmware update?

liangjia2019, 2025-12-09:
I have communicated with the corresponding product manager again, and we will have a plan for future support of this feature. You can look forward to it