Koenkk/zigbee2mqtt #23307: Sonoff BSP Water Flow Device - flow always returns zero

scargill, 2024-07-09:
### What happened?

Latest Zigbee2MQTT working with Home Assistant - the BSP can be turned on and off - and that's it - no flow data when running.

### What did you expect to happen?

I expected the vulume of water passing through to be exposed.

### How to reproduce it (minimal and precise)

Hook the BSP into Zigbee2mqtt - turn it on with water flowing through - note that flow value is zero.

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

 20210120

### Adapter

ZZH

### Setup

Running in Docker on Raspberry Pi 5

scargill, 2024-07-10:
Understand we should have a fix next week. As soon as it's working I'll update the blog entry at https://tech.scargill.net

acarlo79, 2024-07-10:
Just updated the valve firmware to 1.0.3 and now I have the flow reading working correctly, so the bug was in the valve firmware and not zigbee2mqtt

scargill, 2024-07-10:
updating - 1 hour 3 minutes z2mqtt says....

scargill, 2024-07-10:
39 minutes let - it's to be hoped I don;t have a power cut.

acarlo79, 2024-07-10:
Did it work out for you? I found it to be very accurate

scargill, 2024-07-10:
Erm I just pur the pipe on for 2 minutes - we don;t have stunning pressue - during that time it returned 0.2 cubic metres/hour - doesnt sound liek a lot - but over to you....  better than it saying nothing. IF I'd had that one for an hour - I'm pertty sure it would have put out more than 0.2 cubic metres... thoughts?

acarlo79, 2024-07-10:
I tested the sprinklers, I got a flow of 0.4 m3/h which is about 6.6 L/m and it seems to be in line with what I expected.
Also, I run some cycles on 5 and 10L and seems to be in line. Tomorrow I will test with an hose and a bucket for the final test.

scargill, 2024-07-10:
Oh do let me know....

scargill, 2024-07-10:
If you have half decent pressure that might be right - ours is low...

acarlo79, 2024-07-11:
So, I run the following test:

Water presure at the tap (no flow)  - 2.6 bar

Run valve cycle based on desired liters:

asked 2 liters - delivered 3
asked 5 liters - delivered 6
asked 10 liters - delivered 11
asked 20 liters - delivered 21

My conclusion is that somehow (maybe due to hose lenght, there is an extra liter of water delivered. Considering that that calculation is done internally using the flow value, I feel very happy to use it as reading since the1 extra liter of water used for the lawn will be ininfluent.

closed 2024-12-28 by github-actions[bot]

FilipDem, 2025-04-27:
Anyone still the issue? I have a SWV with firmware 1.0.4... But the flow remains at zero.
To be completely correct, my water pump is not giving a lot of pressure... Could this be the problem?
Filip

jesualdomatos-mds, 2025-06-04:
Estou com o mesmo problema

rayone, 2025-11-28:
> I had a discussion with Sonoff support team about the topic... Unfortunately it went a different route with discussions about reception signals etc, restart pairing, ... Finally I could work around the problem myself by adding the flow explicitly to the reporting settings of the device. Sonoff told me that this would not be required as it should work "out of the box", however this solved my problem. See https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request-device-configure-reporting for more information. Of course this can be done through the UI. Just be aware that you need a minimal water debit before a flow is reported. And a final remark: even if the support team of Sonoff did not solve my problem immediately, they give a fast support (and this triggered also my search to the solution).

Thanks, this is the only action that worked for me on fw 1.04, but I don't understand "Just be aware that you need a minimal water debit before a flow is reported".

idelectrik, 2026-08-24:
> > > I had a discussion with Sonoff support team about the topic... Unfortunately it went a different route with discussions about reception signals etc, restart pairing, ... Finally I could work around the problem myself by adding the flow explicitly to the reporting settings of the device. Sonoff told me that this would not be required as it should work "out of the box", however this solved my problem. See https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request-device-configure-reporting for more information. Of course this can be done through the UI. Just be aware that you need a minimal water debit before a flow is reported. And a final remark: even if the support team of Sonoff did not solve my problem immediately, they give a fast support (and this triggered also my search to the solution).
> > 
> > 
> > Thanks, this is the only action that worked for me on fw 1.04, but I don't understand "Just be aware that you need a minimal water debit before a flow is reported".
> 
> I am using the Sonoff for irrigation with a drip system... Due to the drip system the water flow (m3/min) is very low... This means means that it is insufficient to make the internal rotor system of the Sonoff rotating... And so no flow is detected...

Hi @FilipDem ,

I have the same issue with my sensor (1.0.4), but I don't understand the shared documentation.
Could you please help ? Thanks

FilipDem, 2026-08-25:
This worked for me.
[image]