Koenkk/zigbee2mqtt #23314: Cannot add install code (0xb2: APS_TABLE_FULL)

TheMagnetar, 2024-07-10:
### What happened?

I am trying to add an install code but I get instead:

z2m: Request 'zigbee2mqtt/bridge/request/install_code/add' failed with error: 'SREQ '--> APP_CNF - bdbAddInstallCode - {"installCodeFormat":1,"ieeeaddr":"0x18FC260000A4DDC1","installCode":{"type":"Buffer","data":[156,84,118,213,111,227,221,156,84,4,155,170,135,77,192,139,194,113]}}' failed with status '(0xb2: APS_TABLE_FULL)' (expected '(0x00: SUCCESS)')'

### What did you expect to happen?

Code successfully added

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Docker

TheMagnetar, 2024-07-10:
Can this limitation be overcome by:

1) deleting older installation codes?
2) Using a different controller and driver?

Koenkk, 2024-07-11:
Does reflashing the adapter solve it?

TheMagnetar, 2024-07-15:
I will try and report back!

TheMagnetar, 2024-07-16:
So @Koenkk I have flashed the adapter again with `cc2538-bsl.py -ewv` and that did not help. Moreover trying to add an install code hard crashes zigbee2mqtt without logging any error.

Koenkk, 2024-07-17:
> install codes are still seen in the web interface

Where do you see this in the web interface? Could you also give https://github.com/Koenkk/Z-Stack-firmware/discussions/505 a try?

TheMagnetar, 2024-07-18:
It is when pressing the smaller button under "Add install code"

[image]

> Could you also give [Koenkk/Z-Stack-firmware#505](https://github.com/Koenkk/Z-Stack-firmware/discussions/505) a try?

I will

TheMagnetar, 2024-07-18:
I have tested it and nothing has changed. The same happens. It seems to me that the install codes are written somewhere in the adapter memory. Is there a way to remove them through the Texas SDK? I have looked at the API but I could not find any way to do that.

Koenkk, 2024-07-20:
Do you see the link keys in the `data/coordinator_backup.json`?

TheMagnetar, 2024-07-21:
Yes, there are some link keys. They correspond to both kind of devices (those requiring install code and those not requiring install code). Install codes are not listed.

Koenkk, 2024-07-21:
Can you try with this FW?  (it's for the sonoff dongle P)

[znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang.hex.zip](https://github.com/user-attachments/files/16324778/znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang.hex.zip)

TheMagnetar, 2024-07-23:
I have tested it and the problem persists. Zigbee2MQTT container crashes when trying to add an install code. No error or warnings are seen in the logs.

Koenkk, 2024-07-24:
> Zigbee2MQTT container crashes when trying to add an install code. No error or warnings are seen in the logs.

Before you got an error right?

> z2m: Request 'zigbee2mqtt/bridge/request/install_code/add' failed with error: 'SREQ '--> APP_CNF - bdbAddInstallCode - [...]

TheMagnetar, 2024-07-25:
Yes. It crashes before I get any error.

TheMagnetar, 2024-07-26:
Since reflashing the device, the error does not appear. 

> Does reflashing the adapter solve it?

Obmawo, 2024-10-13:
Hi,

I have the exact same problem with my Sonoff Dongle Plus... I have already flashed the latest firmware... but I get always the "APS_TABLE_FULL" error...

How can I remove the old install codes? Or what is the best advice here?

SteffenLoges, 2025-01-15:
I'm facing the exact same problem. Is there a way to delete old install codes?

Obmawo, 2025-01-17:
@Koenkk 

Happy to hear any advices from you :-)

Koenkk, 2025-01-19:
We currently don't have a way to nicely remove install codes, a re-flash of the adapter is the only way for now (to be improved)