Koenkk/zigbee2mqtt #23320: Options for new devices not saving

YewTreeWeb, 2024-07-11:
### What happened?

Since updating from 1.37 to 1.38 any settings I set for newly added devices get removed on restart. If I try to add a setting in the UI the checkboxes instantly clear themselves and do not save. If I add a device setting option in the z2m configuration.yml file the saved option doesn’t show in the UI and is removed from the config after restarting. This now means I cannot retain any new devices.

### What did you expect to happen?

Device settings should save in UI and config and persist after restarting.

### How to reproduce it (minimal and precise)

Add a device option in z2m config and restart Home Assistant.

### Zigbee2MQTT version

1.38.1

### Adapter firmware version

20240710.0

### Adapter

Combi II

### Setup

Home Assistant OS in a Proxmox VM

maciekdnd, 2024-07-16:
Same here with 1.38.0 with Z2M in LXC + SLZB6M. Unable to save debounce. It is nearly impossible to save value. It clears right away (I have to hit number than enter, and than when I have a luck, it may stay.

YewTreeWeb, 2024-07-18:
I have updated to version 1.39.0 and the issue is still there.

Wielebny666, 2024-07-18:
1.39.0-dev commit: 8c71afd this same.
Z2M Docker
I have the impression that Z2M is regressing instead of progress

jesse-kaufman, 2024-07-24:
I've had this issue on and off for the last number of releases as well as on the dev branch .. same issue whether I access directly or go through a reverse proxy ..

it seems like there's a background process pulling the config and updating the UI every second or so .. sometimes I can paste the name of a device and hit enter before it refreshes the values, other times it happens too fast 

I wonder if it's related to the issue I'm also seeing where clicking on the check for update and update firmware buttons sometimes require 5 or 6 clicks before anything actually ends up happening .. it's like the UI is constantly refreshing (but not a full page load)

RubenKelevra, 2024-09-02:
I can confirm it with 1.40.0, too:

[Screencast_20240902_022120.webm](https://github.com/user-attachments/assets/782efd37-6934-4683-8b0a-752c7f6ece8e)

This is not limited to this checkbox, but all checkboxes. (probably more elements)

@jesse-kaufman yeah found that workaround, too. So I can disable and enable devices. Click on the checkbox and hit enter instantly will save the settings.

But that's not ideal.

@Koenkk please have a look if you have time. :)

Koenkk, 2024-09-02:
I think every time there is an update the UI resets, but anyway, this issue should go into the frontend repo.

RubenKelevra, 2024-09-03:
@Koenkk can you move the issue or should we recreate it?

Koenkk, 2024-09-03:
I cannot move issues to the z2m frontend unfortunately.

Skeletorjus, 2024-09-03:
This issue is already in the frontend repo.
https://github.com/nurikk/zigbee2mqtt-frontend/issues/2080

closed 2024-09-04 by Koenkk