Koenkk/zigbee2mqtt #23323: Devices that not support genGroups should not be able to be added to a group

JoostV, 2024-07-11:
### What happened?

I have (a lot) of SONOFF ZBMINIL2 devices: https://www.zigbee2mqtt.io/devices/ZBMINIL2.html. I found out the firmware of this device does not (currently) support genGroups, so it won't respond to any group messages. Futhermore if you remove if from the group it stays in there forever even if you remove if from the file.

Apart from it not working on the group message, I think it would be much better if the device is not allowed to be added to a group if it does not support the genGroups.

### What did you expect to happen?

Not be allowed to add a device that does not support the genGroups. So either a error message that it does not support genGroups or to filter it out, so that you can't add it to the group.

### How to reproduce it (minimal and precise)

Add a SONOFF ZBMINIL2 to a group.
Send a group command.

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

7.4.2 [GA]

### Adapter

ember

### Setup

Add-On on HA

JoostV, 2024-07-15:
This is a normal working lamp / switch
[image]

This is a SONOFF ZBMINIL2:
[image]

As you can see, it misses the "genGroups" so it does not respond to group commands. But you can add it to a group. I think it would be better to not be able to add devices to a group that do not have the genGroups implemented.

Furthermore, if you want to remove a SONOFF ZBMINIL2 from a group you are not able to do so. That behavior should in my opinion also be changed.

Koenkk, 2024-07-16:
Do you get any errors when adding it to the group? It seems the ZBMINIL2 supports it "a bit', otherwise the `genGroups` `add` command should fail when sending it to the device.

hmmbob, 2024-08-08:
Duplicate of #22355

closed 2024-12-28 by github-actions[bot]