Koenkk/zigbee2mqtt #23325: Wrong firmware flashed on IKEA LED2201G8, bulbs only do flash now, offline

MaxRower, 2024-07-11:
### What happened?

Today new a firmware upgrade for IKEA bulbs was offered, so I started to upgrade the corresponding bulbs.
Sometime later I noticed my LED2201G8 in the basement do flash every few seconds. They are offline, factory reset does not work.
After looking for the cause, I noticed the new 3.0.21 is not for the LED2201G8, but only for my other LED2107C4, which do work fine after the upgrade.
Since the firmware versions of both bulbs types where the same before, I think, there has to be some sort of error by IKEA? Or is it z2m?

### How to reproduce it (minimal and precise)

n/a

### Zigbee2MQTT version

1.38.0 commit: fe048e6

### Adapter firmware version

20221226

### Adapter

Sonoff USB P

### Setup

docker in lxc container

### Debug log

Logs for LED2201G8:
[2024-07-11 13:33:44] info:     z2m: Update available for 'Keller Lampe 2'
[2024-07-11 13:33:44] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Keller Lampe 2', payload '{"brightness":254,"color_mode":"xy","color_options":{"execute_if_off":true},"last_seen":"2024-07-11T13:33:44+02:00","linkquality":200,"power_on_behavior":"on","state":"OFF","update":{"installed_version":50331664,"latest_version":50331681,"state":"available"},"update_available":null}'
[2024-07-11 13:33:44] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Keller Lampe 2","updateAvailable":true},"status":"ok","transaction":"izs9s-2"}'
[2024-07-11 13:33:44] info:     z2m: Update available for 'Keller Lampe 1'
[2024-07-11 13:33:44] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Keller Lampe 1', payload '{"brightness":254,"color_mode":"xy","color_options":{"execute_if_off":true},"last_seen":"2024-07-11T13:33:43+02:00","linkquality":196,"power_on_behavior":"on","state":"OFF","update":{"installed_version":50331664,"latest_version":50331681,"state":"available"},"update_available":null}'
[2024-07-11 13:33:44] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Keller Lampe 1","updateAvailable":true},"status":"ok","transaction":"izs9s-1"}'

Logs for LED2107C4:
[2024-07-11 13:52:20] info:      z2m:mqtt: MQTT publish: topic 'fkg/Schlafzimmer Nachtkästchenlampe rechts', payload '{"action":"off","brightness":1,"color_mode":"color_temp","color_options":null,"color_temp":370,"last_seen":"2024-07-11T13:52:20+02:00","linkquality":144,"power_on_behavior":"on","state":"OFF","update":{"installed_version":50331664,"latest_version":50331681,"state":"available"},"update_available":null}'

MaxRower, 2024-07-11:
Corresponding entries in http://fw.ota.homesmart.ikea.net/feed/version_info.json
Maybe the wrong entry is already removed?

{"fw_binary_url":"http://fw.ota.homesmart.ikea.net/global/GW1.0/01.21.031/bin/zingo_ws-0x2200-3.0.21-prod.ota.ota.signed","fw_file_version_LSB":33,"fw_file_version_MSB":768,"fw_filesize":307758,"fw_image_type":8704,"fw_manufacturer_id":4476,"fw_type":2},
{"fw_binary_url":"http://fw.ota.homesmart.ikea.net/global/GW1.0/01.21.031/bin/zingo_ws_soc-0x2206-3.0.21-prod.ota.ota.signed","fw_file_version_LSB":33,"fw_file_version_MSB":768,"fw_filesize":306714,"fw_image_type":8710,"fw_manufacturer_id":4476,"fw_type":2}

Bkrickl, 2024-07-11:
I also flashed of my LED2201G8 to the newest version.
It is till working for me but now when i change the brightness the temperature is reset to some default value.

CV8R, 2024-07-12:
Same problem - is there a way to roll the firmware back or is the bulb dead?

sverrevh, 2024-07-12:
Came here to comment the same. Seems that 3.0.10 broke the bulb. @Koenkk can you prevent others from flashing this firmware to their bulbs asap?

Bkrickl, 2024-07-12:
I also flashed two of my LED2201G8 and it seems to work fine for me.
I am currently on Z2M 1.36.1.

The only thing i needed to do was to re add the device because it had some strange behaviour (everytime i increased/decrease the brightness the color temp was reseted to some default value)

MaxRower, 2024-07-12:
> I also flashed two of my LED2201G8 to 3.0.21 and it seems to work fine for me. I am currently on Z2M 1.36.1.
> 
> The only thing i needed to do was to re add the device because it had some strange behaviour (everytime i increased/decrease the brightness the color temp was reseted to some default value)

Maybe you have a different hardware revision of the bulb? I bought mine not even a month ago. I will take a close look at the bulb itself.

phrozen77, 2024-07-12:
Errr.. sh... upgrade is in progress right now (z2m: 1.36.1) - i guess i'll find out what's in store for me as somehow stopping the upgrade is probably a worse idea than potentially bricking one bulb?

MaxRower, 2024-07-12:
> Errr.. sh... upgrade is in progress right now (z2m: 1.36.1) - i guess i'll find out what's in store for me as somehow stopping the upgrade is probably a worse idea than potentially bricking one bulb?

I'd turn it off now, before it is at 100%. It happens all the time an update can't be finished, because someone turns the switch off.

phrozen77, 2024-07-12:
Lets see what happens - switched the bulb off, Z2M still pretends the update is in progress...

MaxRower, 2024-07-12:
The problem seems to exist when using the tradfri gateway as well, there's a reddit post as well, with my comments:
https://www.reddit.com/r/tradfri/comments/1e0ohmc/why_are_my_lamps_in_disco_mode/

MaxRower, 2024-07-12:
> Lets see what happens - switched the bulb off, Z2M still pretends the update is in progress...
> 
> Edit: red popup came up, saying that the device didn't request any further data and called the update off...
> 
> Edit2: everything still seems to work correctly - update was only at 3.5%ish or something tho...

You're a lucky man!

phrozen77, 2024-07-12:
Indeed! I need to level up to a "wise man" tho - read sh*t about updates before clicking any damn buttons :D

MaxRower, 2024-07-12:
> Indeed! I need to level up to a "wise man" tho - read sh*t about updates before clicking any damn buttons :D

Well, this seems to be part of our education to "wise man".

Bkrickl, 2024-07-12:
> Maybe you have a different hardware revision of the bulb? I bought mine not even a month ago. I will take a close look at the bulb itself.

I bought mine like 3-4 month ago.
Here is a screenshot of my bulb which is updated and working.

[image]

phrozen77, 2024-07-12:
> After looking for the cause, I noticed the new 3.0.21 is not for the LED2201G8, but only for my other LED2107C4, which do work fine after the upgrade.

Just for the record: the LED2201G8 is the only lightbulb i've got on my Zigbee network... So, z2m (or Ikea, rather) addresses this bulb with that firmware update...

enbyted, 2024-07-12:
I had the update offered for LED2201G8, however after re-triggering the update check a few minutes ago the update disappeared, so I guess the issue is resolved now?

AndreasFlies, 2024-07-12:
Same here. Bought today. Old firmware was 3.0.8 (still shown in zigbee2mqtt)

enbyted, 2024-07-12:
> Same here. Bought today. Old firmware was 3.0.8 (still shown in zigbee2mqtt)

Latest valid firmware for LED2201 is 3.0.10, so if you have 3.0.8 then you should see the update. See the release notes page: https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html

wakuwaku, 2024-07-12:
Same here today. Slow flashing cold white light after updating my 2 LED2201G8 bulbs. Home Assistant/zigbee2mqtt has no control anymore after the update.

If they a bricked, how to swamp them for new bulbs at IKEA ?

robertnisipeanu, 2024-07-13:
Well, just upgraded mine today and it's bricked. Sadly found out too late about this github issue.

Is there any way to reflash it or do I have to try convince IKEA to exchange the bulbs?

bmork, 2024-07-13:
Ouch.  Same experinece here.  Interestingly enough, I updated two LED2201G8 from 3.0.10 to  3.0.21 and one them is working fine, while the other is blinking continuously.  They were bought together but are probably from different batches, since they came with different firmware revisions originally.

Meckslo, 2024-07-13:
> Well, just upgraded mine today and they're bricked. Sadly found out too late about this github issue.
> 
> Is there any way to reflash it or do I have to try convince IKEA to exchange the bulbs?

My devices were bricked yesterday by the upgrade. I went to IKEA today and exchanged them. They accepted them without any issues. However, it didn't seem like the saleswoman was aware of the error.

wakuwaku, 2024-07-13:
Just wonder if you get the same issue when installing the new bulbs. If we get new LED2201G8 and (auto) update them, do we repeat the issue?

Meckslo, 2024-07-13:
> Just wonder if you get the same issue when installing the new bulbs. If we get new LED2201G8 and (auto) update them, do we repeat the issue?

I now have firmware 3.0.8 on the new bulbs. Currently, no new update is being offered through zigbee2mqtt.

wakuwaku, 2024-07-13:
IKEA has confirmed to sent me 2 new LED2201G8 bulbs. 
I have just called customer service and explained them on internet there are bulb owners complaining about their bulbs has been bricked after the (auto) firmware update and leave them with out-of-control flashing bulbs.

Hope the "wrong" firmware is removed.

sverrevh, 2024-07-13:
I only updated one LED2201G8 I had. I just returned the bricked bulb to IKEA and bought a new one. This one has firmware 2.4.5. Zigbee2mqtt still wants to update it to 3.0.21 (I think?, HomeAssistant lists it as 50331681) and also for the other LED2201G8 the update that bricks them is still available.

fldc, 2024-07-15:
Found this thread after one of my bulbs starting blinking after a firmware update, my newest LED2201G8 IKEA bulb survived this update while an older one did not?

MaxRower, 2024-07-15:
Maybe we should document, which bulbs survive the firmware update, and which do not? On my two bulbs, there are numbers printed, that differ somewhat, so maybe this relates to a hardware revision? Numbers are following the text "TRADRFI WS".
Those do not work with firmware 3.0.21, and had firmware 3.0.10 before the attempted upgrade:
2346 50050
2337 50050

I already contaced IKEA support for Austria, and they said, I should return the bulbs in the shop.

bmork, 2024-07-15:
Can't check my working LED2201G8 right now, but the failing one is
2347 50050

ieee addresses might be an interesting variable as well?  My two LED2201G8 bulbs use different address series/OUIs
failing: 0x881a14fffexxxxxx
working: 0x980c33fffexxxxxx

fldc, 2024-07-16:
> Can't check my working LED2201G8 right now, but the failing one is
> 
> 2347 50050
> 
> 
> 
> ieee addresses might be an interesting variable as well?  My two LED2201G8 bulbs use different address series/OUIs
> 
> failing: 0x881a14fffexxxxxx
> 
> working: 0x980c33fffexxxxxx
> 
> 
> 
> 
> 
> 

The failed ones were all 0x881, but I also have a 0xb43522fffexxxxxx which updated fine.

sverrevh, 2024-07-16:
I bought two lamps at the same time, returned the other one that broke during the update already. Remaining one indeed starts with 0x811a1.....

Bkrickl, 2024-07-17:
Yesterday they released a new firmware update for these bulbs: https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html

Someone already checked if this updated is working?

vincentkoevoets, 2024-07-17:
> Yesterday they released a new firmware update for these bulbs: https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html
> 
> Someone already checked if this updated is working?

I would love to, but the bulbs are no longer reachable so how would we do - that? Mine appear to be completely bricked.

lukasgierth, 2024-07-17:
I was one of the lucky ones where my bulp is still working with 3.0.10 and the update to the broken 3.0.21 never run through.
Would be able to test it, but can i see somewhere in zigbee2mqtt if the version it wants to update to is actually 3.0.22? I dont see anything useful in the Logs section. Don't want to risk my device trying to update to the definitely not working 3.0.21...

bmork, 2024-07-17:
> Would be able to test it, but can i see somewhere in zigbee2mqtt if the version it wants to update to is actually 3.0.22? 

If you use the z2m web frontend:  Select the "state" tab on the device page.  It will  display the available version as "latest_version" under "update".  For example from my LED2201G8 currently running 3.0.21 (the lucky one that survived):
```
{
    "brightness": 198,
    "color_mode": "color_temp",
    "color_temp": 370,
    "linkquality": 18,
    "state": "OFF",
    "update": {
        "installed_version": 50331681,
        "latest_version": 50331682,
        "state": "available"
    },
    "color_options": null,
    "power_on_behavior": null,
    "update_available": null
}
```

Now those numbers are a bit difficult to read.  You need to convert them to hex to make any sense out of it:
```
50331681 = 0x3000021
50331682 = 0x3000022
```

If you run Home Assistant , then it will also report that decimal number as the available version.

lukasgierth, 2024-07-17:
[image]

Update ran through without problems, LED works absolutely fine with v3.0.22.

bmork, 2024-07-17:
And to add another success:  I went and got a LED2201G8 instead of the LED2103G5 mistake (starting to know my way between the "Returns" and "Trådfri" departments now :-).

The new LED2201G8 came with version 2.4.5. Updated it to 3.0.22 and it still works.

sverrevh, 2024-07-17:
Did someone already manage to update their LED2201G8 with address 0x881... successfully to 3.0.22?

McFlyPartages, 2024-07-18:
WARNING, I've updated to 50331682 (3.0.22) for LED2201G8 but I'm having the same problem, so I think it's best to wait.

McFlyPartages, 2024-07-18:
> I already updated 10 of my bulbs an no issues so far. But bad to hear that it is not working for you.

Yes, it's weird, I'll see if the bulb reference is the same as the Z2MQTT tonight.

m42j, 2024-07-18:
Only have a single samplepoint.
LED2201G8 with IEEE Address 0x881a...
Updated fw 3.0.10 to 50331682/3.0.22 and it started blinking, ZCL commands timeout. =7

vincentkoevoets, 2024-07-18:
I might have stumbled onto something. Today I swapped my 6 bricked bulbs at the store for new ones. I paired them all in Z2M and everything worked fine. Then I went forward with updating, and from these 6 only 5 succeeded. The IEEE's were all seemingly random, the succeeded ones started with:
- 0x60ef
- 0x8c6f
- 0x980c
- 0x8c6f
- 0x8c6f
And the failed one started with 0xb0c7. So I'm starting to thing it has really nothing to do with the IEEE's. 
The thing that stood out for me was the firmware version prior to updating. The one that failed was the only one that was on version 3.0.8, while all the ones that succeeded were on version 2.4.5. I could see that for all the bulbs the offered firmware was the latest 3.0.22. This made me think that the problem maybe lies in the prior firmware version? The one that is present when the update starts? I don't know how that could make any sense but the different firmware version was the only thing that stood out.

McFlyPartages, 2024-07-19:
In my case, it's only the LED2201G8s that have had this problem.

Here are the IEEE addresses, they were all 3.10 before or 3.08 before the update.
* 0x28dba7fffecfee8b
(0xD4CD)
* 0x28dba7fffea9e67b
(0xF26C)
* 0x881a14fffe13e037
(0x0E36)
* 0x881a14fffe2323d0
(0xE012)
* 0x881a14fffe36d8ee
(0x0DB0)
* 0x28dba7fffea9c8b9
(0xA81A)

[image]
[image]
[image]

LeoeLeoeL, 2024-07-20:
Same device different bugs?
https://github.com/Koenkk/zigbee2mqtt/issues/21820

jnswb, 2024-07-20:
Updated yesterday 3 bulbs to 3.0.22 and everything works fine.

vincentkoevoets, 2024-07-21:
> In my case, it's only the LED2201G8s that have had this problem.
> 
> Here are the IEEE addresses, they were all 3.10 before or 3.08 before the update.
> 
> * 0x28dba7fffecfee8b
>   (0xD4CD)
> * 0x28dba7fffea9e67b
>   (0xF26C)
> * 0x881a14fffe13e037
>   (0x0E36)
> * 0x881a14fffe2323d0
>   (0xE012)
> * 0x881a14fffe36d8ee
>   (0x0DB0)
> * 0x28dba7fffea9c8b9
>   (0xA81A)
> 
> ![Capture d’écran du 2024-07-19 15-06-50](https://private-user-images.githubusercontent.com/44530252/350403771-22a495f8-a538-4102-9112-895154fe2b58.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE1NTY1MzgsIm5iZiI6MTcyMTU1NjIzOCwicGF0aCI6Ii80NDUzMDI1Mi8zNTA0MDM3NzEtMjJhNDk1ZjgtYTUzOC00MTAyLTkxMTItODk1MTU0ZmUyYjU4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzIxVDEwMDM1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE4MDQzMjFjY2ZiMjRmODZmNGZhMTE0OTMxNmIzZWU0ZTA1OWZmMjQzNTNmZDczNzU1Nzg3MGY2ZDJmNTc2NmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.yf7WcH5KUzoXyyySOTP6R0cmyHfXBzqMdCi3SjVSDRU) ![Capture d’écran du 2024-07-19 15-06-25](https://private-user-images.githubusercontent.com/44530252/350403781-f02fb173-1c63-445f-ab33-632f4f437e0a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE1NTY1MzgsIm5iZiI6MTcyMTU1NjIzOCwicGF0aCI6Ii80NDUzMDI1Mi8zNTA0MDM3ODEtZjAyZmIxNzMtMWM2My00NDVmLWFiMzMtNjMyZjRmNDM3ZTBhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzIxVDEwMDM1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIzY2I1MDQ1N2VkYmI5ZTk0NWI3Yjc3ODczYzVhMmIzZjQzMWUyNWM0ZDk1NTljN2RkODIyNjEyNjYyYzRlZGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8ZKcWIC2z2jZ2u3Hfo7WggObMnFQIJzCBnVCrAOf59E) ![Capture d’écran du 2024-07-19 15-05-33](https://private-user-images.githubusercontent.com/44530252/350403790-cb925720-2313-47df-b4b6-565986be9bc0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE1NTY1MzgsIm5iZiI6MTcyMTU1NjIzOCwicGF0aCI6Ii80NDUzMDI1Mi8zNTA0MDM3OTAtY2I5MjU3MjAtMjMxMy00N2RmLWI0YjYtNTY1OTg2YmU5YmMwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzIxVDEwMDM1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM1YzQ1MGMxY2E5NWJkN2EzMmE2YzA3NTUwZGQwYzM3MWIyNjIwZGJmNzllZWFlYTlkNjBlM2IwNjA1NDhlYjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.1DR8ZSaffkFO9GqelOBKangf6muKjYrIT2BQus7c0Sc)

So it begins to look more like some sort of russian roulette now. I don't see anything else that sets the failing ones apart from the ones that do successfully update. Could it be something in the update sequence or circumstances? I know from experience these bulbs need to be very close to the coordinator while pairing, maybe they need to be just as close while updating? These might be the cheapest bulbs on the market, but they sure also seem to be the pickiest.

m42j, 2024-07-21:
> So it begins to look more like some sort of russian roulette now. I don't see anything else that sets the failing ones apart from the ones that do successfully update. Could it be something in the update sequence or circumstances? I know from experience these bulbs need to be very close to the coordinator while pairing, maybe they need to be just as close while updating? These might be some of the cheapest bulbs on the market, but they sure also seem to be the pickiest.

I read McFly's list as failed lights, so IEEE 0x881a would fall in line with other failed lights,  no? (And adding 0x28db to the possibly troublesome ones.) IEEE range could indicate hardware revisions/production runs, that's why it's interesting to me. But only IKEAs supplier(s) actually know if that's the case.

G-glop, 2024-07-24:
I got a replacement bulb with fw 3.0.8 and OTA immediately killed it, don't know what version was offered because it doesn't show in the log.

fldc, 2024-07-25:
> I got a replacement bulb with fw 3.0.8 and OTA immediately killed it, don't know what version was offered because it doesn't show in the log.
> 
> Could we not be making e-waste?

Looks like bulbs shipping with 3.0.8 is the ones bricking, that was the case here at least.

sverrevh, 2024-07-27:
> Did someone already manage to update their LED2201G8 with address 0x881... successfully to 3.0.22?

To answer my own question, just updated my 0x881.. Bulb to 3.0.22 from 3.0.10 without issues.

LeoeLeoeL, 2024-07-27:
> To answer my own question, just updated my 0x881.. Bulb to 3.0.22 from 3.0.10 without issues.

I made the same but contunue to suffer about "spontaneous illuminations"

vincentkoevoets, 2024-07-28:
> > I got a replacement bulb with fw 3.0.8 and OTA immediately killed it, don't know what version was offered because it doesn't show in the log.
> > Could we not be making e-waste?
> 
> Looks like bulbs shipping with 3.0.8 is the ones bricking, that was the case here at least.

That might be it. I returned my last bricked bulb yesterday and got a new one, this on again has 3.0.8. I think I will keep this one just the way it is for now, just to be sure.

G-glop, 2024-07-31:
Update: one of the bulbs i've previously updated to 3.0.22 just started flashing after i messed with the "Execute if off" setting.

sverrevh, 2024-07-31:
@G-glop thanks, I had the same. With that setting on the light goes on, goes off and goes on again when turning on. After disabling this is not happening anymore

Belaial, 2024-09-09:
Add me to the list of failed bulbs.... found this thread to late :disappointed: 

I have 3 x LED2201G8 and all are bricked now, bought them 2 weeks ago

0x28dba7fffe5fb1ff
0x881a14fffe0ea0a6
0x28dba7fffe5ff182

They all had 3.0.8 from factory, update was to 3.0.22, one bulb managed to report the correct version before it died it seems but the other two are still showing 3.0.8, all are offline now and just flashing white / orange, re-pairing does not work..... feels great.....

[image]

LeoeLeoeL, 2024-09-09:
I don't understand why these bulbs have so many different behaviours.

vincentkoevoets, 2024-09-09:
It remains a mystery indeed. Today I updated my last 3.0.8, expecting it to fail but it upgraded without problems. It did took some precautions, like turning off the 'Execute if off' setting in Z2M, and putting the bulb right next to the Zigbee coordinator. So maybe that helped, but either way there is no way to know for sure I'm afraid.

LeoeLeoeL, 2024-09-09:
I can pair these bulbs to Z2M or Deconz and they work for a little time.
Then they start to switch on without any command or they leave the network.
Under Hue bridge they have the opposite behaviour.
Sometimes when i want to switch them on they appear on in HA or Hue app but they stay off. To fix it I have to power them off.

Belaial, 2024-09-10:
Spent a few hours troubleshooting this last night but they are truly bricked, nothing I tested brings them back or puts them in paring mode again, worst part is there is no IKEA where I live so will be a month before I can get them replaced.

This is the first time ever I had any issues at all with a Z2M OTA of any kind, amazed it turned out to be a big player like IKEA.

Also reading on Reddit there are even people with IKEAs own hub having these issues so..... yeah..... I don't think this is related to Z2M at all.

Belaial, 2024-09-10:
Came to think of the title on this issue "Wrong firmware flashed on IKEA LED2201G8", do we actually know this or was this the assumption when the issue first started?

Looking at the release notes here for the version I flashed yesterday https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html

```
Release Version: 1.27.1.1
16th July 2024
New features and changes in Accessories:
TRÅDFRI white spectrum bulb E27 (3.0.22)
Product ID: LED2201
◆ Bug fixes.
```

It does seem to be a planned release for the correct hardware type, so I guess the title is old and not updated on this issue.

[12 more replies]

closed 2024-12-28 by github-actions[bot]