Koenkk/zigbee2mqtt #23334: Improved images of existing Zigbee devices for your repo/master

mackonsti, 2024-07-13:
### Is your feature request related to a problem? Please describe

I'd like to contribute with a bunch of cleaner images for existing devices in your database, that I own (and saw on front-end of Zigbee2MQTT) that I processed myself on Photoshop for their transparency.

### Describe the solution you'd like

Please check and accept my small contribution in the below PNG graphical elements.

### Describe alternatives you've considered

Keep using existing images but lower resolution and background not removed well.

### Additional context

These are 3+1 images here of devices that one can find on Asian sellers, I did my best to render them transparent, at least they are higher quality than existing ones. I hope you will approve them. I am not too familiar with git procedures (clone etc.) and there was no mention anywhere how to contribute with images only @Koenkk ! So apologies to use this type of submission, here. Let me know where I can keep contributing.

P.S. The existing door/window sensor PNG is in good quality so I am not submitting a replacement image.

[image]

[image]

[image]

[image]

[image]

Koenkk, 2024-07-13:
Thanks for your contribution, the following instructions can be used to update images:

-----------------

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `MODEL.png` (update `model` accordingly) 
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

mackonsti, 2024-07-13:
Thanks @Koenkk for this, I will try doing so this morning!
And thank you in general for this nice work 👍

mackonsti, 2024-07-13:
@Koenkk if images are not so big such as 512x512 would you still accept transparent ones but smaller? (and square) e.g. 400x400 etc.? Thanks.

P.S. There are thousands of image files 😬 I wanted to link the generic files but seems I need to do more digging 😆

Koenkk, 2024-07-13:
Images should either be 512x512 or 150x150

closed 2024-12-28 by github-actions[bot]