Koenkk/zigbee2mqtt #23335: z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'includes')

janivr, 2024-07-13:
### What happened?

Suddenly, my device stopped working and I get this error for any action on the device.

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
//const extend = require('zigbee-herdsman-converters/lib/extend');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_7a5ob7xq',
        },
    ],
    //zigbeeModel: ['TS0601'],
    model: 'TS0601_new',
    vendor: 'Tuya',
    description: 'Uri Curtain Switch plus Lights', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    //extend: [],
    //fromZigbee: [fz.ignore_basic_report, fz.tuya_switch],
    //toZigbee: [tz.tuya_switch_state],
    fromZigbee: [tuya.fz.datapoints], // We will add this later
    toZigbee: [tuya.fz.datapoints], // Should be empty, unless device can be controlled (e.g. lights, switches).
	//fromZigbee: [tuya.fzDataPoints],
    //toZigbee: [tuya.tzDataPoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l3').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l4').setAccess('state', ea.STATE_SET)
    ],

    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
        ],
    },

    //meta: {multiEndpoint: true},
    //exposes: [e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
    //        e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
    //        e.switch().withEndpoint('l3').setAccess('state', ea.STATE_SET),
    //        e.switch().withEndpoint('l4').setAccess('state', ea.STATE_SET)], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
	//endpoint: (device) => {
    //        // Endpoint selection is made in tuya_switch_state
    //        return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1};
    //    },
};

module.exports = definition;

`

### What did you expect to happen?

my device will work, all 4 actions.

### How to reproduce it (minimal and precise)

happens all time

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

20220219

### Adapter

sonoff

### Setup

home assistant

### Debug log

[2024-07-13 14:39:45] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/0x04cd15fffefb6680/set' with data '{"state_l1":"ON"}'
[2024-07-13 14:39:45] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'includes')
[2024-07-13 14:39:45] debug: 	z2m: TypeError: Cannot read properties of undefined (reading 'includes')
    at /app/lib/extension/publish.ts:232:60
    at Array.find (<anonymous>)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:232:42)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:23)
    at EventEmitter.emit (node:events:529:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:115:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:147:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:138:27)
    at WebSocket.emit (node:events:517:28)
    at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1211:20)
[2024-07-13 14:39:46] debug: 	zh:zstack:unpi:parser: <-- [254,25,68,129,0,0,10,0,69,194,1,1,0,43,0,43,147,114,0,0,5,0,246,0,7,0,69,194,29,222]

janivr, 2024-08-01:
Hello,
When will I get an answer or help?
Thanks

janivr, 2024-08-04:
Still Exists

janivr, 2024-08-05:
Please help

KoKolaj, 2024-09-02:
After today's update

`[2024-09-02 17:36:39] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:36:49] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:37:00] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:37:27] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:37:39] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:38:06] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:38:21] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:38:33] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:39:03] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:39:19] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')`

KoKolaj, 2024-09-02:
After today's update
```
[2024-09-02 17:36:39] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:36:49] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:37:00] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:37:27] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:37:39] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:38:06] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:38:21] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:38:33] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:39:03] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 17:39:19] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
```

janivr, 2024-09-02:
Solved?

hoskerism, 2024-09-02:
Same here. Every device reporting:
EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')

Started after upgrading yesterday.

maddoglee, 2024-09-02:
I get this error as well

rafapoch, 2024-09-03:
running version 1.40.0 getting the same issue

Domochip, 2024-09-03:
Having the same error using  ZDMS16-1 from Avatto
Here is my z2m relevant infos:
 Zigbee2MQTT version: 1.40.0 commit: 6fa1dab
 Coordinator type: zStack3x0
 Coordinator revision: 20240315

`z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')`

KoKolaj, 2024-09-03:
https://github.com/Koenkk/zigbee2mqtt/issues/23821

silkyclouds, 2024-09-05:
yep, same here 

[2024-09-05 21:17:51] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-05 21:17:51] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')

closed 2024-12-28 by github-actions[bot]