Koenkk/zigbee2mqtt #23336: Zemismart TB26-3 incorrectly detected as TB26-1

jsamr, 2024-07-13:
### What happened?

I switched to Z2M edge version hoping to have my new TB26-3 setup and working. Initially, I thought it was working. After some testing, I realized the wrong model had been identified, and TB26-3 didn't work as expected, see section below.

### What did you expect to happen?

## The device should work (2-ways sync)

- Physical presses on the devices should be reflected in Z2M
- Z2M switch presses should be reflected in the physical device

## The device model should be correctly detected

[image]

### How to reproduce it (minimal and precise)

Pressing the switches from the "Exposes" tab _did_ work. I pressed the 3 UI switches and the 3 lights were switched on physically.
[image]

Switching OFF these UI switches was also reflected physically.

After which I pressed the device physical button, and it was NOT reflected in the UI
[image]

This is not always reproducible. At times, the state resulting from pressing physically on the device is reflected in the UI.

### Zigbee2MQTT version

1.39.0-dev commit: e3fa0bf

### Adapter firmware version

20210708

### Adapter

Sonoff ZBDongle Plus

### Setup

Add-on, HAOS Proxmox VM x86

### Debug log

*I did enable debug level logging, but couldn't find the `data/log` folder in HAOS. Checked in the docker container and couldn't find anything in `/data/log`.

jsamr, 2024-07-13:
Quick update: it seems my reporting missed one big feature I hadn't enabled: 
[image]

After enabling this feature, the state is properly reported after pressing physical buttons. The mismatched model name doesn't seem so far to be an issue. Will test for a few days and confirm in this post.

Koenkk, 2024-07-13:
It seems the wrong model name was used (just a cosmetic thing), updated it.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

jsamr, 2024-07-15:
@Koenkk Confirming it works perfectly fine, thanks!

closed 2024-07-15 by jsamr