Koenkk/zigbee2mqtt #23337: [New device support]: MTG075-ZB-RL - TuYa Zigbee Human Presence Sensor - alternate Manufacturer ID

AlexMKX, 2024-07-13:
### Link

https://www.aliexpress.com/item/1005005909374871.html

### Database entry

{"id":87,"type":"Router","ieeeAddr":"0x70ac08fffef6290e","nwkAddr":13471,"manufId":4098,"manufName":"_TZE200_sbyx0lm6","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_sbyx0lm6","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":68,"powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1720897176867}

### Comments

Just the same device as in https://github.com/Koenkk/zigbee2mqtt/issues/21445 but with different ID.

### External definition

```shell
it was just easier to add the fingerprint _TZE200_sbyx0lm6
```

Koenkk, 2024-07-13:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-07-13 by Koenkk

AlexMKX, 2024-07-14:
You're awesome 👍