Koenkk/zigbee2mqtt #23339: [Feature request]: Further support for PM related clusters (PM1, PM25, PM10)

turboproc, 2024-07-14:
### Is your feature request related to a problem? Please describe

For now there is limited support for PM related attributes:
- only PM10 and PM25 are exposed
- only PM25 cluster has been defined (0x042A)

### Describe the solution you'd like

Would like to suggest to add clusters for PM1 and PM10. Fully aware that these have not been defined in the Zigbee 3.0 standard but at the same time aware that the Matter 1.3 standard follows the same cluster codes, we could define ```msPM1 (0x042C)``` and ```msPM10 (0x042D)```.
Further `pm1` would have to be added to exposes.js

### Describe alternatives you've considered

Of course we could follow the manufacturerSpecfic route.

### Additional context

Background being that I'm developing some new sensors and would like to minimize the amount of device specific configuration.

Koenkk, 2024-07-15:
Would you mind making a PR to add the clusters [here](https://github.com/Koenkk/zigbee-herdsman/blob/f73f65fc0494555dac88a60266ed3c021e8e5bea/src/zspec/zcl/definition/cluster.ts#L2490)?

turboproc, 2024-07-15:
Sure, let me try. Always having little fights with PRs

turboproc, 2024-07-15:
@Koenkk, a PR should have been created. Please check, as said, this is always a little battle.

Koenkk, 2024-07-16:
Merged, thanks!

closed 2024-07-16 by Koenkk