Koenkk/zigbee2mqtt #23343: Ikea e27 LED2109G6 wrong color temp range

capjack92, 2024-07-15:
### What happened?

I'm taking inspiration from another post to highlight a problem that I hope can be solved regarding the temperature range of this light bulb. It should be from 6000k to 1780k, but instead it goes from 4000k to 1780k:

Based on the manual of the bulb, it should have a way broader spectrum of white tones. 

In the generated definition, the range is 250 to 454 mireds:
> ```shell
> extend: [light({"colorTemp":{"range":[250,454]},"color":{"modes":["xy","hs"],"enhancedHue":true}})]
> ```

This is what [the manual](https://www.ikea.com/nl/nl/manuals/tradfri-led-lamp-e27-806-lumen-draadloos-dimbaar-gekleurd-en-wit-spectrum-globe-opaalwit__AA-2402817-2-2.pdf) states:
[image]

Based on my calculations the range should be from 167 to 562.
Cold sky: 1.000.000 / 6000 = 166,66666666666666666666666666667
Candle light: 1.000.000 / 1780 = 561,79775280898876404494382022472

@Koenkk Is that something that is not detected properly by Zigbee2MQTT when the external definition is generated? If you change these values, will that work?

The definition also shows that it should support XY and HS colors, but in the Zigbee2MQTT front-end, only XY is shown, does that need to be fixed? If we have HS, would it add something to the light?
[image]

_Originally posted by @YannickvanVugt in https://github.com/Koenkk/zigbee2mqtt/issues/21755#issuecomment-2179440909_

### Zigbee2MQTT version

1.39

### Adapter firmware version

20220219

### Adapter

Sonoff dongle P

### Setup

add-on on Home assistant os on mini pc

YannickvanVugt, 2024-07-18:
I have been wondering if it could be that the Ikea Dirigera Hub mixes in tints of red for warmer colors and blue for colder colors from it's RGB LEDs. Also not sure what the hub would show if you link it up to Home Assistant via Matter, I don't have one. Maybe someone who looks at this post?

I haven't opened up one of the lights to check if the white channel are separate LEDs, as the bulbs were a gift for someone.

Very curious what is going on and if we can get it fixed, I am sure there are more lights that suffer from this.

capjack92, 2024-07-18:
These are the only ones that suffer from this problem... I have several Ikea bulbs, Hue etc.... For example, the Ikea GU10s have the same color spectrum and temperature as the E27s, but on Z2M with the GU10s I don't have this problem because the temp color range goes from 153 to 500. For the hue White Ambiance and hue color e27 the range are the same @YannickvanVugt

luuuuurin, 2024-09-12:
Did you find a way to fix it?

capjack92, 2024-09-12:
No, i think @Koenkk needs to modify the database or something similar to fix the problem

Koenkk, 2024-09-13:
The values you see in the generated definition are read from the device (so they should be correct). 

 > Based on my calculations the range should be from 167 to 562.

What if you update the external definition with this? Does setting 562 make a diff compared to 454?

capjack92, 2024-09-13:
> The values you see in the generated definition are read from the device (so they should be correct).
> 
> > Based on my calculations the range should be from 167 to 562.
> 
> What if you update the external definition with this? Does setting 562 make a diff compared to 454?

Thank you very much for your attention. Regarding the problem, as you can see from the first post, in the manual there are two readings of the device:
- The first limited if you don't have an IKEA Hub
- The second with a much wider range of colors and white tones if you have the IKEA hub

on z2m it is practically the definition of the first reading, the limited one.
The problem lies in the fact that when you have this bulb in a group with other IKEA Gu10s or Hues, you have different temperature and color results if set simultaneously.
I'll give you an example:
I have a group with these e27s and some IKEA gu10s, I set a temperature of coolest white for the whole group. I end up with the gu10s which indicate that they are in temp color in a cold white, while the e27s fall into the x/y colors like light blue but this is a software limit because in truth with the IKEA hub they can reach the same (or almost) same color temperature range

Koenkk, 2024-09-13:
@capjack92 I guess they set the XY color and not the color temperature then?

capjack92, 2024-09-13:
This happens because the color temperature and color range are incorrect on definition in z2m.
Look at these screenshot taken from the manual of this e27.
[image]
From the GU10 manual report the same exact information but on z2m the gu10 in color temp the range is 153 to 500 (like hue bulb), while the e27 in color temp the range is 250 to 454. Actually, according to the IKEA manual, the E27 bulb should have the same color range and temperature as the GU10 (and the Hue).

Is it possible to modify the database to have the same definitions of gu10 as ranges?

Sorry for my english, is not my first language

Koenkk, 2024-09-14:
Try to stop z2m, edit `data/database.db` and modify the color temp range (`colorTempPhysicalMin`/`colorTempPhysicalMax`) and use the generated definition.

YannickvanVugt, 2024-09-16:
> Try to stop z2m, edit `data/database.db` and modify the color temp range (`colorTempPhysicalMin`/`colorTempPhysicalMax`) and use the generated definition.

Thanks Koen!

Quickly picked up such a bulb for myself yesterday to try your suggestion.

I did just that, the new generated definition shows a wider temperature range, but it does not allow me to select that range in either the z2m interface or Home Assistant. What's next to try?

```
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TRADFRI bulb E27 CWS globe 806lm'],
    model: 'TRADFRI bulb E27 CWS globe 806lm',
    vendor: 'IKEA of Sweden',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[167,562]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};

module.exports = definition;
```

[image]

[image]

Koenkk, 2024-09-18:
Is the converter loaded? (check that the description in the about page is now `Automatically generated definition`)

YannickvanVugt, 2024-09-18:
> Is the converter loaded? (check that the description in the about page is now `Automatically generated definition`)

I didn't get that you wanted me to create a custom definition with those details. 
I just did, but when going from cool (250) to coolest (167) or warm (454) to warmest (562), it does not change any color.
If I change the color via the Home Assistant front end, it just pushes the value back to 250 or 454 after setting it to the lower or higher values. 

So I guess it is a feature only supported by the Ikea Dirigera hub then, it probably does some custom commands to mix in more color, but I am unable to check if it really shows a broader range.

Would there be any way to check that? Or would that require some sniffing of a Dirigera hub, if you could intercept the PAN ID and security keys, to see the packets that it is sending to the bulb?

I guess this ticket can be closed for now as long as there are no other options. Thanks for your help and time! 🙌🏻

capjack92, 2024-09-18:
@YannickvanVugt 

what I noticed is that the gu10 (always IKEA) in the z2m database reports "colorTempPhysicalMin":250,"colorTempPhysicalMax":454 ... But on z2m dashboard it actually has a wider range 153-500

YannickvanVugt, 2024-09-18:
> @YannickvanVugt
> 
> what I noticed is that the gu10 (always IKEA) in the z2m database reports "colorTempPhysicalMin":250,"colorTempPhysicalMax":454 ... But on z2m dashboard it actually has a wider range 153-500

Looking at the source code of Zigbee2MQTT, I see the following for these bulbs:

```
    {
        zigbeeModel: ['TRADFRI bulb GU10 CWS 345lm', 'TRADFRI bulb GU10 CWS 380lm'],
        model: 'LED1923R5',
        vendor: 'IKEA',
        description: 'TRADFRI bulb GU10, color/white spectrum, 345/380 lm',
        extend: [
            addCustomClusterManuSpecificIkeaUnknown(),
            ikeaLight({colorTemp: {range: [153, 500], viaColor: true}, color: true}), // light is pure RGB (XY), advertise 2000K-6500K
            identify(),
        ],
    },
```

Could you get me the generated definition of these? "viaColor: true" looks promosing, but I need to see it from the definition as the format is different and I do not have these bulbs.

Thanks for your reply, it maybe got us on the right path now!

capjack92, 2024-09-18:
`{"id":46,"type":"Router","ieeeAddr":"0xa46dd4fffe4d1934","nwkAddr":52975,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb GU10 CWS 345lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64636],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"onOff":1}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454,"colorMode":1,"currentX":20512,"currentY":13762}},"genLevelCtrl":{"attributes":{"currentLevel":51}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":113,"hwVersion":1,"dateCode":"20230913","swBuildId":"1.0.38","zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1726690613727}
`

This is what results from database.db
sorry but I'm new to z2m so I'm struggling

YannickvanVugt, 2024-09-18:
> `{"id":46,"type":"Router","ieeeAddr":"0xa46dd4fffe4d1934","nwkAddr":52975,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb GU10 CWS [...]

That's one useful part as well, please check step 2. Creating the external definition, the data from there is what I need. You don't need to do 2.1. 😀

capjack92, 2024-09-18:
here it is
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

`const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TRADFRI bulb GU10 CWS 345lm'],
    model: 'TRADFRI bulb GU10 CWS 345lm',
    vendor: 'IKEA of Sweden',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[250,454]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};

module.exports = definition;`

capjack92, 2024-09-18:
@Koenkk 
I also tried as a yannik with the same results
This is the generated definition

`const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TRADFRI bulb E27 CWS globe 806lm'],
    model: 'TRADFRI bulb E27 CWS globe 806lm',
    vendor: 'IKEA of Sweden',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};

module.exports = definition;`

As you can see from the previous post, with the IKEA gu10 I have the same ranges as the e27 in the generated definitions, but in truth the selectable range is 153-500

YannickvanVugt, 2024-09-18:
> here it is
> 
> `const {light} = require('zigbee-herdsman-converters/lib/modernExtend');
> 
> const definition = { zigbeeModel: ['TRADFRI bulb GU10 CWS 345lm'], model: 'TRADFRI bulb GU10 CWS 345lm', vendor: 'IKEA of Sweden', description: 'Automatically generated definition', extend: [light({"colorTemp":{"range":[250,454]},"color":{"modes":["xy","hs"],"enhancedHue":true}})], meta: {}, };
> 
> module.exports = definition;`
> 
> ![gu10](https://private-user-images.githubusercontent.com/108417387/368753610-b96758e6-6ac1-4ad0-8d60-911bd88ee83c.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY2OTY3MDAsIm5iZiI6MTcyNjY5NjQwMCwicGF0aCI6Ii8xMDg0MTczODcvMzY4NzUzNjEwLWI5Njc1OGU2LTZhYzEtNGFkMC04ZDYwLTkxMWJkODhlZTgzYy5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxOFQyMTUzMjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNzNkZWE3M2U0ZGI2NzgwMWUzNmYxMzYxN2I2NjFjODFlM2ViYzQ5ZjQyMDhkYThiYWMzYTI1MDk0ZTE5YmNhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.T3s97lWC12EcCfDvw6YGXiECA1M9qGzh4V2OuHdmcAI)
> 
> as you can see it shows the range as 250-454 in generated definition but in truth from the dashboard I can do 153-500 (and it is a real range compared with hue bulbs seen up close)

**IT WORKS!!!!**

Thanks so much for your help and input, I don't really think that by the original specification these bulbs will/would go further than their advertised generated definition, but thanks to the 'viaColor: true' option it does, but this really works for **ANY** Ikea bulb that would allow that option! It is an option that is only there for Ikea bulbs.

By my testing, colors are EXACTLY the same when using it with and without viaColor, so that is a **VERY GOOD** thing.

I have used some unsupported stuff that's not in the official documentation to test this. 🤭
I have been messing with the following piece of code:

```
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');
const {ikeaLight} = require('zigbee-herdsman-converters/lib/ikea');
// const {DefinitionWithExtend} = require('zigbee-herdsman-converters/lib/types');

const definition = {
    zigbeeModel: ['TRADFRI bulb E27 CWS globe 806lm'],
    model: 'TRADFRI bulb E27 CWS globe 806lm',
    vendor: 'IKEA of Sweden',
    description: 'Automatically generated definition',
    extend: [ikeaLight({"colorTemp":{"range":[100,1000]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    // extend: [ikeaLight({"colorTemp":{"range":[100,1000],"viaColor":true},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    // extend: [addCustomClusterManuSpecificIkeaUnknown(), ikeaLight({"colorTemp":{"range":[167,562],"viaColor":true},"color":{"modes":["xy","hs"],"enhancedHue":true}}), identify()],
    meta: {},
};

module.exports = definition;

// : DefinitionWithExtend[]
```

The currently used, so the not uncommented "extend:", only allows me the default range from 250 to 454.

The second "extend:", first uncommented, allows me fully from the warm red tone (just pure red, 1000k) and "Candle light" (1780k) to "Blue sky" (10000k). As long as you use viaColor, anything will work. So now I can even operate this light outside it's, by manual advertised, range and it all works.

The third "extend:" was just an example, but didn't add much for testing.

Literally any range you throw at it when using the 'viaColor' option will work. From extreme cold (100, 10000k, RGB: 201, 218, 255) to extremely warm (1000, 1000k, almost completely red, RGB: 255, 67, 0). Warmer is not possible, colder is possible, but I haven't checked past 12000k as it's very unlikely that anybody would use that.

@Koenkk : How would you go about the color range?

[image]

Koenkk, 2024-09-19:
When using `viaColor`, are the color temperatures as accurate as when not using it? This was only meant to be used on IKEA bulbs which support color but not color temp.

YannickvanVugt, 2024-09-19:
> When using `viaColor`, are the color temperatures as accurate as when not using it? This was only meant to be used on IKEA bulbs which support color but not color temp.

I only have one bulb at the moment, but it seems to be very accurate from the few datapoints I checked.

I switched everytime between the definition in Z2M and the custom definition and applied the next or previous value to that current temperature and most of the time it didn't change. It's spot on.

Is it with the viaColor option just an RGB value that is send to the bulb? 

I will do some more testing if you'd like.

YannickvanVugt, 2024-09-20:
> @YannickvanVugt I have several bulbs, I can try it myself... Now I'm reading your previous comment, hoping to be able to replicate what you did
> 
> @YannickvanVugt i need your help to do what you did

You will need a second Zigbee2MQTT instance and Zigbee stick to test this! You can not apply this to a single bulb, it will be applied to all bulbs of the same type in an instance.

There is several ways to get this going.
If you only have a single Home Assistant installation you can add the repository for the Z2M addon once more with an extra /
`https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/`
A third instance is possible with
`https://github.com/zigbee2mqtt/hassio-zigbee2mqtt//`

You will need to configure Zigbee2MQTT to use another folder, MQTT topic, Zigbee channel, PAN_ID and different network_key.

In the Zigbee2MQTT folder, create the file 'ikea-test.js' with this code:
```
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');
const {ikeaLight} = require('zigbee-herdsman-converters/lib/ikea');

const definition = {
    zigbeeModel: ['TRADFRI bulb E27 CWS globe 806lm'],
    model: 'TRADFRI bulb E27 CWS globe 806lm',
    vendor: 'IKEA of Sweden',
    description: 'Automatically generated definition',
    extend: [ikeaLight({"colorTemp":{"range":[100,1000],"viaColor":true},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};

module.exports = definition;
```

Add the following to the Zigbee2MQTT configuration.yaml:
```
external_converters:
    - ikea-test.js
```

That should be enough to get that going. 😄

capjack92, 2024-09-22:
@YannickvanVugt 
@Koenkk 
Below I explain what I did:
- Created a virtual machine with HA
- Everything configured and z2m installed with alternative zigbee coordinator
- followed your steps (Creating files in the z2m folder, added the lines of code in configuration.yaml always in the z2m folder
-added an ikea e27 equal to the others on z2m modified and connected next to an ikea e27 without modifications to see the 
  differences
-compare the differences also with an e27 hue color

Assumption: The modified IKEA light bulb has a fuchsia label in the photos. As regards the ranges set in the modification file, I didn't enter 100,1000 but I entered the ranges that both the IKEA gu10 and the e27, gu10 hue color have, i.e. 153,500

First comparison, modified e27 vs original e27 (brightness set to "10" otherwise it was impossible to notice the differences from the photos due to the high brightness).. temp range 500vs454..
I noticed that the modified bulb becomes slightly warmer but the lumens of the bulb automatically increase as shown in the photo below:
[image]

Then I increased the lumens of the original bulb (brightness range to "50"), without changing the temp range and leaving the brightness of the modified E27 unchanged, to make a more accurate comparison with similar bulb brightness:
[image]

Then I made the comparison with the same bulbs, brightness range "10", temp range "153" vs "250"..
Here the difference is clear:
[image]

Next I tested the modified e27 with an e27 hue color. As for the hue color, it has a crazy low brightness capacity (I don't know if it's a limitation of the hardware of the IKEA bulbs or the software, I don't have an IKEA hub to try).
which is why, to make a comparison that makes sense, I increased the brightness range of the hue color to 50, compared to the "10" range of the modified e27.
Here is the comparison with temp range 500vs500:
[image]

Here is the comparison with temp range 153vs153:
[image]

As you can see the color temperature seems really similar.
If you need other tests, just ask, although there is the possibility of trying to somehow bring the brightness of the IKEA bulbs lower.

YannickvanVugt, 2024-09-22:
> @YannickvanVugt
>
> @Koenkk
>
> Below I explain what I did:
>
> - Created a virtual machine with HA
>
> - Everything configured and z2m installed with alternative zigbee coordinator
>
> - followed your steps [...]

Thanks for your tests, it is important to know if both the LED2109G6, if set to the same value, has the same colors.

Of course 454 and 500 do not compare, but it would be great to know if 250 without and with the 'viaColor' option compare. And 280, 300, 320 and so on. To see if the 'viaColor' option is accurate.

Good to know it at least is quite comparable to Hue lights.

capjack92, 2024-09-22:
> Thanks for your tests, it is important to know if both the LED2109G6, if set to the same value, has the same colors.

Do you mean to use both bulbs on the same z2m with the same settings to see if they differ? @YannickvanVugt

YannickvanVugt, 2024-09-22:
> > Thanks for your tests, it is important to know if both the LED2109G6, if set to the same value, has the same colors.
> 
> 
> 
> Do you mean to use both bulbs on the same z2m with the same settings to see if they differ? @YannickvanVugt 

Nope. That wouldn't add anything.

The 'viaColor' option sends an RGB value to the bulb, which is produced by Z2M.

A stock Z2M instance will send a temperature command to the bulb that is processed by the bulb.

We'd like to know if both create the same or somewhat comparable color temperature to each other.

capjack92, 2024-09-22:
@YannickvanVugt 
yes, in fact, I edited the previous comment because I later understood what you meant. As I wrote in the previous comment, tried (but only with temp range 454) and they seemed identical) soon I will also try other ranges

capjack92, 2024-09-22:
Ok done, I think they are identical. @YannickvanVugt @Koenkk 

post the photos below:

temp range 250:
[image]

temp range 300:
[image]

temp range 350:
[image]

temp range 400:
[image]

temp range 450:
[image]

tested also rgb.
Red:
[image]

green:
[image]

blue:
[image]

YannickvanVugt, 2024-09-22:
> Ok done, I think they are identical. Perhaps from the photos it is confusing which of the two has the label, so to be safe I tell you that the modified e27 is the one on the top right. @YannickvanVugt @Koenkk

Thanks mate! That's amazing!

So the 'viaColor' option is very reliable to use as a replacement for color temperature.

@Koenkk Did you only add this for Ikea bulbs for a (specific) reason or would this be a possible solution for other bulbs and strips as well? As long as we know that the calculation from temperature to RGB is reliable, anything is possible right?

Nobody uses either end of the possible spectrum, but a broad range would be nice. Let's say 1700k/1780k up to 6000k or something, as mentioned in the Ikea manual for that bulb and kind of normal range as mentioned on [Wikipedia](https://en.wikipedia.org/wiki/Color_temperature).  

What would be a good temperature range to allow?
1780k to 6000k would be 167 to 562, as with what we started the topic.

Thanks for both your help and input!

capjack92, 2024-09-23:
> > Did you only add this for Ikea bulbs for a (specific) reason or would this be a possible solution for other bulbs and strips as well?
> 
> this is currently only done for LED1624G9 and LED1923R5 since do not support the color temp commands. This bulb does so I'm a bit hesitant to add this (since it bypasses the native commands). Why don't you just set the color? Because now we are basically abusing color temp to set colors. Note that this will also not work when you try to control multiple bulbs via a group.

Yes, in fact, I used the "standard" color temperature values ​​(153,500), those used by Hue and the IKEA Gu10. In doing so, in my opinion, the IKEA E27 performed very well even from a mixed lighting control perspective (Hue , gu10 ikea and e27 ikea),
while with the original configuration I have problems controlling the lights together because the e27 is limited via software, this is the reason why I opened this issue.

@Koenkk What do you think based on the comparison photos I took above?

Koenkk, 2024-09-24:
With controlling lights together, do you mean a Zigbee group?

capjack92, 2024-09-25:
> With controlling lights together, do you mean a Zigbee group?

Yes, I use them both in Zigbee groups and with a customized addon that simulates the functionality of the Philips Hue app on Home Assistant where I can manage them individually and simultaneously in a very simple and clear way.
I'm attaching a screenshot for you to see

[image]

This is a mix of gu10 and e27 ikea

when I go to set a temperature that the e27 does not allow (without modification) automatically sets the maximum temperature that the bulb allows (for example gu10 at 500 while e27 at 454). Except that afterwards it starts to give a series of problems if I try again, for example it doesn't understand if I'm on temperature or color.

Which doesn't happen for example in other groups where I don't have Ikea E27 but I have Hue, Ikea Gu10 and Stripled Zigbee RGBCCT lights. In this group everything works perfectly.

As far as standard light control is concerned, for example, if I create a lighting group (IKEA GU10 and E27) via the home assistant helper or via Z2M and try to set a temperature at which the E27 cannot reach while the GU10 can then brings the temperature of all bulbs to what the e27 allows, limiting the gu10

dirrgang, 2024-10-20:
I just ran into this issue myself. Though it seems like a fix has been found? Is someone else going to do a PR to modify the converters, or should I give it a try?

My bulb in question is a LED2111G6 (E14 globe)

capjack92, 2024-10-20:
> I just ran into this issue myself. Though it seems like a fix has been found? Is someone else going to do a PR to modify the converters, or should I give it a try?
> 
> My bulb in question is a LED2111G6 (E14 globe)

At the moment the only solution is the one proposed above by Yannick, manually modifying the light bulb database by activating the viacolor option with the difference of inserting 153,500 as colortemp range

dirrgang, 2024-11-06:
> At the moment the only solution is the one proposed above by Yannick, manually modifying the light bulb database by activating the viacolor option with the difference of inserting 153,500 as [...]

I'm not that deep in the zigbee2mqtt contribution community, but couldn't someone just... add an appropriate converter for that particular model? Like I said, I wouldn't mind giving it a try, but I'm wondering if I'm missing anything.

RubenKelevra, 2024-11-08:
> Thank you very much for your attention. Regarding the problem, as you can see from the first post, in the manual there are two readings of the device:
>
> * The first limited if you don't have an IKEA [...]

Okay, but z2m should be able to support all of the second options. If the hardware can support it. You don't need an Ikea hub for that.

capjack92, 2024-11-09:
@RubenKelevra as you can see in the following posts with @YannickvanVugt we found a possible solution that solves the problem. The point now is, should we make this change official on Z2M or solve it by Yannick's method? Because i don't know if Yannick's method can cause problems with future Z2M updates

RubenKelevra, 2024-11-09:
> @RubenKelevra as you can see in the following posts with @YannickvanVugt we found a possible solution that solves the problem. The point now is, should we make this change official on Z2M or solve [...]

Well my point was: The documentation by Ikea does not apply for the z2m use case. It only applies if you use Ikea bulbs with say Philips Hue or something like this.

Z2M should be able to support everything the hardware offers.

And if the hardware database in z2m is wrong, just fix it. There's no reason why we should restrict the hardware capabilities IMHO.

So feel free to do a PR. :)

Mushoz, 2024-12-13:
> this is currently only done for LED1624G9 and LED1923R5 since do not support the color temp commands. This bulb does so I'm a bit hesitant to add this (since it bypasses the native commands). Why [...]

I understand you are hesitant about abusing colors to set the temperature range. But what if it was possible for each bulb to have two ranges in the database? One regular as we have right now, and a second optional extended color temperature range. The standard range will have viaColor set to False, while the extended range will have viaColor set to True. This way, as much as the range is controlled the proper way as desired. But you are still able to go outside the normal range into the extended range. Right now, using zigbee2mqtt is a pretty big step down in terms of temperature range versus controlling it through Ikea's app. Hopefully this allows you to please both sides?

RubenKelevra, 2024-12-13:
I don't see a reason why z2m should support the limited color range. It should just expose the maximum possible and the appropriate commands to the bulb.

capjack92, 2025-02-12:
@Koenkk After 3 months of use with z2m modified to fully support what the hardware of this E27 can offer (and which is limited using the original version of z2m) without having found No problem with the usage, both alone and in a group with other zigbee bulbs (Hue bulb and other Ikea Gu10 bulbs) but only advantages. I ask you if it is possible to insert the modification officially or introduce an option In the device settings on z2m such as "true color: on; off" to allow those, like me, who would like to use the full potential of this e27. I apologize for my insistence on the request but in daily use this modification has been a lifesaver for me. Thanks

Mushoz, 2025-02-12:
Just to voice my opinion as well: I understand the desire to use the native commands, and refrain from abusing colors to set the color temperature. However, this change would ONLY impact these bulbs. And right now, z2m is not a good choice for these bulbs (unless modified) since you're effectively giving up a big part of the color range that is available through Ikea's hub / app. So using these native commands looks better from a developer's point of view. But for the actual end-user, it's really not a good implementation.

I share @capjack92's opinion that it might be a good idea to make this configurable OR as I suggested before, maybe we can implement two ranges. The restricted range, which would use native commands to set the color temperature, and an extended "full range". If a color temperature is chosen that is outside the restricted range, the color is used to set the temperature. This way, z2m uses the native command as much as possible, but still supports the full range that is available in Ikea's app.

Koenkk, 2025-02-13:
I still don't think it is a good idea, just because the fact the original hub widens the color temp range artificially by translating the color temp to x/y values doesn't mean Zigbee2MQTT should do the same. In fact, you can already achieve exactly this by setting the x/y color.

Mushoz, 2025-02-13:
You can indeed achieve this by setting the x/y color, which is certainly doable when changing the lights manually. However, some people might be using Adaptive Lightning in home assistant, which changes the color temperature based on the sun's color temperature. Because Adaptive Lightning sets a color temperature in Kelvins, it can only do the restricted range. That instantly makes the lights much less useful than they would be if I could use the entire range.

The fact is, is that the light is advertised with a 1780 to 6000 kelvin color temperature range. Even the manual states as such. This range is available when using the Ikea app. But this range is not available when using the bulb through z2m. I understand why, but for most end-users, all they know is that they are getting an inferior experience with z2m over Ikea's official app.

capjack92, 2025-02-13:
@Koenkk But for the gu10 it doesn't work the same way? Because they also support a reduced range outside the Ikea hub, with the hub instead it supports a wider range, literally the same thing what happens with the E27s. In fact, if I remember correctly @YannickvanVugt  came up with the solution by looking at the settings that the GU10s have on the Z2M to solve this "problem" on the E27s. This is exactly what happened in the responses between September 18th and 19th.

RubenKelevra, 2025-04-15:
Obviously not fixed

RubenKelevra, 2025-04-15:
> You can indeed achieve this by setting the x/y color, which is certainly doable when changing the lights manually. However, some people might be using Adaptive Lightning in home assistant, which [...]

Not entirely correct: Adaptive Lighting can also use x/y colors, if forced, but this will affect all lights in the group. 

So if you got RGB+CCT lights they will switch to RGB mode with atrocious color rendering, less energy efficiency and often a lot less brightness. 

If the light would *only* offer RGB via x/y colors, adaptive lighting would handle them correctly, however the fact that they offer also CCT is the issue here.

capjack92, 2025-04-15:
The solution is there, but it is not considered. With the solution of @YannickvanVugt There are no problems in using it, either alone or in a group, as I posted in the previous comments there are also photos that demonstratethe bulb is finally being used to its full potential. True color option enabled Is the way

RubenKelevra, 2025-04-15:
@Koenkk wrote:

> I still don't think it is a good idea, just because the fact the original hub widens the color temp range artificially by translating the color temp to x/y values doesn't mean Zigbee2MQTT should do the same. In fact, you can already achieve exactly this by setting the x/y color. Doing it for just this bulb would make it extremely inconsistent with other implementations.

How about offering a generic expansion of color ranges?

Meaning an additional setting you can enable in z2m for all RGB+CCT lights, with the minimum and maximum color temperature you like to expand it too?

This would allow users which like to expand *any* light bulbs color range, just to enable this for the light bulbs and z2m could show an expanded CCT color range to HA while converting the color range from the CCT value to X/Y colors.

That would be great as many RGB+CCT bulbs only offer 2400 K or 2700 K as minimum, while others do internal blending or have very warm white LEDs and offer down to 1800 K, which is great for very late night lighting.

While yes, there may be a "step" in the blending behavior if z2m does this, it at least allows users to set warm colors in groups without some of the lights being stuck at cooler colors.

Mushoz, 2025-04-15:
I have been using the workaround for a few months now with no downsides whatsoever. Ideally this becomes available for less technically inclined people as well, as right now the z2m experience is worse than what a end-user would get by using Ikea's default bridge + app.

closed 2025-06-23 by github-actions[bot]

RubenKelevra, 2025-06-28:
Sorry you have to endure such a fucking bot. No idea why @Koenkk activated it here.

Anyway, I've picked up one of those lamps today to investigate.

RubenKelevra, 2025-06-28:
For some reason I cannot pair my E2001 to it, so I can't tell if they would act via Ikea differently, but I doubt it.

What I found out so far is:

* That this is an RGB Bulb with what I think is just an additional 3000 K or 4000 K LED on top. The different white color tones are therefore just achieved by mixing the RGB and the one white LED.
* Changing between CCT and RGB mode makes no difference for this bulb, if I select nearly the same color tone in RGB I get the same brightness as in CCT mode.

Therefore, if Z2M would offer an extended range via MQTT and convert anything outside of the offered CCT range of the bulb to x/y colors, it would be not different to what the Ikea hub does and also not impact light quality, because there's no "pure RGB" mode with less color fidelity.

I still think this should be a general setting for all bulbs which offer RGB, where the user can just specify the CCT range he/she want to have and the colors not available via CCT mode will be converted to x/y colors within z2m for the user. 

I think this would offer a lot more usability, as groups of different lights will offer different lowest (or in this case even highest) CCT temperature, and thus create an uneven white tone in the room, if lowered further than the light source with the tightest boundary of offered colors.

Since many lights already do this internally (I got at least two which create warmer tones by adding Red into the warm white LED) this is not really an issue.