Koenkk/zigbee2mqtt #23344: [zh:ember] Warning when using Itead Dongle-E Zigbee NCP Firmware (7.4.3)

yoch, 2024-07-15:
### What happened?

### Issue Description
After upgrading Z2M and flashing my ZBDongle-E with [Itead's latest firmware](https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/NCP_7.4.3) (v7.4.3) to use the `ember` adapter, the following warning appears:

```
zh:ember: [EzspConfigId] Failed to SET "MULTICAST_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
```

### Investigation
Comparison of configuration parameters between Itead and Darkxst firmware versions:

| Parameter                  | Itead | Darkxst |
|----------------------------|-------|---------|
| CONFIG_PACKET_BUFFER_COUNT | 250   | 255     |
| CONFIG_ADDRESS_TABLE_SIZE  | 32    | 16      |
| CONFIG_MULTICAST_TABLE_SIZE| 8     | 16      |

### Questions
1. Why does Z2M rely on `CONFIG_ADDRESS_TABLE_SIZE=16`?
2. Is there a risk of system malfunction due to this parameter mismatch?
3. What are the reasons for these configuration differences between Itead and Darkxst firmware versions?

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

7.4.3 [GA]

### Adapter

 EmberZNet

### Setup

Plain install on RPi4

baspax1, 2024-07-15:
I have the exact same adapter and performed the exact same upgrade with the intention of improving the potential connection issues I had with my devices. However, after the upgrade and the switch to ember, things are worse, as shown in the logs here:

Zigbee2MQTT
version: 1.39.0
 
**Log:**
```Starting Zigbee2MQTT without watchdog.
[12:49:12] INFO: Preparing to start...
[12:49:12] INFO: Socat not enabled
[12:49:14] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[12:50:00] INFO: Preparing to start...
[12:50:01] INFO: Socat not enabled
[12:50:03] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[12:51:32] INFO: Preparing to start...
[12:51:33] INFO: Socat not enabled
[12:51:35] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
```
Configuration:
https://prnt.sc/nOC-7ELnDYL7

Here are the devices I have:
https://prnt.sc/GO97YqLaTuXu

I am facing many issues related to data updates; they are not stable at all. Even the sensor that is close to the Zigbee adapter does not receive data every 3 minutes as I want it to.

yoch, 2024-07-16:
#### NOTE
I saw [here](https://www.zigbee2mqtt.io/guide/adapters/emberznet.html) that I could set non-default values ​​with a `stack_config.json` file :

```json
{
  "ADDRESS_TABLE_SIZE": 32,
  "MULTICAST_TABLE_SIZE": 8
}
```

By doing so, the warning can be avoided, but I am still curious to know why these specific values ​​are applied by Z2M?
Maybe I better use the Darkxst firmware?

Nerivec, 2024-07-21:
A config failing to set means the adapter will use in-firmware value instead, as the message says. Z2M adapts to the effective value when necessary, so it should not be a problem. Different values means different "priorities" though, see below.

Most of these values (in-firmware ones) were picked a while back (NabuCasa, darkxst, etc.) to optimize the general use case for ZHA/Z2M based on the available adapters. It's all about optimizing memory allocation on the adapter to what's of better use for most users.
Since Silabs changed several areas regarding these values a while back, runtime modification was restricted a bit further. As such, a lot of these configs will no longer be explicitly set by Z2M (after next release), and instead in-firmware values will be used. Also new adapters (like MG24-based ones) can handle much larger values than previous.

_The new firmware building system allows easy customization of firmware per adapter (even easy enough for anyone to clone the repo, change the values, and trigger the GitHub workflow to get a firmware built for themselves if they want to try something else)._

I would say, better to use a darkxst firmware; haven't checked what values Itead put in all their new builds...

closed 2024-07-21 by Nerivec

ReX1983, 2025-01-13:
Old thread but I don't think my question deserves a new issue.

As of now (Jan 2025), is it recomanded to install the Itead 7.4.3 firmware or the the latest darkxst one (7.4.5 at the time of writing)?

I am on darkxst 7.4.3 and wonder where I should go.

ReX1983, 2025-01-13:
Shall I go for latest (which is 8.0.2 from darkxst) or official (which is the Itead version I found)?

Do you mean that now 8.0.x has bugs as 7.3.x and the future 8.1.x would correct these?

Nerivec, 2025-01-13:
darkxst moved more recent releases to `Releases` page instead of directly inside the repository (same for all other builders more or less):
https://github.com/darkxst/silabs-firmware-builder/releases
https://github.com/nabucasa/silabs-firmware-builder/releases
https://github.com/nerivec/silabs-firmware-builder/releases

_Note: Technically, we're at 8.1.0 already (released in December), but that one is definitely not stable yet, we'll have to wait for fixes from Silabs._

ReX1983, 2025-01-13:
You are the man, thanks for taking the time to help me.

Definitely trying 8.0.2 tonight :-)