Koenkk/zigbee2mqtt #23347: Busch-Jaeger 6735/6736/6737: Operating relay/dimmer with Zigbee can have long delays / sometimes several attempts needed

Ra72xx, 2024-07-16:
### What happened?

This is a follow-up of https://github.com/Koenkk/zigbee-herdsman-converters/issues/7733:
In this bug report, a configurable polling intervall was introduced in order to reduce the network load caused by the polling which is needed for these devices. 
This patch worked, however now there are very often (but not always) delays and also failures when operating the relay/dimmer (which is being polled) with Zigbee commands. 
This does not affect the button panel, obviously the topmost row of the panel always operates the relay/dimmer directly, and the additional rows send their commands without delay. We are only talking about wireless Zigbee operation here.

### What did you expect to happen?

The relay/dimmer reacts instantly to Zigbee commands, like the other devices in the network.

### How to reproduce it (minimal and precise)

Install any latest-dev version of Zigbee2MQTT after the patch mentioned above.

### Zigbee2MQTT version

 1.39.0-dev commit: bd78f56

### Adapter firmware version

7.4.2 [GA]

### Adapter

EmberZNet

### Setup

Podman container

### Debug log

This happens when turning from "off" state to "on" and "off" again, i.e. two mouseclicks in the Z2M ui (example with only slight delays):

`Info 2024-07-16 04:13:20z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts', payload '{"brightness":254,"brightness_relay":254,"device":{"applicationVersion":1,"dateCode":"20161209","friendlyName":"WZ: Dimmer-Bedienelement-Kombi Tür rechts","hardwareVersion":0,"ieeeAddr":"0xd85def11a10015b8","manufacturerID":4398,"manufacturerName":"Busch-Jaeger","model":"6735/6736/6737","networkAddress":140,"powerSource":"Mains (single phase)","softwareBuildID":"1.2.0","stackVersion":1,"type":"Router","zclVersion":1},"last_seen":"2024-07-16T02:13:20.929Z","linkquality":148,"state":"ON","state_relay":"OFF"}'
Info 2024-07-16 04:13:20z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts/relay', payload '{"brightness":254,"state":"OFF"}'
Info 2024-07-16 04:13:22z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts', payload '{"brightness":254,"brightness_relay":254,"device":{"applicationVersion":1,"dateCode":"20161209","friendlyName":"WZ: Dimmer-Bedienelement-Kombi Tür rechts","hardwareVersion":0,"ieeeAddr":"0xd85def11a10015b8","manufacturerID":4398,"manufacturerName":"Busch-Jaeger","model":"6735/6736/6737","networkAddress":140,"powerSource":"Mains (single phase)","softwareBuildID":"1.2.0","stackVersion":1,"type":"Router","zclVersion":1},"last_seen":"2024-07-16T02:13:22.135Z","linkquality":148,"state":"ON","state_relay":"OFF"}'
Info 2024-07-16 04:13:22z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts/relay', payload '{"brightness":254,"state":"OFF"}'
Info 2024-07-16 04:13:22z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts', payload '{"brightness":254,"brightness_relay":254,"device":{"applicationVersion":1,"dateCode":"20161209","friendlyName":"WZ: Dimmer-Bedienelement-Kombi Tür rechts","hardwareVersion":0,"ieeeAddr":"0xd85def11a10015b8","manufacturerID":4398,"manufacturerName":"Busch-Jaeger","model":"6735/6736/6737","networkAddress":140,"powerSource":"Mains (single phase)","softwareBuildID":"1.2.0","stackVersion":1,"type":"Router","zclVersion":1},"last_seen":"2024-07-16T02:13:22.469Z","linkquality":148,"state":"ON","state_relay":"OFF"}'
Info 2024-07-16 04:13:22z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts/relay', payload '{"brightness":254,"state":"OFF"}'
Info 2024-07-16 04:13:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts', payload '{"brightness":254,"brightness_relay":254,"device":{"applicationVersion":1,"dateCode":"20161209","friendlyName":"WZ: Dimmer-Bedienelement-Kombi Tür rechts","hardwareVersion":0,"ieeeAddr":"0xd85def11a10015b8","manufacturerID":4398,"manufacturerName":"Busch-Jaeger","model":"6735/6736/6737","networkAddress":140,"powerSource":"Mains (single phase)","softwareBuildID":"1.2.0","stackVersion":1,"type":"Router","zclVersion":1},"last_seen":"2024-07-16T02:13:23.552Z","linkquality":148,"state":"ON","state_relay":"OFF"}'
Info 2024-07-16 04:13:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts/relay', payload '{"brightness":254,"state":"OFF"}'
Info 2024-07-16 04:13:30z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts', payload '{"brightness":254,"brightness_relay":254,"device":{"applicationVersion":1,"dateCode":"20161209","friendlyName":"WZ: Dimmer-Bedienelement-Kombi Tür rechts","hardwareVersion":0,"ieeeAddr":"0xd85def11a10015b8","manufacturerID":4398,"manufacturerName":"Busch-Jaeger","model":"6735/6736/6737","networkAddress":140,"powerSource":"Mains (single phase)","softwareBuildID":"1.2.0","stackVersion":1,"type":"Router","zclVersion":1},"last_seen":"2024-07-16T02:13:30.732Z","linkquality":148,"state":"ON","state_relay":"OFF"}'
Info 2024-07-16 04:13:30z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts/relay', payload '{"brightness":254,"state":"OFF"}'
Info 2024-07-16 04:13:30z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts', payload '{"brightness":254,"brightness_relay":254,"device":{"applicationVersion":1,"dateCode":"20161209","friendlyName":"WZ: Dimmer-Bedienelement-Kombi Tür rechts","hardwareVersion":0,"ieeeAddr":"0xd85def11a10015b8","manufacturerID":4398,"manufacturerName":"Busch-Jaeger","model":"6735/6736/6737","networkAddress":140,"powerSource":"Mains (single phase)","softwareBuildID":"1.2.0","stackVersion":1,"type":"Router","zclVersion":1},"last_seen":"2024-07-16T02:13:30.732Z","linkquality":148,"state":"ON","state_relay":"ON"}'
Info 2024-07-16 04:13:30z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts/relay', payload '{"brightness":254,"state":"ON"}'
Info 2024-07-16 04:13:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts', payload '{"brightness":254,"brightness_relay":254,"device":{"applicationVersion":1,"dateCode":"20161209","friendlyName":"WZ: Dimmer-Bedienelement-Kombi Tür rechts","hardwareVersion":0,"ieeeAddr":"0xd85def11a10015b8","manufacturerID":4398,"manufacturerName":"Busch-Jaeger","model":"6735/6736/6737","networkAddress":140,"powerSource":"Mains (single phase)","softwareBuildID":"1.2.0","stackVersion":1,"type":"Router","zclVersion":1},"last_seen":"2024-07-16T02:13:31.882Z","linkquality":148,"state":"ON","state_relay":"ON"}'
Info 2024-07-16 04:13:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts/relay', payload '{"brightness":254,"state":"ON"}'
Info 2024-07-16 04:13:34z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts', payload '{"brightness":254,"brightness_relay":254,"device":{"applicationVersion":1,"dateCode":"20161209","friendlyName":"WZ: Dimmer-Bedienelement-Kombi Tür rechts","hardwareVersion":0,"ieeeAddr":"0xd85def11a10015b8","manufacturerID":4398,"manufacturerName":"Busch-Jaeger","model":"6735/6736/6737","networkAddress":140,"powerSource":"Mains (single phase)","softwareBuildID":"1.2.0","stackVersion":1,"type":"Router","zclVersion":1},"last_seen":"2024-07-16T02:13:34.694Z","linkquality":148,"state":"ON","state_relay":"ON"}'
Info 2024-07-16 04:13:34z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts/relay', payload '{"brightness":254,"state":"ON"}'
Info 2024-07-16 04:13:34z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts', payload '{"brightness":254,"brightness_relay":254,"device":{"applicationVersion":1,"dateCode":"20161209","friendlyName":"WZ: Dimmer-Bedienelement-Kombi Tür rechts","hardwareVersion":0,"ieeeAddr":"0xd85def11a10015b8","manufacturerID":4398,"manufacturerName":"Busch-Jaeger","model":"6735/6736/6737","networkAddress":140,"powerSource":"Mains (single phase)","softwareBuildID":"1.2.0","stackVersion":1,"type":"Router","zclVersion":1},"last_seen":"2024-07-16T02:13:34.694Z","linkquality":148,"state":"ON","state_relay":"OFF"}'
Info 2024-07-16 04:13:34z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/WZ: Dimmer-Bedienelement-Kombi Tür rechts/relay', payload '{"brightness":254,"state":"OFF"}'
[... cut ...] }, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":18,"options":4160,"groupId":0,"sequence":37})'`

I'll try to get a debug log of this situation also, this happened only after I disabled debug logging for the above log :-(.

Ra72xx, 2024-12-20:
Issue still present with current software versions.

l-mb, 2025-01-18:
I hadn't noticed this using the legacy `switch.xxxx`, but after upgrading to 2.x, I am stuck with only the `light.xxxx_relay` endpoint and that both incorrectly insists on being dimmable as well as having a very long delay when used with on/off actions.

I don't understand why the `switch.___` was removed. Can https://github.com/Koenkk/zigbee-herdsman-converters/commit/088f6e78c1550cf89d5cd85f5ef8401a165bf0e0 be reverted? It says "wrong", but it did have the advantage of working :-)

Ra72xx, 2025-01-19:
I never used the legacy switches, so I didn't notice that these were without delay. But as also in my case the problem was introduced by a Z2M software change, I assume that it could solved by software. And yes, the pure relais version is incorrectly treated as dimmable, that's true.
 
However, for me the delay issue was fixed when switching to more powerful coordinator hardware (from SkyConnect to SLZV-07MG24, see [this post](https://github.com/Koenkk/zigbee2mqtt/discussions/25804)).

klada, 2025-03-09:
> Issue still present with current software versions.

@Ra72xx What you are describing here sounds like an issue with your ZigBee network or ZigBee adapter. Did you plan your channel layout carefully beforehand? I am talking about WiFi and ZigBee channels and coexistence. Did you perform an [energy scan](https://github.com/zigpy/zigpy-cli?tab=readme-ov-file#performing-an-energy-scan) prior to choosing a ZigBee channel?

How many devices do you have in your ZigBee network? According to the official specs they support networks with up to 20 devices. From my personal experience the devices also work great in networks with more than 20 devices, but are starting to have binding issues in networks with 50+ devices.

klada, 2025-03-09:
> I hadn't noticed this using the legacy `switch.xxxx`, but after upgrading to 2.x, I am stuck with only the `light.xxxx_relay` endpoint and that both incorrectly insists on being dimmable as well as having a very long delay when used with on/off actions.

@l-mb I think your problem is unrelated to this issue. I don't fully understand the delays you are talking about. Could you please explain under which conditions they occur? There are three possible ways of controlling these devices, which one are you referring to?

1. Send on/off through MQTT / Home Assistant
2. Press top-row buttons on the device
3. Control through ZigBee group binding

In my environment feedback from 1 and 3 is instant. The delay from 2 depends on the poll interval you have set. It used to be hard-coded to 5 seconds, in Z2M 1.39.1 it changed to (user-configurable) 60 seconds. My guess is that you are referring to that polling interval which you may now change through the per-device `state_poll_interval` setting.

You can lower this again if you wish, but I would not recommend values lower than 10 seconds.

You are correct that the light is always exposed as a dimmable light, even when the control panel is installed on top of a relay instead of a dimmer. The reason for this is that (AFAIK) it's impossible to detect whether the light is dimmable or not. When installed on a relay these BJ devices simply advertise the same capabilities as when installed on a dimmer. Same endpoints, same clusters, same model number...

However always exposing the brightness does not hurt. That's the same thing that the Hue Bridge does when these devices are used in Hue environments. When using Home Assistant you could even hide the brightness if it really bothers you through the zigbee2mqtt configuration file like this:

```yaml
'0xd85macaddress':
  filtered_attributes:
    - brightness
  homeassistant:
    light_relay:
      brightness: false
  state_poll_interval: 10
```

In fact I think it's a good idea to add this to the official docs. I'll prepare a PR.

> I don't understand why the `switch.___` was removed. Can [Koenkk/zigbee-herdsman-converters@088f6e7](https://github.com/Koenkk/zigbee-herdsman-converters/commit/088f6e78c1550cf89d5cd85f5ef8401a165bf0e0) be reverted? It says "wrong", but it did have the advantage of working :-)

The devices used to be exposed both as a switch and a (dimmable) light. This was confusing in many ways:

- Two Home Assistant entities for the same thing, although the switch did not support dimming (obviously)
- The switch would never update it's state when the state was changed through a ZigBee binding. It was therefore broken for people who rely on direct control.

That's why the official device docs had a note that the switch should be ignored. It had only been there for compatibility reasons anyways. With Z2M 2.x a lot of things have changed (including payloads for legacy users) so this was a good chance to get rid of it.

Ra72xx, 2025-03-09:
>     1. Send on/off through MQTT / Home Assistant
> 
>     2. Press top-row buttons on the device
> 
>     3. Control through ZigBee group binding

For me:
1. has delays of a few seconds (from almost instantly, say, 2 secs, to over 10 secs)
2. has no delays, as obviously the relay is operated directly, without Zigbee. However, it might take some time until Homeassistant notices this, according to the configured polling intervall
3. I don't do that.

l-mb, 2025-04-23:
Sorry for the slow response from my side.

After the various troubles with the device and me believing that it was due to the z2m upgrade (because, for some reason, they started at the same time ...), it's finally "fully" died on me (as in, not switching anything, even w/o Zigbee in the mix).

I've since replaced it with a different model/brand, so from my side, I'll be blaming this on an aging/failed piece of hardware.

Apologies for what turned out to be unrelated.

closed 2025-07-01 by github-actions[bot]