Koenkk/zigbee2mqtt #23355: Z2M cannot start with SLZB-06M

maciekdnd, 2024-07-17:
### What happened?

I'm unable to start Z2M. I see only errors below:

[2024-07-17 16:25:33] error:    zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-07-17 16:25:33] error:    zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-17 16:25:33] error:    zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-07-17 16:25:33] error:    zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[2024-07-17 16:25:36] error:    zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-07-17 16:25:36] error:    zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-17 16:25:36] error:    zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-07-17 16:25:36] error:    zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[2024-07-17 16:25:54] error:    zh:ember: !!! NCP FATAL ERROR reason=ERROR_SERIAL_INIT. ATTEMPTING RESET... !!!
[2024-07-17 16:25:55] error:    zh:ember: Failed to reset and init NCP. Error: Stack type 0 is not expected!
[2024-07-17 16:25:55] error:    z2m: Adapter disconnected, stopping

Strangely enough, Z2M was running perfectly fine for few weeks. Right now I can say it's random. One time it can start without any problems, other time it fails like this.
I also tried to update to latest firmware (7.4.3) but I got same response.
I have two SLZB-06M in two separate networks (two LXC containers), both behave same.

Part of my config:
mqtt:
  base_topic: z2m_1
  server: mqtt://192.168.3.100:1883
  user: mqtt_ha
  password: mqttha12345
  version: 5
  keepalive: 60
  reject_unauthorized: true
serial:
  port: tcp://192.168.3.11:6638
  baudrate: 115200
  rtscts: false
  adapter: ember

I tried to reset controllers by reset by web console, disconnecting them from POE, rebooting containers. Nothing works. During start both works (I see web ui for controllers).

What should I do to get it working?

### What did you expect to happen?

Clean Z2M start without errors

### How to reproduce it (minimal and precise)

run Z2M: cd /opt/zigbee2mqtt && npm start

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20240510

### Adapter

SLZB-06M

### Setup

Z2M in Proxmox LXC container

gravcyber, 2024-07-17:
Experienced the same problem right now, thanks to @maciekdnd discovery i could fix it instantly

maciekdnd, 2024-07-19:
For some reason I'm unable to start Z2M today. Even when I rm coordinator_backup.yaml. Right now I have no idea what is going on. One time I have clean, perfect start and other I can't start even once. Can someone help?

maciekdnd, 2024-07-20:
Another day, another finding. I can restart Z2M in LXC container using `systemctl restart zigbee2mqtt.service`. Looks like clean start and hope for near feature.

andyblac, 2024-07-27:
I found that if you restart reboot HA, you also need to reboot your SLZB-06M as well.

maciekdnd, 2024-07-27:
My routine after restarting or powering down for now is:
```
rm opt/zigbee2mqtt/data/coordinator_backup.json
systemctl restart zigbee2mqtt.service
```

When I configure watchdog, like this:
```
WatchdogSec=10s
Restart=always
RestartSec=10s
```
I have to delete coordinator_backup.json every time anyway. With the file present, Z2M is unable to boot in my configuration, in all 3 networks.

itinfosec, 2024-07-30:
I have been having a similar issue with setting up Z2MQTT with an SLZB-06M and have been getting the following errors which haven't been overly useful in figuring out what is wrong.

> [2024-07-30 05:15:00] error: 	zh:ember:uart:ash: Received frame with CRC error
[2024-07-30 05:15:00] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-07-30 05:15:00] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-30 05:15:00] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-07-30 05:15:00] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-07-30 05:15:00] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-07-30 05:15:01] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-07-30 05:15:01] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-07-30 05:15:01] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-07-30 05:15:02] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-07-30 05:15:02] error: 	zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-07-30 05:15:02] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-30 05:15:02] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-07-30 05:15:02] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!

yeyeoke, 2024-07-30:
Same issue here, running in docker. When restarting the Zigbee2mqtt container, It fails to start. It's solved by restarting the container once again, then everything comes back up without issue.

andyblac, 2024-07-30:
I get similar on restarting HA OS

```
[13:41:48] INFO: Preparing to start...
[13:41:48] INFO: Socat not enabled
[13:41:49] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-07-30 13:41:50] error: 	zh:ember:uart:ash: Received frame with CRC error
[2024-07-30 13:41:50] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-07-30 13:41:50] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-30 13:41:50] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-07-30 13:41:53] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-07-30 13:41:53] error: 	zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
[2024-07-30 13:41:53] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | NCP status: ASH_NCP_FATAL_ERROR
[2024-07-30 13:41:53] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-07-30 13:41:53] error: 	zh:ember: !!! NCP FATAL ERROR reason=HOST_FATAL_ERROR. ATTEMPTING RESET... !!!
[2024-07-30 13:42:50] error: 	zh:ember: !!! NCP FATAL ERROR reason=ERROR_SERIAL_INIT. ATTEMPTING RESET... !!!
[2024-07-30 13:42:54] error: 	z2m: Not connected to MQTT server!
[2024-07-30 13:42:54] error: 	z2m: Cannot send message: topic: 'zigbee2mqtt_slzb-06m/bridge/state', payload: '{"state":"offline"}
```

once I reboot the SLZB-06M Z2M starts normally.
```
[13:42:55] INFO: Preparing to start...
[13:42:55] INFO: Socat not enabled
[13:42:55] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
```

pepe59, 2024-08-02:
After updating Z2M 1.39.1-1, it can no longer run.
I reverted to Z2M 1.39.0-1 and it works as expected.
I am using Z2M addon in HA.
Does anyone have the same problem, or how to make it work on the new version of Z2M 1.39.1-1?

MilesTEG1, 2024-08-02:
> After updating Z2M 1.39.1-1, it can no longer run. I reverted to Z2M 1.39.0-1 and it works as expected. I am using Z2M addon in HA. SLZB-06M coordinator Does anyone have the same problem, or how to make it work on the new version of Z2M 1.39.1-1?

For me, no problem with the 1.39.1-1 Z2M version.
I don't encounter the addon can't start after boot. But I'll have to verify this.

GordonSmith-db, 2024-08-07:
I have a SLZB-06 (not M) and just got this today:
[Notes.txt](https://github.com/user-attachments/files/16520139/Notes.txt)

Any ideas?

GordonSmith-db, 2024-08-07:
I am using a SLZB-06 (not M) and just got this today. Any ideas? I noticed this an hour later and found Z2M wasn't running. When I clicked "start", it started just fine.

```
[2024-08-06 20:11:09] info: 	zh:zstack:znp: Socket error
[2024-08-06 20:11:09] info: 	zh:zstack:znp: Port closed
[2024-08-06 20:11:09] info: 	zh:zstack:znp: closing
[2024-08-06 20:11:09] error: 	z2m: Adapter disconnected, stopping
[2024-08-06 20:11:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-08-06 20:11:09] info: 	z2m: Disconnecting from MQTT server
[2024-08-06 20:11:09] info: 	z2m: Stopping zigbee-herdsman...
[2024-08-06 20:11:09] info: 	z2m: Stopped zigbee-herdsman
[2024-08-06 20:11:09] info: 	z2m: Stopped Zigbee2MQTT
[20:11:11] INFO: Preparing to start...
[20:11:12] INFO: Socat not enabled
[20:11:18] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-08-06 20:11:28] info: 	z2m: Logging to console, file (filename: log.log)
[2024-08-06 20:11:28] info: 	z2m: Starting Zigbee2MQTT version 1.39.1 (commit #unknown)
[2024-08-06 20:11:28] info: 	z2m: Starting zigbee-herdsman (0.55.3)
[2024-08-06 20:11:28] info: 	zh:zstack:znp: Opening TCP socket with 192.168.68.83:6638
[2024-08-06 20:11:28] info: 	zh:zstack:znp: Socket error
[2024-08-06 20:11:28] error: 	z2m: Error while starting zigbee-herdsman
[2024-08-06 20:11:28] error: 	z2m: Failed to start zigbee
[2024-08-06 20:11:28] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-08-06 20:11:28] error: 	z2m: Exiting...
[2024-08-06 20:11:29] error: 	z2m: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:185:24)
    at Socket.emit (node:events:529:35)
    at emitErrorNT (node:internal/streams/destroy:151:8)
    at emitErrorCloseNT (node:internal/streams/destroy:116:3)
    at processTicksAndRejections (node:internal/process/task_queues:82:21)

/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:189:25)
    at Logger.info (/app/lib/util/logger.ts:202:14)
    at Znp.onPortClose (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:107:16)
    at Object.onceWrapper (node:events:632:26)
    at Socket.emit (node:events:517:28)
    at TCP.<anonymous> (node:net:350:12)
[21:32:40] INFO: Preparing to start...
[21:32:41] INFO: Socat not enabled
[21:32:47] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-08-06 21:32:57] info: 	z2m: Logging to console, file (filename: log.log)
[2024-08-06 21:32:57] info: 	z2m: Starting Zigbee2MQTT version 1.39.1 (commit #unknown)
[2024-08-06 21:32:57] info: 	z2m: Starting zigbee-herdsman (0.55.3)
[2024-08-06 21:32:58] info: 	zh:zstack:znp: Opening TCP socket with 192.168.68.83:6638
[2024-08-06 21:32:58] info: 	zh:zstack:znp: Socket connected
```

jgdub, 2024-08-08:
> I am using a SLZB-06 (not M) and just got this today on Z2M 1.39.1-1. Any ideas? I noticed this an hour later and found Z2M wasn't running. When I clicked "start", it started just fine.
> 
> ```
> [2024-08-06 20:11:09] info: 	zh:zstack:znp: Socket error
> [2024-08-06 20:11:09] info: 	zh:zstack:znp: Port closed
> [2024-08-06 20:11:09] info: 	zh:zstack:znp: closing
> [2024-08-06 20:11:09] error: 	z2m: Adapter disconnected, stopping
> [2024-08-06 20:11:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
> [2024-08-06 20:11:09] info: 	z2m: Disconnecting from MQTT server
> [2024-08-06 20:11:09] info: 	z2m: Stopping zigbee-herdsman...
> [2024-08-06 20:11:09] info: 	z2m: Stopped zigbee-herdsman
> [2024-08-06 20:11:09] info: 	z2m: Stopped Zigbee2MQTT
> [20:11:11] INFO: Preparing to start...
> [20:11:12] INFO: Socat not enabled
> [20:11:18] INFO: Starting Zigbee2MQTT...
> Starting Zigbee2MQTT without watchdog.
> [2024-08-06 20:11:28] info: 	z2m: Logging to console, file (filename: log.log)
> [2024-08-06 20:11:28] info: 	z2m: Starting Zigbee2MQTT version 1.39.1 (commit #unknown)
> [2024-08-06 20:11:28] info: 	z2m: Starting zigbee-herdsman (0.55.3)
> [2024-08-06 20:11:28] info: 	zh:zstack:znp: Opening TCP socket with 192.168.68.83:6638
> [2024-08-06 20:11:28] info: 	zh:zstack:znp: Socket error
> [2024-08-06 20:11:28] error: 	z2m: Error while starting zigbee-herdsman
> [2024-08-06 20:11:28] error: 	z2m: Failed to start zigbee
> [2024-08-06 20:11:28] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
> [2024-08-06 20:11:28] error: 	z2m: Exiting...
> [2024-08-06 20:11:29] error: 	z2m: Error: Error while opening socket
>     at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:185:24)
>     at Socket.emit (node:events:529:35)
>     at emitErrorNT (node:internal/streams/destroy:151:8)
>     at emitErrorCloseNT (node:internal/streams/destroy:116:3)
>     at processTicksAndRejections (node:internal/process/task_queues:82:21)
> 
> /app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264
>   var er = new ERR_STREAM_WRITE_AFTER_END();
>            ^
> Error: write after end
>     at writeAfterEnd (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264:12)
>     at DerivedLogger.Writable.write (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:300:21)
>     at DerivedLogger.log (/app/node_modules/winston/lib/winston/logger.js:231:12)
>     at Logger.log (/app/lib/util/logger.ts:189:25)
>     at Logger.info (/app/lib/util/logger.ts:202:14)
>     at Znp.onPortClose (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:107:16)
>     at Object.onceWrapper (node:events:632:26)
>     at Socket.emit (node:events:517:28)
>     at TCP.<anonymous> (node:net:350:12)
> [21:32:40] INFO: Preparing to start...
> [21:32:41] INFO: Socat not enabled
> [21:32:47] INFO: Starting Zigbee2MQTT...
> Starting Zigbee2MQTT without watchdog.
> [2024-08-06 21:32:57] info: 	z2m: Logging to console, file (filename: log.log)
> [2024-08-06 21:32:57] info: 	z2m: Starting Zigbee2MQTT version 1.39.1 (commit #unknown)
> [2024-08-06 21:32:57] info: 	z2m: Starting zigbee-herdsman (0.55.3)
> [2024-08-06 21:32:58] info: 	zh:zstack:znp: Opening TCP socket with 192.168.68.83:6638
> [2024-08-06 21:32:58] info: 	zh:zstack:znp: Socket connected
> ```

Are you using ember?

andyblac, 2024-08-09:
> Are you using ember?

make sure you using the latest firmware.

Koenkk, 2024-08-24:
In case you are using ember, make sure to configure the `adapter` properly in the z2m `configuration.yaml`, it should have:

```yaml
serial:
    adapter: ember
```

(I see `z-stack` errors in your log so it's not using the ember adapter)

andyblac, 2024-08-24:
> In case you are using ember, make sure to configure the `adapter` properly in the z2m `configuration.yaml`, it should have:
> 
> ```yaml
> serial:
>     adapter: ember
> ```
> 
> (I see `z-stack` errors in your log so it's not using the ember adapter)

yup, i'm using ember. it's still has stack in error logs.

```yaml
serial:
  port: tcp://slzb-06m.local:6638
  baudrate: 115200
  adapter: ember
  rtscts: false
```

prmfeddema, 2024-09-04:
Hi have the same here. Whatever i do it keeps dropping off after starting it. Any help would be appreciated.

Z2M version: 1.40.0-1
SLZB06M firmware versions:
Core: 2.5.2
Zigbee coordinator: 20240510

Settings
Socat:
enabled: false
master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
options: "-d -d"
log: false

mqtt:
server: mqtt://10.0.0.87:1883
user: <>
password: <>
base_topic: zigbee2mqtt1
channel: 4
permit join: true

serial:
port: tcp://10.0.0.37:6638
baudrate: 115200
adapter: ember
disable_led: false
tscts: false

Log file:
[12:57:44] INFO: Preparing to start...
[12:57:45] INFO: Socat not enabled
[12:57:48] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-09-04 12:57:54] info: 	z2m: Logging to console, file (filename: log.log)
[2024-09-04 12:57:54] info: 	z2m: Starting Zigbee2MQTT version 1.40.0 (commit #unknown)
[2024-09-04 12:57:54] info: 	z2m: Starting zigbee-herdsman (0.57.1)
[2024-09-04 12:57:54] info: 	zh:ember: Using default stack config.
[2024-09-04 12:57:54] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-09-04 12:57:54] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-09-04 12:57:54] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-09-04 12:57:54] info: 	zh:ember:uart:ash: Socket ready
[2024-09-04 12:57:54] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-09-04 12:57:56] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-09-04 12:57:56] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-09-04 12:57:56] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-09-04 12:57:56] info: 	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2024-09-04 12:57:56] info: 	zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.1 [GA]","build":0,"major":7,"minor":4,"patch":1,"special":0,"type":170}
[2024-09-04 12:57:56] info: 	zh:ember: [STACK STATUS] Network up.
[2024-09-04 12:57:56] info: 	zh:ember: [INIT TC] Adapter network matches config.
[2024-09-04 12:57:56] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1247ms until next broadcast.
[2024-09-04 12:57:56] info: 	z2m: zigbee-herdsman started (resumed)
[2024-09-04 12:57:56] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":1,"revision":"7.4.1 [GA]","special":0,"type":170},"type":"EmberZNet"}'
[2024-09-04 12:57:56] info: 	z2m: 0x84fd27fffece9881 (0x84fd27fffece9881): TS011F_plug_1 - Tuya Smart plug (with power monitoring) (Router)
[2024-09-04 12:57:56] info: 	z2m: Currently 1 devices are joined.
[2024-09-04 12:57:56] info: 	z2m: Zigbee: disabling joining new devices.
[2024-09-04 12:57:57] info: 	z2m: Connecting to MQTT server at mqtt://10.0.0.87:1883
[2024-09-04 12:57:57] info: 	z2m: Connected to MQTT server
[2024-09-04 12:57:57] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt1/bridge/state', payload '{"state":"online"}'
[2024-09-04 12:57:57] info: 	z2m: Started frontend on port 8099
[2024-09-04 12:57:57] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt1/bridge/state', payload '{"state":"online"}'
[2024-09-04 12:57:57] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt1/0x84fd27fffece9881', payload '{"ac_frequency":null,"child_lock":null,"countdown":null,"current":0,"energy":100.57,"indicator_mode":null,"linkquality":255,"power":0,"power_apparent":null,"power_outage_memory":null,"state":"ON","update":{"installed_version":67,"latest_version":67,"state":"idle"},"update_available":null,"voltage":234}'
[2024-09-04 12:57:57] info: 	z2m: Zigbee2MQTT started!
[2024-09-04 12:58:02] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/122105103981011015010911311611649_0xe0798dfffe8f8f6e/connection_state/config', payload '{"device":{"hw_version":"EmberZNet 7.4.1 [GA]","identifiers":["zigbee2mqtt_bridge_0xe0798dfffe8f8f6e"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.40.0"},"device_class":"connectivity","entity_category":"diagnostic","name":"Connection state","object_id":"zigbee2mqtt_bridge_connection_state","origin":{"name":"Zigbee2MQTT","sw":"1.40.0","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt1/bridge/state","unique_id":"bridge_0xe0798dfffe8f8f6e_connection_state_zigbee2mqtt1","value_template":"{{ value_json.state }}"}'
[2024-09-04 12:58:15] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2024-09-04 12:58:15] error: 	zh:ember:uart:ash: Received unexpected reset from adapter, with reason=RESET_SOFTWARE.
[2024-09-04 12:58:15] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | Adapter status: ASH_NCP_FATAL_ERROR
[2024-09-04 12:58:15] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2024-09-04 12:58:15] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[... 4 lines cut ...]
[2024-09-04 12:58:15] info: 	zh:ember:uart:ash:   DATA bytes  : RX=403, TX=412
[2024-09-04 12:58:15] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2024-09-04 12:58:15] info: 	zh:ember:uart:ash:   ACK frames  : RX=3, TX=44
[2024-09-04 12:58:15] info: 	zh:ember:uart:ash:   NAK frames  : RX=1, TX=0
[2024-09-04 12:58:15] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-09-04 12:58:15] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[... cut ...]
[2024-09-04 12:58:15] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-09-04 12:58:15] error: 	z2m: Adapter disconnected, stopping
[2024-09-04 12:58:15] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt1/bridge/state', payload '{"state":"offline"}'
[2024-09-04 12:58:15] info: 	z2m: Disconnecting from MQTT server
[2024-09-04 12:58:15] info: 	z2m: Stopping zigbee-herdsman...
[2024-09-04 12:58:15] info: 	z2m: Stopped zigbee-herdsman
[2024-09-04 12:58:15] info: 	z2m: Stopped Zigbee2MQTT

prmfeddema, 2024-09-04:
After deleting the coordinator_backup.json i now have the following error / log file:

[13:16:15] INFO: Preparing to start...
[13:16:16] INFO: Socat not enabled
[13:16:18] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-09-04 13:16:25] info: 	z2m: Logging to console, file (filename: log.log)
[2024-09-04 13:16:25] info: 	z2m: Starting Zigbee2MQTT version 1.40.0 (commit #unknown)
[2024-09-04 13:16:25] info: 	z2m: Starting zigbee-herdsman (0.57.1)
[2024-09-04 13:16:25] info: 	zh:ember: Using default stack config.
[2024-09-04 13:16:25] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-09-04 13:16:25] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-09-04 13:16:25] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-09-04 13:16:25] info: 	zh:ember:uart:ash: Socket ready
[2024-09-04 13:16:25] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-09-04 13:16:26] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-09-04 13:16:26] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-09-04 13:16:26] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-09-04 13:16:26] info: 	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2024-09-04 13:16:26] info: 	zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.1 [GA]","build":0,"major":7,"minor":4,"patch":1,"special":0,"type":170}
[2024-09-04 13:16:27] info: 	zh:ember: [STACK STATUS] Network up.
[2024-09-04 13:16:27] info: 	zh:ember: [INIT TC] Adapter network matches config.
[2024-09-04 13:16:27] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1247ms until next broadcast.
[2024-09-04 13:16:27] info: 	zh:controller: Wrote coordinator backup to '/config/zigbee2mqtt/coordinator_backup.json'
[2024-09-04 13:16:27] info: 	z2m: zigbee-herdsman started (resumed)
[2024-09-04 13:16:27] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":1,"revision":"7.4.1 [GA]","special":0,"type":170},"type":"EmberZNet"}'
[2024-09-04 13:16:27] info: 	z2m: 0x84fd27fffece9881 (0x84fd27fffece9881): TS011F_plug_1 - Tuya Smart plug (with power monitoring) (Router)
[2024-09-04 13:16:27] info: 	z2m: Currently 1 devices are joined.
[2024-09-04 13:16:27] info: 	z2m: Zigbee: disabling joining new devices.
[2024-09-04 13:16:42] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=54:"SEND_BROADCAST" Seq=39 Len=7].
[2024-09-04 13:16:42] error: 	zh:ember: Adapter fatal error: ASH_ERROR_TIMEOUTS
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   Total frames: RX=51, TX=86
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   DATA frames : RX=49, TX=41
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   DATA bytes  : RX=442, TX=436
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=43
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   NAK frames  : RX=1, TX=1
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=6
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2024-09-04 13:16:42] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-09-04 13:16:42] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2024-09-04 13:16:42] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-09-04 13:16:42] error: 	z2m: Adapter disconnected, stopping
[2024-09-04 13:16:42] error: 	z2m: Failed to set permit join to false (ASH_ERROR_TIMEOUTS)
[2024-09-04 13:16:42] info: 	z2m: Connecting to MQTT server at mqtt://10.0.0.87:1883
[2024-09-04 13:16:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt1/bridge/state', payload '{"state":"offline"}'
[2024-09-04 13:16:42] info: 	z2m: Connected to MQTT server
[2024-09-04 13:16:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt1/bridge/state', payload '{"state":"online"}'
[2024-09-04 13:16:42] info: 	z2m: Disconnecting from MQTT server
[2024-09-04 13:16:42] info: 	z2m: Stopping zigbee-herdsman...
[2024-09-04 13:16:42] info: 	z2m: Stopped zigbee-herdsman
[2024-09-04 13:16:42] error: 	z2m: MQTT error: client disconnecting
[2024-09-04 13:16:42] info: 	z2m: Stopped Zigbee2MQTT

/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:189:25)
    at Logger.error (/app/lib/util/logger.ts:194:14)
    at Controller.start (/app/lib/controller.ts:190:20)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at start (/app/index.js:154:5)

prmfeddema, 2024-09-04:
Next step: hook it up to a usb port instead of ethernet to see if things improve

prmfeddema, 2024-09-05:
Confirming that the slzb-06m works fine when connected through usb. It does NOT when connected through ethernet (usb powered)

thecode, 2024-09-05:
> Confirming that the slzb-06m works fine when connected to usb. So this reboot/zigbee failure only seems to happen when connected through ethernet (no poe - usb powered for test purposes)

What core firmware you have on the device? I had issues with core v2.3.6 which I think are fixed with v.2.5.2, "I think" because it is not enough time since I updated and sometimes I had few disconnections a day and sometimes days without a problem (average was about 1 disconnection a day)

prmfeddema, 2024-09-05:
Hi,

I have 2.5.2 installed. In my case it disconnects immediately.

Kind regards,

Pascal

prmfeddema, 2024-09-05:
Log files from both HA as well as the SLZB-06m

[HomeAssistant_log_file.zip](https://github.com/user-attachments/files/16887744/HomeAssistant_log_file.zip)
[SLZB-06M Log file.zip](https://github.com/user-attachments/files/16887745/SLZB-06M.Log.file.zip)

prmfeddema, 2024-09-07:
Further testing shows that this only happens whab using the latest zigbee coordinator firmware (with ember). Whe. Reverting back to version 200240330 and setting ezsp things seem to work fine.

prmfeddema, 2024-09-07:
> What core firmware you have on the device? I had issues with core v2.3.6 which I think are fixed with v.2.5.2, "I think" because it is not enough time since I updated and sometimes I had few [...]

I have 2.5.2 and reverted back 1 version to test. This made no difference. Went back to 2.5.2 and downgraded the zigbee coordinator version to 20240330 and ezsp. This solved the problem.

andyblac, 2024-09-07:
I've not had this issue since updating to version `2.5.2`

prmfeddema, 2024-09-07:
I did some further testing - the issue is in the zigbee coordinator. When i downgraded that one to 20240330 and ezsp (instead of ember) the problem seems to be gone (what i can see from my limited testing)Verstuurd vanaf mijn iPhone

MilesTEG1, 2024-09-07:
Hello,
Since some recents updates of Z2M, and since I use [this firmware (https://github.com/Nerivec/silabs-firmware-builder/releases/download/v2024.6.1/smlight_slzb06m_ncp-uart-sw_115200_8.0.1.0.gbl)](https://github.com/Nerivec/silabs-firmware-builder/releases/download/v2024.6.1/smlight_slzb06m_ncp-uart-sw_115200_8.0.1.0.gbl) for my SLSB-06M I don't see anymore the errors I had and Z2M starts well.

prmfeddema, 2024-09-07:
Unfortunately i still have the same problem with this firmware 😥

prmfeddema, 2024-09-08:
I’ve done more digging - seems there is a faulty network cable at play here.. investigatingVerstuurd vanaf mijn iPhone

prmfeddema, 2024-09-08:
Ok - further digging being done and removed and identified an issue with a
switch in the process (switch replaced)  - now this seems to be caused by
the network config. As soon as i move the SLZB06m on the same switch as the
home assistant server it works fine. As soon as 2 more switches are in the
home server instance and the SLZB-06M  between the zigbee stick starts to
drop.

Therefore: not sure if this is a firmware issue but maybe a timing issue?
...strange...

ka0r5, 2024-09-16:
Hi, I also encountered the error like zh:ember:uart:ash: Received unexpected reset from NCP, with reason=RESET_SOFTWARE.
For me it was due to the conflict with the auto start service of my z2m instance which was looping

prmfeddema, 2024-09-17:
Thanks - over here things seem to running ok after i moved it to it's destined location (Shed in the garden, POE powered) and no switches in between. Will keep an eye on it.

huesofredy, 2024-11-14:
I have this error, how can I solve it?
2024-11-14 22:39:13] info: 	zh:ember: [STACK STATUS] Network down.
[2024-11-14 22:39:14] error: 	z2m: Error while starting zigbee-herdsman
[2024-11-14 22:39:14] error: 	z2m: Failed to start zigbee
[2024-11-14 22:39:14] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-11-14 22:39:14] error: 	z2m: Exiting...
[2024-11-14 22:39:14] error: 	z2m: Error: [BACKUP] Current backup file is not for EmberZNet stack.
    at EmberAdapter.getStoredBackup (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1161:23)
    at EmberAdapter.initTrustCenter (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:937:29)
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:726:18)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1598:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)

closed 2025-01-21 by github-actions[bot]

devkaiwang, 2025-04-01:
> with SLZB-06M I have this error, how can I solve it? 2024-11-14 22:39:13] info: zh:ember: [STACK STATUS] Network down. [2024-11-14 22:39:14] error: z2m: Error while starting zigbee-herdsman [2024-11-14 22:39:14] error: z2m: Failed to start zigbee [2024-11-14 22:39:14] error: z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions [2024-11-14 22:39:14] error: z2m: Exiting... [2024-11-14 22:39:14] error: z2m: Error: [BACKUP] Current backup file is not for EmberZNet stack. at EmberAdapter.getStoredBackup (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1161:23) at EmberAdapter.initTrustCenter (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:937:29) at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:726:18) at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1598:24) at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29) at Zigbee.start (/app/lib/zigbee.ts:69:27) at Controller.start (/app/lib/controller.ts:161:27) at start (/app/index.js:154:5)

Deleting the `coordinator_backup.json` solved the issue here.

tispokes, 2025-11-25:
Hi, writing this comment, to help other lost souls, because I had problems with the **migration from zstack to ember** with the error `z2m: Error: [BACKUP] Current backup file is not for EmberZNet stack.` even after changing the IEEE according to https://www.zigbee2mqtt.io/guide/faq/#how-do-i-migrate-from-one-adapter-to-another (was a bit pain under W11 with WSL [see usbipd]), too.

1. stop z2m
2. backup from zigbee2mqtt (or data) directory (this is to make a clean rollback if needed)
- configuration.yaml
- coordinator_backup.json
- database.db (warning: under VSCode it's usually hidden!)
- state.json
3. rename `coordinator_backup.json` to e.g. `coordinator_backup.save.json`
4. In the zigbee2mqtt/configuration.yaml
- change serial adapter to `adapter: ember` (maybe `port`, for me it stayed `port: /dev/ttyUSB0`
- maybe in `advanced` to `transmit_power: 20` 
5. start z2m to generate a new `coordinator_backup.json`
- now the error occurs
6. stop z2m
7. in `coordinator_backup.json` 
- copy everything from `coordinator_backup.save.json` **expect** the first 2 blocks (`metadata` and `stack_specific`) and save
For me I first tried to keep the new `coordinator_ieee` (although I changed and checked it according to the link above) and the `extended_pan_id`, which was a mess. 
8. start z2m again

For me this worked flawless after my initial fail

techie-dreamer, 2026-01-30:
> Hi, writing this comment, to help other lost souls, because I had problems with the **migration from zstack to ember** with the error `z2m: Error: [BACKUP] Current backup file is not for EmberZNet [...]

Thanks for documenting these steps.
I migrated from SLZB-06 (CC2652P chipset - due to massive stability issues with a large 'noisy' network) to SLZB-06M & was stuck at this step (getting z2m to start up again after stack setting change in configuration.yaml) - renaming coordinator_backup.json made it boot up again.

Happy Days! :)