Koenkk/zigbee2mqtt #23357: [Feature request]: Available update version details on OTA page

sverrevh, 2024-07-17:
### Is your feature request related to a problem? Please describe

Currently only the installed firmware version is shown on the OTA page. It would be very helpful (as https://github.com/Koenkk/zigbee2mqtt/issues/23325 has shown) to also know which version is available there for update. 

### Describe the solution you'd like

Create an extra column with the available software version or put the available software version behind the current software version.

### Describe alternatives you've considered

HomeAssistant shows the available software version, but sometimes it is hard to read as it displays it in decimal instead of hexadecimal for IKEA bulbs.

### Additional context

,.

mrmaximas, 2024-07-17:
It's frontend by @nurikk feature

Koenkk, 2024-07-18:
It's not possible to determine the `swBuildId` for an image without flashing it on the device. The versioning used in OTA is different than the `dateCode` and `swBuildId`.

sverrevh, 2024-07-18:
Ok but Zigbee2mqtt shows this kind of info:
{"installed_version":50331664,"latest_version":50331681,"state":"available"}

Decimal  50331681 is 3000021 in hexadecimal. The installed version shows 3.0.21 for this example.

Koenkk, 2024-07-20:
@sverrevh that may be true for this device, but I believe it doesn't apply to all.

closed 2024-12-28 by github-actions[bot]