Koenkk/zigbee2mqtt #23366: [New device support]: HOMMYN (ORVIBO) MS-21-Z  presence and Illuminance sensor

Psytoshgen, 2024-07-18:
### Link

https://hommyn.app/products/sensor-dvizheniya-i-sveta-hommyn-ms-21-z/

### Database entry

{"id":40,"type":"EndDevice","ieeeAddr":"0x00158d00010c5e65","nwkAddr":14899,"manufId":4151,"manufName":"ORVIBO","powerSource":"Battery","modelId":"84780a9533b64db29050e3d226e7ffbe\u00002017-04-06\u0000\u0000","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280],"outClusterList":[0,1280],"clusters":{"genBasic":{"attributes":{"modelId":"84780a9533b64db29050e3d226e7ffbe\u00002017-04-06\u0000\u0000","manufacturerName":"ORVIBO","powerSource":3,"zclVersion":1,"appVersion":6,"stackVersion":2,"hwVersion":1,"dateCode":"2017-04-06","swBuildId":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffec10e29","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffec10e29","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffec10e29","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffec10e29","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":262,"inClusterList":[0,3,1024],"outClusterList":[0,1024],"clusters":{"genBasic":{"attributes":{"swBuildId":""}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":20792}}},"binds":[{"cluster":1024,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffec10e29","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffec10e29","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":6,"stackVersion":2,"hwVersion":1,"dateCode":"2017-04-06","swBuildId":"","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1721307998153}

### Comments

here some info

{
    "ac_status": false,
    "alarm_1": true,
    "alarm_2": false,
    "battery": 100,
    "battery_defect": false,
    "battery_low": false,
    "illuminance_2": 22553,
    "illuminance_lux_2": 180,
    "linkquality": 255,
    "restore_reports": false,
    "supervision_reports": false,
    "tamper": false,
    "test": false,
    "trouble": false,
    "device": {
        "applicationVersion": 6,
        "dateCode": "2017-04-06",
        "friendlyName": "Движение вход",
        "hardwareVersion": 1,
        "ieeeAddr": "0x00158d00010c5e65",
        "manufacturerID": 4151,
        "manufacturerName": "ORVIBO",
        "model": "84780a9533b64db29050e3d226e7ffbe\u00002017-04-06\u0000\u0000",
        "networkAddress": 14899,
        "powerSource": "Battery",
        "softwareBuildID": "",
        "stackVersion": 2,
        "type": "EndDevice",
        "zclVersion": 1
    }
}

### External definition

```shell
const {deviceEndpoints, battery, identify, iasZoneAlarm, illuminance} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['84780a9533b64db29050e3d226e7ffbe 2017-04-06 '],
   model: '84780a9533b64db29050e3d226e7ffbe 2017-04-06 ',
    vendor: 'ORVIBO',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), battery(), identify(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]}), illuminance({"endpointNames":["2"]})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

it16122512, 2024-12-25:
Hi, did you manage to add it?

Psytoshgen, 2024-12-27:
> Hi, did you manage to add it?

nope, cant understand instructions to make converter. I used yandex smart home to to add it... only thing that it shows this sensor as vibration.

closed 2025-03-05 by github-actions[bot]