Koenkk/zigbee2mqtt #23367: unsupported device with Zigbee model 'TS0601' and manufacturer name ' _TZE204_dapwryy7'

actabi, 2024-07-18:
### Link

https://fr.aliexpress.com/item/1005007050686755.html?spm=a2g0o.order_list.order_list_main.11.2f1a5e5bGjqnNt&gatewayAdapt=glo2fra

### Database entry

?

### Comments

Warning 2024-07-18 17:17:41z2m: Device '0xa4c138cabf76350a' with Zigbee model 'TS0601' and manufacturer name '_TZE204_dapwryy7' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_dapwryy7',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

i-m-d, 2024-07-21:
I have the same device and problem:

[2024-07-21 17:50:15] warning: 	z2m: Device '0xa4c138e38b5f5848' with Zigbee model 'TS0601' and manufacturer name '_TZE204_dapwryy7' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support
-new-devices/01_support_new_devices.html

FR3JZ, 2024-07-22:
Same here device not supported

Koenkk, 2024-07-23:
To find out which definition works, you can try various definitions from [`tuya.ts`](https://github.com/Koenkk/zigbee-herdsman-converters/blob/775eb63dad43f06efd0563faf3e3b2af64dc3ac9/src/devices/tuya.ts#L6476) (search for `human presence`) by stopping z2m, change `_TZE204_dapwryy7` to e.g. `_TZE200_auin8mzr` and start z2m.

FR3JZ, 2024-07-23:
https://github.com/Koenkk/zigbee2mqtt/issues/23249

Solution is posted here

Koenkk, 2024-07-24:
Ah yes, `_TZE204_dapwryy7` will be supported out-of-the-box in the 1 August release.

closed 2024-07-24 by Koenkk