Koenkk/zigbee2mqtt #23368: Can't add devices to or delete a re-created group and can't add to new groups

YewTreeWeb, 2024-07-18:
### What happened?

When I either create or re-create a group after deleting I cannot add devices to that group since upgrading to 1.38.0. If I try and deleted a re-create group I also get an error. The error message for both problems is always the ID does not exist in the database. Some times you can get the device to still add if you click on one of the buttons for the scene name.

[image]

### What did you expect to happen?

Once a Zigbee group has been created/re-created the group should be able to add devices and be deleted.

### How to reproduce it (minimal and precise)

Create a new group and add devices or delete an existing group and re-create a new group with the same name and ID

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

20240710.0

### Adapter

Combee II

### Setup

Home Assistant OS on Proxmox

hdejongh, 2024-07-26:
i have the same problem here, also renaming doesnt work
[image]

uncendave, 2024-11-09:
Same problem here. I do seem to be able to rename the group. Once renamed, I still can't add devices but generally can delete the group!

Pammetje, 2024-11-15:
same issue here

DigiluxAI, 2024-11-17:
same issue we are facing, we are stuck to create new group with members with the same group Id. @Koenkk Please let us know how we can resolve this problem.

HappyBassWalker, 2024-11-18:
Some problem here when I try to add new device

pietrouk, 2024-12-15:
Same problem for me too

nicola-spreafico, 2024-12-17:
I have a similar problem, I deleted and recreated the group, now the new group has again the ID=1.
When I try to add devices it simply does... nothing. There is a toast "Adding device xxx to group yyy" but then nothing, no confirmation toast, no error toast, and the device does not appear in the gorup

Gw3n4, 2024-12-30:
Hello. Same behaviour for me.

KennethLavrsen, 2025-01-02:
Confirmed. I have it too. Is there a work around?

KennethLavrsen, 2025-01-02:
To anyone looking for work around see https://github.com/Koenkk/zigbee2mqtt/issues/22904

leecrossland, 2025-01-08:
Hi, also having the same issue. I can create a group but if when I:

1. Try to add a device I get an error and device doesn't get added to the group (pic 1).
2. If I try to delete the group I also get an error and it fails to delete (pic 2).

I can remove the group by deleting from the yaml file and restarting, but this seems to be a bug.
[image]
[image]

Koenkk, 2025-01-09:
Could you provide the debug log when re-producing this + your data/database.db?

KennethLavrsen, 2025-01-09:
I do not have my buggy database but let me describe how I got into trouble and out again

First I created a group successfully and added devices to it

Then I created a group that I called Utility Group with group id 2. I added two devices and suddenly noted that the group should be Utility Room. Worried about the nane in Home Assistant I removed the group
I then created right away a new group again with id 2. Then I wanted to add devices and I got the errors like the original poster.
So I tried to remove it which also failed the first time. But I could rename it to some angry name
So now I created a new group with the right name and group id 3.
While I did that the angry group 2 suddenly showed members again that I had deleted.
Then I tried to delete the group again which went ok.
I now had group id 1 and 3.
Later I add a new group and give it group id 2 which was available. And I got errors again.

The way I finally fixed it was to stop Zigbee2MQTT, open the database in a text editor, and searched and found 3 instances in the database with group id 2. And they had no members listed.
I made a safety copy of the database and..
I manually deleted the lines with the group id 2 from the database file and started Zigbee2MQTT.
And all worked and I could add new groups and add members.

Something seems to go wrong when deleting groups and adding a new with same group id.

Gunni, 2025-01-10:
Why are group id's even exposed to the user at all?

Are they not just an implementation detail that can be hidden away?

Also, use transactions to ensure data consistency?

Begin transaction, Insert into db:

- Failure = rollback
- Success = commit

Databases 101?

nicola-spreafico, 2025-01-10:
Right now I have one group with id 2, it is used to collect 2 windows of the same room.
If I try to delete it, I'm getting this notification and then nothing:
[image]

I'm also unable to remove specific devices from the group, If I try to delete one device from the group I have similar behaviour, toast notification and then nothing
[image]

If I open `configuration.yaml` I can see there is the group definition, but I don't know if I can simply remove from here in order to delete it:
[image]

Also in the database.db` I have this
```
{"id":113,"type":"Group","groupID":2,"members":[{"deviceIeeeAddr":"0x00158d0009e7d6e0","endpointID":1},{"deviceIeeeAddr":"0x00158d008afd7204","endpointID":1}],"meta":{}}
```

I enabled debug log and I tried again to use the delete button (for the entire group or single device in the group), there is not trace at all of the operation

Gunni, 2025-03-12:
Not stale

pbel78, 2025-03-15:
Having a similar issue: after creating my first group ever after years new devices don't join anymore with error: Interview failed with "database entry xyz does not exist"

Gunni, 2025-05-15:
Still broken afaik

KennethLavrsen, 2025-07-15:
Ping bot

Gunni, 2025-09-15:
I feel like I saw a PR for this recently, didn't I?

Gunni, 2025-09-15:
https://github.com/Koenkk/zigbee2mqtt/issues/23929 https://github.com/Koenkk/zigbee2mqtt/pull/28583 does this solve the problem?

Gunni, 2025-09-15:
@mhavill your problem looks like you added devices to a group bypassing the z2m Groups UI, not sure if it is discoverable. Delete and recreate using the UI if you want it to show up in the UI.

Anyway, this ticket is a duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/23929 and should be closed.

Gunni, 2025-11-15:
Not stale

mhavill, 2025-11-15:
Hi. I tried all that first and then tried again after deleting the group. Still unable to add Sonoff TRVs to a group. RegardsMick HavillMob +61 423 023 ***@***.***

Koenkk, 2025-11-15:
https://github.com/Koenkk/zigbee2mqtt/issues/23929 is fixed now, maybe this one also? Please try with z2m 2.6.3

closed 2026-05-24 by github-actions[bot]

Gunni, 2026-05-24:
Why not close as fixed?