Koenkk/zigbee2mqtt #23372: tuya WKD-LS100 rain and sun sensor

conta1988, 2024-07-19:
### Link

https://it.aliexpress.com/item/1005007269624052.html?spm=a2g0o.order_list.order_list_main.43.7b613696tX5dxn&gatewayAdapt=glo2ita

### Database entry

{"id":19,"type":"EndDevice","ieeeAddr":"0xa4c138f98bbf43b4","nwkAddr":56375,"manufId":4742,"manufName":"zbeacon","powerSource":"Battery","modelId":"DS01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}},"genBasic":{"attributes":{"modelId":"DS01","manufacturerName":"zbeacon","powerSource":3,"zclVersion":3,"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x0000000000000000","zoneState":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":false,"meta":{},"lastSeen":1721362616873} {"id":20,"type":"EndDevice","ieeeAddr":"0xa4c1383a82f0d0fa","nwkAddr":24874,"manufId":4742,"manufName":"zbeacon","powerSource":"Battery","modelId":"DS01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}},"genBasic":{"attributes":{"modelId":"DS01","manufacturerName":"zbeacon","powerSource":3,"zclVersion":3,"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00212effff08049d","zoneState":1}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00212effff08049d","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00212effff08049d","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0,"manufacturerCode":null},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0,"manufacturerCode":null}],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-14895530},"lastSeen":1721362479267} {"id":21,"type":"EndDevice","ieeeAddr":"0xa4c138ed05ad0d64","nwkAddr":49127,"manufId":4417,"manufName":"_TZ3210_tgvtvdoc","powerSource":"Battery","modelId":"TS0207","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,4,5,1,1280,61184],"outClusterList":[3,4,6,4096,10,25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0207","manufacturerName":"_TZ3210

### Comments

i can't integrate the product, can you help me?

### External definition

```shell
battery state
battery level
rain
sun (illumination)
```

u236, 2024-07-25:
We've added support for this sensor in HOMEd ZigBee tonight, hope it will help.

HOMEd ZigBee device library record: https://github.com/u236/homed-service-zigbee/commit/329d07e46248d52285fdb76b8520ba69c3ab3ac9

Sensor sends battery state via default "Power Configuration" cluster (like any other battery-powered zigbee device).

Binary rain presence state sent via "IAS Zone" cluster (like occupancy/door and other "security" sensors), using firs bit (Alarm 1).

Illuminance intensity and rain intensity sent via "TUYA Data Points" cluster (0xEF00), and using not-standard command 0x06 (instead standard 0x02) for it. Datapoint 101 is illuminance intensity and datapoint 105 is rain intensity. Both units is "millivolts".

Andrik45719, 2024-07-25:
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0207',
            manufacturerName: '_TZ3210_tgvtvdoc',
        },
    ],
    model: 'TS0207_rain_sensor',
    vendor: 'Tuya',
    description: 'TUYA TS0207 Rain Sensor RB-SRAN01',
    fromZigbee: [tuya.fz.datapoints, fz.battery],
    toZigbee: [],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        e.water_leak(), e.battery_low(), e.battery(),
        exposes.numeric('illuminance', ea.STATE).withDescription('Raw measured illuminance').withUnit('mV'),
        exposes.numeric('rainIntensity', ea.STATE).withDescription('Raw measured rain intensity').withUnit('mV')
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [101, 'illuminance', tuya.valueConverter.raw],
            [105, 'rainIntensity', tuya.valueConverter.raw]
        ],
    },
};

module.exports = definition;

```

u236, 2024-07-25:
> ```js
>     fromZigbee: [tuya.fz.datapoints, fz.battery],
> ```

Think you forgot item for "water leak" :)
And you not need "magic request", sensor works without it.

conta1988, 2024-07-25:
for now, 
Thank you very much!

Xander-V, 2024-08-18:
Found [this review](https://smarthomescene.com/reviews/tuya-zigbee-solar-rain-sensor-rb-srain01-review/) about the sensor including an external converter:

`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            modelID: 'TS0207',
            manufacturerName: '_TZ3210_tgvtvdoc',
        },
    ],
    model: 'TS0207_rain_sensor',
    vendor: 'Tuya',
    description: 'Solar Rain Sensor RB-SRAIN01',
    fromZigbee: [tuya.fz.datapoints, fz.battery, fz.ias_water_leak_alarm_1],
    toZigbee: [],
    exposes: [
        e.water_leak('raining').withDescription('Droplet Detection (Raining)'), 
        e.numeric('illuminance', ea.STATE).withUnit('mV').withDescription('Illuminance level'),
        e.numeric('illuminance_average_20min', ea.STATE).withUnit('mV').withDescription('Illuminance average for the last 20 minutes'),
        e.numeric('illuminance_maximum_today', ea.STATE).withUnit('mV').withDescription('Illuminance maximum for the last 24 hours'),
        e.binary('cleaning_reminder', ea.STATE, 'ON', 'OFF').withDescription('Cleaning reminder'),
        e.numeric('rain_intensity', ea.STATE).withUnit('mV').withDescription('Rainfall intensity'),
        e.numeric('battery', ea.STATE).withUnit('%').withDescription('Battery level'),
        e.battery_low(),        
    ],
    meta: {
        tuyaDatapoints: [
            [4, 'battery', tuya.valueConverter.batteryPercentage],
            [101, 'illuminance', tuya.valueConverter.raw],
            [102, 'illuminance_average_20min', tuya.valueConverter.raw],
            [103, 'illuminance_maximum_today', tuya.valueConverter.raw],
            [104, 'cleaning_reminder', tuya.valueConverter.trueFalse],
            [105, 'rain_intensity', tuya.valueConverter.raw],
        ],
    },
    extend: [],
};

module.exports = definition;
`

Might help in adding it as a device. It seems however it needs some extra work to create f.i. options for heavy rain, light rain etc. since the values now are raw mV values if drops of rain are on the device.

sajikur, 2024-09-03:
This device completely changed the exposed sensors in HA during the last update. It was previously recognized as "https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector.html#tuya-ts0207_water_leak_detector" and in addition to its solar sensors, it exposed the "water_leak". The "rain intensity" gives a value of 2896 if the sensor gets wet for testing which makes little sense for developing automations. Is it therefore possible to expose the old sensors again?
Could you fix?
thank you!!!

closed 2024-12-28 by github-actions[bot]