Koenkk/zigbee2mqtt #23373: Presence sensor reports presence all the time (_TZE204_ztqnh5cg)

petepal, 2024-07-19:
### What happened?

A while ago, both my _TZE204_ztqnh5cg started to report presence at all times.

```
{
    "detection_delay": 1,
    "fading_time": 5,
    "illuminance_lux": 249,
    "linkquality": 96,
    "maximum_range": 5,
    "minimum_range": 0.5,
    "presence": true,
    "radar_sensitivity": 7,
    "target_distance": 0
}
```

In HA it reports statuses like "Home" and "Away" (instead of detect presence true / false).

Any ideas on how to debug?

### Zigbee2MQTT version

1.39.0-1

### Adapter firmware version

7.4.3 [GA]

### Adapter

Skyconnect, ember

### Setup

Add on on HA, set up as Proxmox's VM

### Debug log

```
[2024-07-19 09:55:03] debug: 	z2m: Received Zigbee message from '[Kuchnia] Czujnik obecności', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":9}],"seq":33337}' from endpoint 1 with groupID 0
[2024-07-19 09:55:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/[Kuchnia] Czujnik obecności', payload '{"detection_delay":1,"fading_time":5,"illuminance_lux":252,"linkquality":96,"maximum_range":5,"minimum_range":0.5,"presence":true,"radar_sensitivity":7,"target_distance":0}'
```

kamashkin, 2024-08-11:
Have the same issue 
[image]

negmos, 2024-09-11:
Same for me :(.. i have like 5 of them, and only one is reporting correctly!! The rest are always True!

negmos, 2024-09-11:
I tried with 1.40.1 and also with different adapters (Zigstar, SMlight) and the same result :(

Koenkk, 2024-09-12:
Could you provide the debug log when triggering presence on the device? (keep it as small as possible, start capturing, trigger presence, wait 5 secs, stop capturing logs)

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

kamashkin, 2024-09-12:
For me, the solution was to set the fading time to 2 or 3 seconds and then trigger the sensor using motion. This resolved the issue with constant presence reporting
[image]

negmos, 2024-09-13:
@kamashkin wow!! that actually worked for me! and only 2 or 3.. any other doesn't work!!
Thanks alot for the tip.

@Koenkk I created a debug log as you requested:
- Start Debug
- Included the device (it was removed)
- Triggered Presence
- Waited until "object distance = 0" but presence still True
- Triggered Presence again
- Wait 3-4 seconds
- Stop
[debuglog.log](https://github.com/user-attachments/files/16991269/debuglog.log)

negmos, 2024-09-13:
@kamashkin it worked for sometime and now not anymore 😞

kamashkin, 2024-09-13:
@negmos 

As a temporary solution, I wrote an automation that changes the fading time each time the presence sensor is triggered
<details><summary>Code of my automation</summary>
<pre>
alias: Kitchen Light LED Brightness Control
description: ""
trigger:
  - platform: state
    entity_id:
      - binary_sensor.sensor_presence_kitchen_presence
condition: []
action:
  - choose:
      - conditions:
          - condition: state
            entity_id: binary_sensor.sensor_presence_kitchen_presence
            state: "on"
        sequence:
          - type: turn_on
            device_id: dc2573b089a3d14145c8e2ead9a53ea0
            entity_id: b3df2c14133303bd542973686e35a1a8
            domain: light
      - conditions:
          - condition: state
            entity_id: binary_sensor.sensor_presence_kitchen_presence
            state: "off"
        sequence:
          - type: turn_off
            device_id: dc2573b089a3d14145c8e2ead9a53ea0
            entity_id: b3df2c14133303bd542973686e35a1a8
            domain: light
  - device_id: d69118a3b6568109cae1530fd1a39364
    domain: number
    entity_id: 3364bc6de117f573e080631806f90226
    type: set_value
    value: 3
  - delay:
      hours: 0
      minutes: 0
      seconds: 2
      milliseconds: 0
  - device_id: d69118a3b6568109cae1530fd1a39364
    domain: number
    entity_id: 3364bc6de117f573e080631806f90226
    type: set_value
    value: 2
mode: single
</pre>
</details>

[image]

Koenkk, 2024-09-14:
Looks like an issue of the device itself, @petepal do you also have this issue?

negmos, 2024-09-14:
@Koenkk i have 5 of these devices. Only one is running normally. The other 4 are doing the same behavior exactly.

negmos, 2024-09-14:
> @negmos
> 
> As a temporary solution, I wrote an automation that changes the fading time each time the presence sensor is triggered
> 
> Code of my automation
> <img alt="Screenshot 2024-09-13 at 17 33 53" width="1103" src="https://private-user-images.githubusercontent.com/3995743/367336531-3965f87b-6cc5-4443-b46d-3aa8833c4dd2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYzNDAwMjMsIm5iZiI6MTcyNjMzOTcyMywicGF0aCI6Ii8zOTk1NzQzLzM2NzMzNjUzMS0zOTY1Zjg3Yi02Y2M1LTQ0NDMtYjQ2ZC0zYWE4ODMzYzRkZDIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkxNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MTRUMTg0ODQzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Zjg4NWM3YWQwZDk2YzA1ODEzY2YzNjRjOTUzZjYxMDBiZTQ0OWJhYzU3YTU2MGVlMGI0ZmUyODgwZDUyMjI2NiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.gFdD8oqkgFyS3i_b3R_A6SERpwSZ3zcbWAKE9dqm4SU">

Thanks alot!! I will definitely try this out

petepal, 2024-09-16:
> Looks like an issue of the device itself, @petepal do you also have this issue?

After changing the fading time to 2 seconds, it seems to work just fine for some time now, no issues. But it's strange that it worked like a charm with different fading time set, until it stopped. Maybe it is a device issue indeed, the important thing is there's a workaround. Thanks, @kamashkin !

dimacoupe, 2024-11-24:
I bought two in a round and a square form factor. both are the same model _TZE204_ztqnh5cg. The square one behaves normally, but the round one behaves as described above. The round one is behaving very strangely. Sometimes it gets stuck in the "true" state. if you stop moving, it starts showing a distance of 0 meters = behaves like a motion sensor, not a presence sensor. I took them apart, inside there are the same chips and component layout, with the exception of some differences due to the shape of the cases.

dimacoupe, 2024-11-24:
so... both (square and round) works identical. few hour of playing and i can't confirm that undestand logic of work, absolutely different behavior when changing even 1 parameter, 

for short distance like wc or shower - doesn't matter how you config, all works well. main problem in distances more than 2-3 meters and maintaining the sensor's attention.

found best settings for me: 
[image]

works well for most scenarios

Jaegerstefan, 2024-11-27:
> @Koenkk found some strange bug in maximum range settings. on some values if u set and change expose tab(on bind, for example0 and back settings not saved and it's some differents setting.
> 
> I spent some time and checked all the values with default step 1,5
> 
> 9,45 >>> 655,34 9,3 >>> 0,02 8,7 >>> 655,34 8,55 >>> 0,02 5,1 >>> 655,35 4,6 >>> 0,01 4,35 >>> 655,35 2,55 >>> 655,28
> 
> i don;t know, thats converter bug or not.

i can confirm it partly. for one sensor i had the same effect, max range was set 9.3m but value in the state was 0.02

closed 2025-02-03 by github-actions[bot]