Koenkk/zigbee2mqtt #23375: Eco-Dim.05 Zigbee duodimmer works only on one side

xsienix, 2024-07-19:
### What happened?

##Problem

After making a ext_converter for this type dimmer and pairing, the dimmer is supported but works only on one side.
[image]
As a test, i deleted the device and paired it again. Now my device is unsupported. 
Supprted or not supported, i can only operate one dimmer.

**There are no error in the logs.**

My ext_converter looks like:
`const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');

const {deviceEndpoints, identify, light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['Eco-Dim.05 Zigbee'],
    model: 'Eco-Dim.05 Zigbee',
    vendor: 'EcoDim BV',
    description: 'SuperDimmer',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), identify(), light(), light()],
    meta: {"multiEndpoint":true},
    exposes: [e.light_brightness().withEndpoint('1'), e.light_brightness().withEndpoint('2')],
	configure: async (device, coordinatorEndpoint, logger) => {
		await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
		for (const ep of [1, 2]) {
			const endpoint = device.getEndpoint(ep);
			await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
			await reporting.onOff(endpoint);
			await reporting.brightness(endpoint);
		}
	},
	endpoint: (device) => {
		return {'left': 2, 'right': 1};
	},
};

module.exports = definition;
};

`
## Database
[image]

### What did you expect to happen?

Working on 2 sides

### How to reproduce it (minimal and precise)

I know, that Ecodim05 is supported, but i have the Ecodim05 Zigbee model. So reparing, deleting about 10X gives me no result. Also using the Ecodim05 code, wich is supported, leads to the exact same results. 

There are no errors in the logs.

### Zigbee2MQTT version

1.39.0

[... 145 lines cut ...]
[2024-07-19 14:19:18] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":54135,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":123,"securityuse":0,"timestamp":7664050,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[17,42,1,0,2,16,45,0,1,0,0,0]}}
[2024-07-19 14:19:18] debug: 	zh:controller: Received payload: clusterID=25, address=54135, groupID=0, endpoint=1, destinationEndpoint=13, wasBroadcast=false, linkQuality=123, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":42,"commandIdentifier":1},"payload":{"fieldControl":0,"manufacturerCode":4098,"imageType":45,"fileVersion":1},"command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}}
[2024-07-19 14:19:18] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-07-19 14:19:18] debug: 	z2m: Received Zigbee message from '0x6c5cb1fffe20b1b8', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1,"imageType":45,"manufacturerCode":4098}' from endpoint 1 with groupID 0
[2024-07-19 14:19:18] debug: 	z2m: Device '0x6c5cb1fffe20b1b8' requested OTA
[2024-07-19 14:19:18] debug: 	zh:controller:endpoint: CommandResponse 0x6c5cb1fffe20b1b8/1 genOta.queryNextImageResponse({"status":152}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-07-19 14:19:18] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x6c5cb1fffe20b1b8:54135/1 (0,0,1)
[2024-07-19 14:19:18] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":54135,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":19,"options":0,"radius":30,"len":4,"data":{"type":"Buffer","data":[25,42,2,152]}}
[2024-07-19 14:19:18] debug: 	zh:zstack:unpi:writer: --> frame [254,14,36,1,119,211,1,1,25,0,19,0,30,4,25,42,2,152,54]
[2024-07-19 14:19:18] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-07-19 14:19:18] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-07-19 14:19:18] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-07-19 14:19:18] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-07-19 14:19:18] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-07-19 14:19:18] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,19,213]
[2024-07-19 14:19:18] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,19,213]
[2024-07-19 14:19:18] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,19] - 213
[2024-07-19 14:19:18] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":19}
[2024-07-19 14:19:18] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-07-19 14:19:18] debug: 	z2m: Responded to OTA request of '0x6c5cb1fffe20b1b8' with 'NO_IMAGE_AVAILABLE'
[2024-07-19 14:21:38] debug: 	z2m: Saving state to file /config/zigbee2mqtt/state.json
[2024-07-19 14:26:38] debug: 	z2m: Saving state to file /config/zigbee2mqtt/state.json
[2024-07-19 14:31:38] debug: 	z2m: Saving state to file /config/zigbee2mqtt/state.json
[2024-07-19 14:36:38] debug: 	z2m: Saving state to file /config/zigbee2mqtt/state.json
`

ozanmakes, 2024-07-19:
Continuing from #22370. I used the following config to make this device "work". Setting brightness of left and right works, and setting brightness to 0 turns the light off. 

If I set the light to be "off" instead however, I can't turn it back on by setting state to "on". For now I am working around this by only using brightness. I would love to have it fully supported though. Feedback welcome.

```js
const {deviceEndpoints, identify, light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['Eco-Dim.05 Zigbee'],
    model: 'Eco-Dim.05 Zigbee',
    vendor: 'EcoDim BV',
    description: 'LED dimmer duo 2x 0-100W',
    extend: [
        deviceEndpoints({endpoints: {"Left": 1, "Right": 2}}),
        light({ endpointNames:["Left"] }),
        light({ endpointNames:["Right"] })
    ],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

ozanmakes, 2024-07-19:
I had some luck by using a custom toZigbee method that sends brightness events when it encounters state changes. This gets me much closer to where I want to be:

```js
const {
  deviceEndpoints,
  identify,
  light,
} = require("zigbee-herdsman-converters/lib/modernExtend");

const customToZigbee = [
  {
    key: ["state"],
    convertSet: async (entity, key, value, meta) => {
      console.log("set state custom", { entity, key, value, meta });
      if (value === "OFF") {
        await entity.command("genLevelCtrl", "moveToLevelWithOnOff", {
          level: 0,
          transtime: 0,
        });
      } else if (value === "ON") {
        // Don't set the brightness value if explicit one is provided
        if (!meta.message.hasOwnProperty("brightness")) {
          await entity.command("genLevelCtrl", "moveToLevelWithOnOff", {
            level: 128, // Update the default brightness value to one you prefer (between 0-254)
            transtime: 0,
          });
        }
      }
    },
  },
];

const ecoDimLight = light({ endpointNames: ["left", "right"] });

const ecoDimLightCustom = {
  ...ecoDimLight,
  toZigbee: [...customToZigbee, ...ecoDimLight.toZigbee],
};

const definition = {
  zigbeeModel: ["Eco-Dim.05 Zigbee"],
  model: "Eco-Dim.05 Zigbee",
  vendor: "EcoDim BV",
  description: "LED dimmer duo 2x 0-100W",
  extend: [
    deviceEndpoints({ endpoints: { left: 1, right: 2 } }),
    identify(),
    ecoDimLightCustom,
  ],
  meta: { multiEndpoint: true },
};

module.exports = definition;
```

Of course, with this the state of the light is not correctly represented in the dashboard. There are also other edge cases such as Home Assistant sending current/new brightness together with state changes. I handled this for some cases but not all.

xsienix, 2024-07-20:
THIS IS AWSOME !!!

I'm sitting here for more then a week now, turning my head around, used different code, and all of them did not work properly.

1. how do you manage to fill in that code so nicely on this page, with colors and all?
2. what was the trick, that your code is working?
3. are you able to give me a hand, because i'm not a code genius, but try to understand what it was that your code works.

Again, you make my weekend!
Thanks so much!

ozanmakes, 2024-07-24:
Glad to hear it worked for you @xsienix. You can use [triple backtick (```js ... \`\`\`) ](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks) syntax for posting code snippet here.

I am a programmer by trade, but new to home automation in general. Your threads here tipped me off to possibilities of zigbee2mqtt and that I can create "converters" that change behavior of the device.

After I observed setting brightness worked, but toggling off/on caused buggy behavior, I implemented a custom "set off" and "set on" behavior for the device using custom converters. When the state is set to OFF or ON, this code issues a "set brightness" command instead.

xsienix, 2024-07-25:
> Glad to hear it worked for you @xsienix. You can use [triple backtick (```js ... ```) [...]

Thank you!

```this is code```

I'm not a programmer but eager to learn what is needed to understand al this. Meanwhile my learning curve is verticaly :)
Zigbee2mqtt converts zigbee information to mqtt information en vice versa. Herdman converters make this happen i presume. 

The dimmer works perfect!
Now i'm searching to get the turning knobs of the device working, so if you turn the knob on the device (or when you push the knob, you activate the dimmer) the light will go on in a specific brightness.

closed 2024-12-28 by github-actions[bot]

morm01, 2025-12-23:
Hello,

I've got a actual eco-dim.05 model with firmware version 1.03 and I operate it with Zigbee2MQTT version 2.7.1
It is showed as supported device and its working, except the dimming on the right channel as described here.
Tried to work with your converter, but the rigth channel (channel 2 in your converter) will not dim.

Here is my db entry for eco-dim.05:
`{"id":14,"type":"Router","ieeeAddr":"0x6c5cb1fffe20ab05","nwkAddr":20629,"manufId":4107,"manufName":"EcoDim BV","powerSource":"Mains (single phase)","modelId":"eco-dim05-zigbee","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"powerSource":0,"appVersion":0,"stackVersion":0,"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":257,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"powerSource":0,"appVersion":0,"stackVersion":0,"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20250314","swBuildId":"1.03","zclVersion":8,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1766529385376}
`

Did you tried your converter also with a newer version (I saw your eco-dim.05 is on version 1.00)?

morm01, 2025-12-29:
Hello,
meanwhile I also found this necessary settings in a other issue regarding the eco-dim.05.
With reporting konfigured like described by you, it will work also without a converter since eco-dim is supported under V. 2.7.1…