Koenkk/zigbee2mqtt #23376: Error: Cluster with name '64529' does not exist" for SONOFF smart water valve SWV

jocamane, 2024-07-19:
### What happened?

After pair SONOFF SWV (https://www.zigbee2mqtt.io/devices/SWV.html#sonoff-swv) the power is not active and in cluster tab appeared 64529

After reconfigure (buttom) the following errors appear in the log: Error: Cluster with name '64529' does not exist"

tested with device version Firmware build date: 20240705 / Firmware version: 1.0.3

[image]

[image]

### What did you expect to happen?

The power field is populate and no exists errors in reconfigure

### How to reproduce it (minimal and precise)

Pair the device. after pair press reconfigure buttom

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

0x26780700

### Adapter

ConBee2

### Setup

Add-on on RPI 4b

### Debug log

LOG PAIR:

Info 2024-07-19 13:22:35MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffe17e71c', payload '{"last_seen":"2024-07-19T13:22:35+01:00","linkquality":255,"update":{"installed_version":-1,"latest_version":-1}}'
Info 2024-07-19 13:22:35Interview for '0x0ceff6fffe17e71c' started
Info 2024-07-19 13:22:35Device '0x0ceff6fffe17e71c' joined
Info 2024-07-19 13:22:35MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0ceff6fffe17e71c","ieee_address":"0x0ceff6fffe17e71c"},"type":"device_joined"}'
Info 2024-07-19 13:22:35MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0ceff6fffe17e71c"},"type":"device_connected"}'
Info 2024-07-19 13:22:35Starting interview of '0x0ceff6fffe17e71c'
Info 2024-07-19 13:22:36MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0ceff6fffe17e71c","ieee_address":"0x0ceff6fffe17e71c","status":"started"},"type":"device_interview"}'
Info 2024-07-19 13:22:36MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0ceff6fffe17e71c"},"type":"pairing"}'
Info 2024-07-19 13:22:37MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffe17e71c', payload '{"last_seen":"2024-07-19T13:22:37+01:00","linkquality":255,"update":{"installed_version":-1,"latest_version":-1}}'
Info 2024-07-19 13:22:38MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffe17e71c', payload '{"last_seen":"2024-07-19T13:22:38+01:00","linkquality":255,"update":{"installed_version":-1,"latest_version":-1}}'
Info 2024-07-19 13:22:38Succesfully interviewed '0x0ceff6fffe17e71c'
Info 2024-07-19 13:22:38Successfully interviewed '0x0ceff6fffe17e71c', device has successfully been paired
Info 2024-07-19 13:22:38Device '0x0ceff6fffe17e71c' is supported, identified as: SONOFF Zigbee smart water valve (SWV)
Info 2024-07-19 13:22:38MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee smart water valve","exposes":[{"access":1,"description":"Current water flow","label":"Flow","name":"flow","property":"flow","type":"numeric","unit":"m³/h"},{"access":5,"category":"diagnostic","description":"Remaining battery in %","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":5,"description":"The water valve is in normal state, water shortage or water leakage","label":"Current device status","name":"current_device_status","property":"current_device_status","type":"enum","values":["normal_state","water_shortage","water_leakage","water_shortage & water_leakage"]},{"features":[{"access":7,"description":"On/off state of the switch","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":7,"description":"Smart water valve cycle timing irrigation","features":[{"access":1,"description":"Number of times it has been executed","label":"Current count","name":"current_count","property":"current_count","type":"numeric","unit":"times"},{"access":3,"description":"Total times of circulating irrigation","label":"Total number","name":"total_number","property":"total_number","type":"numeric","unit":"times","value_max":100,"value_min":0},{"access":3,"description":"Single irrigation duration","label":"Irrigation duration","name":"irrigation_duration","property":"irrigation_duration","type":"numeric","unit":"seconds","value_max":86400,"value_min":0},{"access":3,"description":"Time interval between two adjacent irrigation","label":"Irrigation interval","name":"irrigation_interval","property":"irrigation_interval","type":"numeric","unit":"seconds","value_max":86400,"value_min":0}],"label":"Cyclic timed irrigation","name":"cyclic_timed_irrigation","property":"cyclic_timed_irrigation","type":"composite"},{"access":7,"description":"Smart water valve circulating quantitative irrigation","features":[{"access":1,"description":"Number of times it has been executed","label":"Current count","name":"current_count","property":"current_count","type":"numeric","unit":"times"},{"access":3,"description":"Total times of circulating irrigation","label":"Total number","name":"total_number","property":"total_number","type":"numeric","unit":"times","value_max":100,"value_min":0},{"access":3,"description":"Single irrigation capacity","label":"Irrigation capacity","name":"irrigation_capacity","property":"irrigation_capacity","type":"numeric","unit":"liter","value_max":6500,"value_min":0},{"access":3,"description":"Time interval between two adjacent irrigation","label":"Irrigation interval","name":"irrigation_interval","property":"irrigation_interval","type":"numeric","unit":"seconds","value_max":86400,"value_min":0}],"label":"Cyclic quantitative irrigation","name":"cyclic_quantitative_irrigation","property":"cyclic_quantitative_irrigation","type":"composite"},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"SWV","options":[{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":true,"vendor":"SONOFF"},"friendly_name":"0x0ceff6fffe17e71c","ieee_address":"0x0ceff6fffe17e71c","status":"successful","supported":true},"type":"device_interview"}'
Info 2024-07-19 13:22:38MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Zigbee smart water valve","friendly_name":"0x0ceff6fffe17e71c","model":"SWV","supported":true,"vendor":"SONOFF"},"type":"pairing"}'
Info 2024-07-19 13:22:38MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffe17e71c/flow/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x0ceff6fffe17e71c/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x0ceff6fffe17e71c"],"manufacturer":"SONOFF","model":"Zigbee smart water valve (SWV)","name":"0x0ceff6fffe17e71c","sw_version":"1.0.3","via_device":"zigbee2mqtt_bridge_0x00212effff07afaa"},"enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x0ceff6fffe17e71c","name":"Flow","object_id":"0x0ceff6fffe17e71c_flow","origin":{"name":"Zigbee2MQTT","sw":"1.38.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/0x0ceff6fffe17e71c","unique_id":"0x0ceff6fffe17e71c_flow_zigbee2mqtt","unit_of_measurement":"m³/h","value_template":"{{ value_json.flow }}"}'
Info 2024-07-19 13:22:38MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffe17e71c/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x0ceff6fffe17e71c/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x0ceff6fffe17e71c"],"manufacturer":"SONOFF","model":"Zigbee smart water valve (SWV)","name":"0x0ceff6fffe17e71c","sw_version":"1.0.3","via_device":"zigbee2mqtt_bridge_0x00212effff07afaa"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x0ceff6fffe17e71c","object_id":"0x0ceff6fffe17e71c_battery","origin":{"name":"Zigbee2MQTT","sw":"1.38.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x0ceff6fffe17e71c","unique_id":"0x0ceff6fffe17e71c_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
[... cut ...] odules/zigbee-herdsman-converters/src/index.ts:173:21) at Configure.configure (/app/lib/extension/configure.ts:121:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Info 2024-07-19 13:30:10MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Sistema Rega Fora"},"error":"Failed to configure (Cluster with name '64529' does not exist)","status":"error","transaction":"xmfj3-3"}'

paulz-gh, 2024-07-19:
Hi, I had the same errors but they seem to have disappeared when I updated to 1.39.0.

jocamane, 2024-07-19:
> Hi, I had the same errors but they seem to have disappeared when I updated to 1.39.0.

Ok, I wait more 2 weeks to update to 1.39.0 and after that I report again

jocamane, 2024-07-19:
However, after 6 hours I already have power value

Kuechenfenster, 2024-07-29:
@jocamane,
can you please share your zigbee2mqtt details setting for this valve!

jocamane, 2024-07-29:
Now, after install th Z2M [1.39.0] the tab "64529" disappeared, and if I reconfigure no more errors.

SOLVED

[image]
[image]
[image]
[image]
[image]
[image]
[image]

closed 2024-07-29 by jocamane

quensen, 2024-07-31:
Hi @jocamane 

I am wondering if you needed to create an external converter for your SWV device? I do, but I never needed to create one before. Could you share yours, please?

Sorry for off-topicness.

Stefano

isaga1234, 2024-09-28:
I'm having the same issue: using version [1.40.1] commit: [403d3c0]. I'll wait for the next release, maybe it gets fixed with an update