Koenkk/zigbee2mqtt #23378: Error: ZCL command failed (Timeout - 49529 - 1 - 193 - 258 - 11 after 10000ms)

kevincaradant, 2024-07-20:
### What happened?

When I try to change the position of my roller, I can see the roller move physically, but the position reported by Z2M and consequently by HA is incorrect; it remains unchanged. I checked the logs and encountered this error. Could you help me understand why this error occurs?

### What did you expect to happen?

I expect to receive the requested position accurately, without any errors or incorrect readings reported by Z2M.

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

20221226

### Adapter

Sonoff Dongle P 

### Setup

x86-64 using docker

### Debug log

[2024-07-20 11:25:33] error: 	z2m: Publish 'set' 'position' to 'XXXXXXXXX' failed: 'Error: ZCL command 0xa4c138e8c1443b63/1 closuresWindowCovering.goToLiftPercentage({"percentageliftvalue":51}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 49529 - 1 - 193 - 258 - 11 after 10000ms)'

davidalonsousal, 2024-08-26:
I have the same problem:

z2m: Publish 'set' 'position' to 'xxxxx' failed: 'Error: ZCL command 0xa4c13868708cecb5/1 closuresWindowCovering.goToLiftPercentage({"percentageliftvalue":3}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 61179 - 1 - 255 - 258 - 11 after 10000ms)'

jsanzherrero, 2024-11-05:
I have this issue too
HomeAssistant: 2024.10.4
RPi4 + Sonoff Dongle P with Ember 7.4.4.0
Z2M: 1.41.0-1

Dexter0007, 2024-11-30:
> Same problem from yesterday

Problem solved by disconnect and coccnect again device power supply

closed 2025-02-06 by github-actions[bot]