Koenkk/zigbee2mqtt #23379: [Wrong device]: Miboxer E2-ZR shown as TuYa TS0502B (re-open)

andrew01144, 2024-07-20:
### Link

https://www.zigbee2mqtt.io/devices/E2-ZR.html

### Model

E2-ZR

### Description

LED controller

### Vendor

MiBoxer

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/E2-ZR.png

### Database entry

{"id":24,"type":"Router","ieeeAddr":"0xa4c138f5a21a2354","nwkAddr":4867,"manufId":4417,"manufName":"_TZB210_ayx58ft5","powerSource":"Mains (single phase)","modelId":"TS0502B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZB210_ayx58ft5","zclVersion":3,"appVersion":64,"modelId":"TS0502B","powerSource":1}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500,"colorTemperature":0}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002e1e9f4e","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b002e1e9f4e","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b002e1e9f4e","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":8,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":768,"attrId":7,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1721467873839}

### Notes

Hi,
This is a re-open of [Wrong device]: Miboxer E2-ZR shown as TuYa TS0502B https://github.com/Koenkk/zigbee2mqtt/issues/21185

I have the same problem, but have not succeeded in fixing it.

Starting with Z2M from around 01-Jun, that's before the fix.
I my E2-ZR successfully joined Z2M, but it appeared as Tuya TS0502B.
I upgraded Z2M; herdsman now shows as 19.62.0.
I removed the device. Re-joined the device, but it still appears as Tuya TS0502B.
Am I missing a step?

I tried bringing up a new instance of Z2M with a different adapter. Still appears as Tuya TS0502B.
I tried commenting the E2-ZR line in devices/tuya.js. Still appears as Tuya TS0502B.

Many thanks for any help!

Also - setting 'transition' seems to have no affect on the actual transition.

pvyleta, 2024-09-04:
I found the original [PR ](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7642)that added support for this device, and my feeling is that the author wanted to add the `E3-ZR` instead.

I will try to add the device as an alias to the `Miboxer FUT035` which is older Dual-white controller of the same manufacturer instead, let's see if that works.

andrew01144, 2024-09-04:
@pvyleta thanks for taking a look. I have an E2-ZR in front of me. Note that it can be configured as "Single Color" or "Dual White" by pressing the button on the unit. Let me know if it needs to be in a particular config, and let me know if you would like me to try anything.

pvyleta, 2024-09-04:
@andrew01144 That should be fine, since the FUT035+ has the same feature. I mean it is basically identical device feature-wise, just in newer box.

So I tried editing the code on my instance, and now it works. It claims it failed the interview for whatever reason, but I can control the color etc anyway. I will create a PR soon.

andrew01144, 2024-09-05:
@pvyleta I think I understand what you have done, so I tried the same. It now identifies as MiBoxer	FUT035Z+ and, as far as I can tell, works correctly. Thanks for the inspiration.

pvyleta, 2024-09-05:
Glad you confirmed the idea was correct! The [PR ](https://github.com/Koenkk/zigbee2mqtt/issues/23379#issuecomment-2332173556)is merged so should be in the next release.

closed 2024-12-28 by github-actions[bot]

jamrijur, 2025-02-03:
I have the same problem. Here are my parameters. I don't understand how I can fix it.

Zigbee2MQTT version
[2.1.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.1.0) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
Coordinator type
EmberZNet
Coordinator revision
7.4.4 [GA]
Coordinator IEEE Address
0x8c65a3fffe521d43
Frontend version
0.9.4
zigbee-herdsman-converters version
21.27.1
zigbee-herdsman version
3.2.5

pvyleta, 2025-02-07:
@jamrijur unlikely. This problem was fixed.

jamrijur, 2025-02-08:
So I can't fix it myself. So I can't fix it myself. I have Miboxer E2-ZR and it reports itself as Zigbee Model: TS0502B
Zigbee Manufacturer: _TZB210_eiwanbeb
I also tried adding external_converters:
  - tuya.ts
I downloaded the latest tuya.ts

Kapteenivamma, 2025-02-26:
I have the exact same problem. I have 3pcs of FUT035+ and they are seen as Tuya ts0502b. At the beginning they were seen correctly and they worked great. Now the transition doesen´t work and some times they are unresponsive. Z2M 2.1.1

pvyleta, 2025-02-28:
> I have 3pcs of FUT035+
@Kapteenivamma This thread is about E2-ZR, so it seems to me you have a different problem. This is a closed issue, you are unlikely to get any help here.