Koenkk/zigbee2mqtt #23388: TRVZB Firmware 1.1.5 --> Better-thermostat broken

Unclezz, 2024-07-21:
### What happened?

Apparently firmware 1.1.5 makes Better-Thermostat custom add-on from HACS unusable.

Everytime you try to set temperature or anything else from HA to the TRV, the TRV goes into OF mode (from the display on the valve itself.

### What did you expect to happen?

Simply able to use the valve leveraging Better-thermostat

### How to reproduce it (minimal and precise)

Add the TRV in Better-Thermostat as new device;
add a Better-Thermostat sensor card to control the TRV and try to change whatever value (from temperature setting to heting off or anything else;
check status on the TRV and notice it is now set to OF.

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

20240710.0

### Adapter

EZSP V8 Tasmota 13.2.0

### Setup

HA on Qemu VM

interru, 2024-09-09:
Wasn't a problem on my side. I assume that you missed the "outdoor temperature threshold".
If the outside temperature is over that threshold, the devices are reset to off by better-thermostat regardless if you manually set a temperature.

I'm not using better-thermostat currently because the local_temperature_calibration is weird and after looking at the calculations I decided to rather do most of the automations myself.

ignisf, 2024-11-02:
I also cannot reproduce the issue.

closed 2025-01-10 by github-actions[bot]