Koenkk/zigbee2mqtt #23393: Ping timeout with Sonoff_Zigbee_3.0_USB_Dongle_Plus on x86, works fine on RPI

Joop9, 2024-07-22:
### What happened?

I've running Zigbee2MQTT in a docker container on a x86 system. It works for a few weeks, but after a reboot I get the error 
z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
With the same dongle and configuration file it works fine on my old Raspberry Pi (non-docker, version 1.33.0)
The first error is a few months ago, the work-around was using the old RPI but that's not fine. I would switch back to a single machine solution. I don't know the exact situation on that moment
I've tested all the issues on https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html, but nothing solved the problem

### What did you expect to happen?

it works on the same way on X64 as RPI.

### How to reproduce it (minimal and precise)

For now, start the docker container, wait a few moments and it stops.

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

20240315

### Adapter

Sonoff Zigbee 3. USB Dongle Plus

### Setup

Docker container on x64 (N3050)

### Debug log

Starting Zigbee2MQTT without watchdog.
[2024-07-22 21:15:27] info: 	z2m: Logging to console, file (filename: log.log)
[2024-07-22 21:15:27] debug: 	z2m: Loaded state from file /app/data/state.json
[2024-07-22 21:15:27] info: 	z2m: Starting Zigbee2MQTT version 1.39.0 (commit #0326926)
[2024-07-22 21:15:27] info: 	z2m: Starting zigbee-herdsman (0.50.1)
[2024-07-22 21:15:27] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\"},\"adapter\":{\"concurrent\":null,\"delay\":null,\"disableLED\":false}}"'
[2024-07-22 21:15:29] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
[2024-07-22 21:15:29] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-07-22 21:15:29] info: 	zh:zstack:znp: Serialport opened
[2024-07-22 21:15:29] debug: 	zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-07-22 21:15:29] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-07-22 21:15:29] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-07-22 21:15:29] debug: 	zh:zstack:unpi:writer: --> buffer [239]
[2024-07-22 21:15:30] debug: 	zh:zstack:unpi:parser: <-- [49,45,48,58,53,49,46,55,46,48,40,48,48,48,42,65,41,13,10,49,45,48,58,55,49,46,55,46,48,40,48,48,48,42,65,41,13,10]
[2024-07-22 21:15:30] debug: 	zh:zstack:unpi:parser: --- parseNext [49,45,48,58,53,49,46,55,46,48,40,48,48,48,42,65,41,13,10,49,45,48,58,55,49,46,55,46,48,40,48,48,48,42,65,41,13,10]
[2024-07-22 21:15:30] debug: 	zh:zstack:unpi:parser: <-- [33,52,54,50,65,13,10]
[2024-07-22 21:15:30] debug: 	zh:zstack:unpi:parser: --- parseNext [49,45,48,58,53,49,46,55,46,48,40,48,48,48,42,65,41,13,10,49,45,48,58,55,49,46,55,46,48,40,48,48,48,42,65,41,13,10,33,52,54,50,65,13,10]
[2024-07-22 21:15:30] debug: 	zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-07-22 21:15:30] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-07-22 21:15:30] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-07-22 21:15:31] debug: 	zh:zstack:unpi:parser: <-- [48,45,48,58,57,54,46,49,52,46,48,40,48,48,48,50,41,13,10]
[2024-07-22 21:15:31] debug: 	zh:zstack:unpi:parser: --- parseNext [49,45,48,58,53,49,46,55,46,48,40,48,48,48,42,65,41,13,10,49,45,48,58,55,49,46,55,46,48,40,48,48,48,42,65,41,13,10,33,52,54,50,65,13,10,48,45,48,58,57,54,46,49,52,46,48,40,48,48,48,50,41,13,10]
[2024-07-22 21:15:31] debug: 	zh:zstack:unpi:parser: <-- [48,45,48,58,57,54,46,55,46,50,49,40,48,48,48,49,52,41,13,10,48,45,48,58,57,54,46,55,46,57,40,48,48,48,48,56,41,13,10]
[2024-07-22 21:15:31] debug: 	zh:zstack:unpi:parser: --- parseNext [49,45,48,58,53,49,46,55,46,48,40,48,48,48,42,65,41,13,10,49,45,48,58,55,49,46,55,46,48,40,48,48,48,42,65,41,13,10,33,52,54,50,65,13,10,48,45,48,58,57,54,46,49,52,46,48,40,48,48,48,50,41,13,10,48,45,48,58,57,54,46,55,46,50,49,40,48,48,48,49,52,41,13,10,48,45,48,58,57,54,46,55,46,57,40,48,48,48,48,56,41,13,10]
[2024-07-22 21:15:31] debug: 	zh:zstack:unpi:parser: <-- [49,45,48,58,51,50,46,51,50,46,48,40,48,48,49,48,51,41,13,10,49,45,48,58,53,50,46,51,50,46,48,40,48,48,48,57,49,41,13,10]
[2024-07-22 21:15:31] debug: 	zh:zstack:unpi:parser: --- parseNext [49,45,48,58,53,49,46,55,46,48,40,48,48,48,42,65,41,13,10,49,45,48,58,55,49,46,55,46,48,40,48,48,48,42,65,41,13,10,33,52,54,50,65,13,10,48,45,48,58,57,54,46,49,52,46,48,40,48,48,48,50,41,13,10,48,45,48,58,57,54,46,55,46,50,49,40,48,48,48,49,52,41,13,10,48,45,48,58,57,54,46,55,46,57,40,48,48,48,48,56,41,13,10,49,45,48,58,51,50,46,51,50,46,48,40,48,48,49,48,51,41,13,10,49,45,48,58,53,50,46,51,50,46,48,40,48,48,48,57,49,41,13,10]
[2024-07-22 21:15:31] debug: 	zh:zstack:unpi:parser: <-- [49,45,48,58,50,49,46,55,46,48,40,48,48,46,48,50,48,42,107,87,41,13,10,49,45,48,58,52,49,46,55,46,48,40,48,48,46]
[2024-07-22 21:15:31] debug: 	zh:zstack:unpi:parser: --- parseNext [49,45,48,58,53,49,46,55,46,48,40,48,48,48,42,65,41,13,10,49,45,48,58,55,49,46,55,46,48,40,48,48,48,42,65,41,13,10,33,52,54,50,65,13,10,48,45,48,58,57,54,46,49,52,46,48,40,48,48,48,50,41,13,10,48,45,48,58,57,54,46,55,46,50,49,40,48,48,48,49,52,41,13,10,48,45,48,58,57,54,46,55,46,57,40,48,48,48,48,56,41,13,10,49,45,48,58,51,50,46,51,50,46,48,40,48,48,49,48,51,41,13,10,49,45,48,58,53,50,46,51,50,46,48,40,48,48,48,57,49,41,13,10,49,45,48,58,50,49,46,55,46,48,40,48,48,46,48,50,48,42,107,87,41,13,10,49,45,48,58,52,49,46,55,46,48,40,48,48,46]
[2024-07-22 21:15:31] debug: 	zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-07-22 21:15:31] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-07-22 21:15:31] debug: 	zh:zstack:unpi:parser: <-- [48,45,49,58,50,52,46,50,46,49,40,50,52,48,55,50,50,50,49,49,53,48,54,83,41,40,48,54,53,51,49,46,49,51,50,42,109,51,41,13,10]
[2024-07-22 21:15:31] debug: 	zh:zstack:unpi:parser: --- parseNext [49,45,48,58,53,49,46,55,46,48,40,48,48,48,42,65,41,13,10,49,45,48,58,55,49,46,55,46,48,40,48,48,48,42,65,41,13,10,33,52,54,50,65,13,10,48,45,48,58,57,54,46,49,52,46,48,40,48,48,48,50,41,13,10,48,45,48,58,57,54,46,55,46,50,49,40,48,48,48,49,52,41,13,10,48,45,48,58,57,54,46,55,46,57,40,48,48,48,48,56,41,13,10,49,45,48,58,51,50,46,51,50,46,48,40,48,48,49,48,51,41,13,10,49,45,48,58,53,50,46,51,50,46,48,40,48,48,48,57,49,41,13,10,49,45,48,58,50,49,46,55,46,48,40,48,48,46,48,50,48,42,107,87,41,13,10,49,45,48,58,52,49,46,55,46,48,40,48,48,46,48,45,49,58,50,52,46,50,46,49,40,50,52,48,55,50,50,50,49,49,53,48,54,83,41,40,48,54,53,51,49,46,49,51,50,42,109,51,41,13,10]
[2024-07-22 21:15:31] debug: 	zh:zstack:unpi:parser: <-- [47,88,77,88,53,76,71,70,48,48,49,48,52,52,52,56,48,54,48,57,56,13,10,13,10,49,45,51,58,48,46,50,46,56,40,53,48,41,13,10]
[2024-07-22 21:15:31] debug: 	zh:zstack:unpi:parser: --- parseNext [49,45,48,58,53,49,46,55,46,48,40,48,48,48,42,65,41,13,10,49,45,48,58,55,49,46,55,46,48,40,48,48,48,42,65,41,13,10,33,52,54,50,65,13,10,48,45,48,58,57,54,46,49,52,46,48,40,48,48,48,50,41,13,10,48,45,48,58,57,54,46,55,46,50,49,40,48,48,48,49,52,41,13,10,48,45,48,58,57,54,46,55,46,57,40,48,48,48,48,56,41,13,10,49,45,48,58,51,50,46,51,50,46,48,40,48,48,49,48,51,41,13,10,49,45,48,58,53,50,46,51,50,46,48,40,48,48,48,57,49,41,13,10,49,45,48,58,50,49,46,55,46,48,40,48,48,46,48,50,48,42,107,87,41,13,10,49,45,48,58,52,49,46,55,46,48,40,48,48,46,48,45,49,58,50,52,46,50,46,49,40,50,52,48,55,50,50,50,49,49,53,48,54,83,41,40,48,54,53,51,49,46,49,51,50,42,109,51,41,13,10,47,88,77,88,53,76,71,70,48,48,49,48,52,52,52,56,48,54,48,57,56,13,10,13,10,49,45,51,58,48,46,50,46,56,40,53,48,41,13,10]
[2024-07-22 21:15:32] debug: 	zh:zstack:unpi:parser: <-- [48,45,48,58,57,54,46,49,52,46,48,40,48,48,48,50,41,13,10]
[... cut ...]
[2024-07-22 21:15:49] error: 	z2m: Exiting...
[2024-07-22 21:15:49] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:101:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:129:29)
    at Zigbee.start (/app/lib/zigbee.ts:63:27)
    at Controller.start (/app/lib/controller.ts:139:27)
    at start (/app/index.js:154:5)

nono20004, 2024-07-28:
Hello, I have the same problem. It x86 version work few minuts and stop it

Joop9, 2024-09-10:
Seems to bee multiple issues. on this moment it works, even na reboot. 
issues found in docker-compose.yml:
- missing `restart: unless-stopped`
- `privileged: true` results in Failed to connect to the adapter (do not use them!)
- `group-add=dialout` might give better results
I hope this changes gives a stable situation

closed 2024-12-28 by github-actions[bot]