Koenkk/zigbee2mqtt #23397: IKEA LED1842G3 turns on when sending off command

alex3305, 2024-07-23:
### What happened?

Recently I upgraded a couple of my [IKEA LED1842G3](https://www.zigbee2mqtt.io/devices/LED1842G3.html) to the offered firmware [1.1.4](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html) from 1.1.006 and since than these IKEA bulbs turn on on 1% brightness when issued the turn off command. Turning them off again after that will turn them off and they stay off.

This issue feels similar to #16388

### How to reproduce it (minimal and precise)

1. Turn on the light, this will turn the light on at the previous brightness
2. Turn off the light, this will turn the light off
3. Turn off the light again, this will turn the light on again at minimum brightness
4. Turn off the light again, this will turn the light off'

After this step it is impossible to repeat 3. You'll need to turn the light on again for step 3 to re-occur.

### Zigbee2MQTT version

1.39.0 commit: 0326926

### Adapter firmware version

20230507

### Adapter

Sonoff ZBDongle-P

### Setup

Plain in a Docker container on Debian server x64

### Debug log

I've added a minimal debug log: [debug-log.txt](https://github.com/user-attachments/files/16346633/debug-log.txt)

Koenkk, 2024-07-24:
This looks very similar to https://github.com/Koenkk/zigbee2mqtt/issues/22030 and can be fixed by removing the `transition` for your device.

alex3305, 2024-07-25:
> This looks very similar to #22030 and can be fixed by removing the `transition` for your device.

It seems you are correct. This is a dupe. Sorry for not searching thoroughly enough. Thank you for taking the time to respond ❤️.

closed 2024-07-25 by alex3305