Koenkk/zigbee2mqtt #23398: [New device support]: Saswell irrigation controller SAS980SWT-7-ZTU model 'TS0601' and manufacturer name ' _TZE204_qtnjuoae'

TechsDomadoo, 2024-07-23:
### Link

https://www.domadoo.fr/fr/jardin-connecte/5886-saswell-electrovanne-et-programmateur-d-arrosage-connecte-zigbee-mesure-de-consommation.html

### Database entry

Type d'appareil EndDevice Modèle Zigbee TS0601 Constructeur Zigbee _TZE204_qtnjuoae État de prise en charge Sans valeur (how_to_add_support)  Adresse IEEE 0xa4c1385bf1c8d119 Adresse réseau 0x15F6

### Comments

Previous Saswell irrigation controller is supported (SAS980SWT-7-Z01), but this new model is unsupported.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_qtnjuoae',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

Eoras, 2024-09-10:
Already not supported ? :(

Mific78, 2024-09-17:
Hello, I bought the new model of the Saswell irrigation controller and I confirm it's not supported by the last release (1.40.1). Do you know when it is scheduled ?

papyrus78, 2024-10-25:
Hello, 
I use this conversion file to test :

```ruby

const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const GALLON_TO_LITER = 3.78541;

const definition = {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE204_qtnjuoae',]),
        zigbeeModel: ['SAS980SWT-7-ZTU'],
        model: 'SAS980SWT-7-ZTU',
        vendor: 'SASWELL',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        description: 'Zigbee smart water valve',
        onEvent: tuya.onEventSetTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn the valve On or Off'),
            e.enum('status', ea.STATE, ['0', '1', '2', '3']).withDescription('Status'),
            e.numeric('countdown', ea.STATE_SET).withUnit('min').withValueMin(1).withValueMax(240).withDescription('Countdown'),
            e.numeric('countdown_left', ea.STATE).withUnit('min').withValueMin(1).withValueMax(240).withDescription('Countdown left'),
            e.numeric('water_current', ea.STATE).withUnit('L/min').withValueMin(0).withValueMax(3785.41).withValueStep(0.1)
                .withDescription('Current water flow (L/min)'), // Changed to liters
            e.battery(),
            e.numeric('water_total', ea.STATE).withUnit('L').withValueMin(0).withValueMax(378541.0).withValueStep(0.001)
                .withDescription('Total water flow (L)'), // Changed to liters
            e.binary('fault', ea.STATE, 'DETECTED', 'NOT_DETECTED').withDescription('Fault status'),
            e.enum('weather_delay', ea.STATE_SET, ['24h', '48h', '72h', 'cancel']).withDescription('Weather delay'),
            e.text('normal_timer', ea.STATE_SET).withDescription('Normal timer'),
            e.numeric('quantitative_watering', ea.STATE_SET).withUnit('L').withValueMin(0).withValueMax(10000)
                .withDescription('Quantitative watering'), // Changed to liters
            e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withDescription('Child lock'),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverter.onOff],
                [3, 'status', tuya.valueConverterBasic.lookup({'Off': 0, 'one': 1, 'Child lock activated': 2, 'On': 3, 'On_btn': 4})],
                [5, 'countdown', tuya.valueConverter.raw],
                [6, 'countdown_left', tuya.valueConverter.raw],
                [9, 'water_current', {
                    from: (v) => {
                        convertedValue = (v * GALLON_TO_LITER / 1000).toFixed(1); // Converts the raw value from gal/min to L/min with one decimal place
                        console.log(`Raw value for water_current: ${v}, Converted value: ${convertedValue}`);
                        return convertedValue;
                    },
                    to: (v) => {
                        const rawValue = Math.round((parseFloat(v) / GALLON_TO_LITER) * 1000); // Converts L/min back to the raw value in gal/min
                        console.log(`Raw value for water_current: ${v}, Raw value: ${rawValue}`);
                        return rawValue;
                    },
                }],
                [11, 'battery', tuya.valueConverter.raw],
                [15, 'water_total', {
                    from: (v) => (v * GALLON_TO_LITER / 1000).toFixed(1), // Converts the raw value from gallons to liters with one decimal place
                    to: (v) => Math.round((parseFloat(v) / GALLON_TO_LITER) * 1000) // Converts liters back to the raw value in gallons
                }],
                [19, 'fault', tuya.valueConverter.raw],
                [37, 'weather_delay', tuya.valueConverter.raw],
                [38, 'normal_timer', tuya.valueConverter.raw],
                [42, 'switch_enabled', tuya.valueConverter.onOff],
                [47, 'smart_irrigation', tuya.valueConverter.raw],
                [101, 'total_flow_reset_switch', tuya.valueConverter.onOff],
                [102, 'quantitative_watering', {
                    from: (v) => (v * GALLON_TO_LITER / 1000).toFixed(1), // Converts the raw value from gallons to liters with one decimal place
                    to: (v) => Math.round((parseFloat(v) / GALLON_TO_LITER) * 1000) // Converts liters back to the raw value in gallons
                }],
                [103, 'flow_switch', tuya.valueConverter.onOff],
                [104, 'child_lock', tuya.valueConverter.onOff],
                [105, 'surplus_flow', tuya.valueConverter.raw],
                [106, 'single_watering_duration', tuya.valueConverter.raw],
                [108, 'single_watering_amount', {
                    from: (v) => {
                        const convertedValue = (v * GALLON_TO_LITER / 1000).toFixed(1);
                        console.log(`Raw value for single_watering_amount: ${v}, Converted value: ${convertedValue}`);
                        return convertedValue;
                    },
                    to: (v) => {
                        const rawValue = Math.round((parseFloat(v) / GALLON_TO_LITER) * 1000);
                        console.log(`Value to be sent for single_watering_amount: ${v}, Raw value: ${rawValue}`);
                        return rawValue;
                    }
                }],
            ],
        },
};

module.exports = definition;

```

closed 2025-01-01 by github-actions[bot]

papyrus78, 2025-01-12:
Hi,

Can someone from the team tell us what you need to finally add this new divice to the next uograde?

13630, 2025-03-08:
Bonjour, 

pensez vous que ce module fonctionnera un jour ?

Merci d'avance.

bicoun, 2025-05-09:
Same issue for me: 
Impossible to do something, even on/off on water!
[image]
[image]
Any information to integrate this valve?

bicoun, 2025-06-03:
Feedback : I have made a migration from native ZHA to zigbee2mqtt integration in ZHA, then I use file given by you @papyrus78 , an it is possible for me to drive on/ff the valve.
Battery  measument is 600 with new battery.
I did not test other features.

LeDevAlsacien, 2025-06-15:
> Hello [@LeDevAlsacien](https://github.com/LeDevAlsacien)
> 
> Can you please give us the config file you are using?
> 
> Thanks

hello i used the file you provided on Oct 25, 2024 here : https://github.com/Koenkk/zigbee2mqtt/issues/23398#issuecomment-2437251266

SuperCurro, 2025-06-28:
Hello, I got this device in Domadoo, reference SAS980SWT-7-ZTU product SASWELL - Zigbee smart solenoid valve and irrigation controller RTX ZVG1. I have the old one, no problem, everything works. This one has not support zigbee2mqtt.

When I import the config file I get 
Manufacturer [SASWELL](https://www.zigbee2mqtt.io/supported-devices/#v=SASWELL)
Model [SAS980SWT-7-ZTU](https://www.zigbee2mqtt.io/devices/SAS980SWT-7-ZTU.html#saswell-sas980swt-7-ztu)
Manufacturer
[SASWELL](https://www.zigbee2mqtt.io/supported-devices/#v=SASWELL)
Model
[SAS980SWT-7-ZTU](https://www.zigbee2mqtt.io/devices/SAS980SWT-7-ZTU.html#saswell-sas980swt-7-ztu)

[image]

to import the reference file I followed these instructions:

Create Your Converter File

Write your custom converter in JavaScript (_TZE200_81isopgh.js).
 `
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const GALLON_TO_LITER = 3.78541;

const definition = {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE204_qtnjuoae',]),
        zigbeeModel: ['SAS980SWT-7-ZTU'],
        model: 'SAS980SWT-7-ZTU',
        vendor: 'SASWELL',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        description: 'Zigbee smart water valve',
        onEvent: tuya.onEventSetTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn the valve On or Off'),
            e.enum('status', ea.STATE, ['0', '1', '2', '3']).withDescription('Status'),
            e.numeric('countdown', ea.STATE_SET).withUnit('min').withValueMin(1).withValueMax(240).withDescription('Countdown'),
            e.numeric('countdown_left', ea.STATE).withUnit('min').withValueMin(1).withValueMax(240).withDescription('Countdown left'),
            e.numeric('water_current', ea.STATE).withUnit('L/min').withValueMin(0).withValueMax(3785.41).withValueStep(0.1)
                .withDescription('Current water flow (L/min)'), // Changed to liters
            e.battery(),
            e.numeric('water_total', ea.STATE).withUnit('L').withValueMin(0).withValueMax(378541.0).withValueStep(0.001)
                .withDescription('Total water flow (L)'), // Changed to liters
            e.binary('fault', ea.STATE, 'DETECTED', 'NOT_DETECTED').withDescription('Fault status'),
            e.enum('weather_delay', ea.STATE_SET, ['24h', '48h', '72h', 'cancel']).withDescription('Weather delay'),
            e.text('normal_timer', ea.STATE_SET).withDescription('Normal timer'),
            e.numeric('quantitative_watering', ea.STATE_SET).withUnit('L').withValueMin(0).withValueMax(10000)
                .withDescription('Quantitative watering'), // Changed to liters
            e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withDescription('Child lock'),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverter.onOff],
                [3, 'status', tuya.valueConverterBasic.lookup({'Off': 0, 'one': 1, 'Child lock activated': 2, 'On': 3, 'On_btn': 4})],
                [5, 'countdown', tuya.valueConverter.raw],
                [6, 'countdown_left', tuya.valueConverter.raw],
                [9, 'water_current', {
                    from: (v) => {
                        convertedValue = (v * GALLON_TO_LITER / 1000).toFixed(1); // Converts the raw value from gal/min to L/min with one decimal place
                        console.log(`Raw value for water_current: ${v}, Converted value: ${convertedValue}`);
                        return convertedValue;
                    },
                    to: (v) => {
                        const rawValue = Math.round((parseFloat(v) / GALLON_TO_LITER) * 1000); // Converts L/min back to the raw value in gal/min
[... 44 lines cut ...]
Standard Installations:
Place your .js file in the Zigbee2MQTT configuration directory. This is typically the same directory as your configuration.yaml.
Example:

text
/config/zigbee2mqtt/my-device.js
or, for newer versions:

text
/config/zigbee2mqtt/external_converters/my-device.js
If the external_converters directory does not exist, you can create it and move your .js files there.

Reference the Converter in Your Configuration

Open your configuration.yaml for Zigbee2MQTT.

Add or update the external_converters section to include your file:

text
external_converters:
  - _TZE200_81isopgh.js

The final result is negative, I am only able to open and close, not able to program anything and time countdown or flow doesn't work. I guess the import file is not the right one. I t should be something like this and it is not

[image]