Koenkk/zigbee2mqtt #23399: [New device support]: Frient Smart Relay 2 SMRZB-342

TechsDomadoo, 2024-07-23:
### Link

https://www.domadoo.fr/en/actuators-dimmers/6519-16a-zigbee-onoff-smart-relay-din-rail-format-consumption-measurement-frient-5713594004495.html

### Database entry

Type d'appareil Router Modèle Zigbee SMRZB-342 Constructeur Zigbee frient A/S État de prise en charge Non pris en charge (how_to_add_support)  Adresse IEEE 0x0015bc0014030dee Adresse réseau 0x9904

### Comments

Previous version SMRZB-332 is supported : https://www.zigbee2mqtt.io/devices/SMRZB-332.html
But not this new version SMRZB-342

### External definition

```shell
const {deviceEndpoints, onOff, deviceTemperature, identify, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['SMRZB-342'],
    model: 'SMRZB-342',
    vendor: 'frient A/S',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), onOff({"powerOnBehavior":false,"endpointNames":["1","2"]}), deviceTemperature({"endpointNames":["2"]}), identify(), electricityMeter()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

Teeseeone, 2024-09-03:
Did you get this to work? I have an unsupported device.

[image]

Teeseeone, 2024-10-02:
Please add this device to Z2M

cpanoma, 2024-12-14:
Hi,

How many time for add a new device ?

Teeseeone, 2025-02-13:
Not sure, did you try zha? Im still not using mine since z2m is not supporting it yet.

cpanoma, 2025-02-16:
Hi,

No, I don't use zha.
My entire system uses z2m.
I'm only missing the power consumption feedback, the on/off action is functional.

closed 2025-04-26 by github-actions[bot]