Koenkk/zigbee2mqtt #23400: Z2M stopped working after changing the Sonoff USB 3.0 dongle to the SLZB-06P7

Volkimic, 2024-07-23:
### What happened?

I moved to SLZB-06P7 over ethernet and removed the zigbee stick on /dev/ttyUSB0 from the configuration. Z2M is installed over docker.
docker-compose.yaml:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:latest
    restart: unless-stopped
    network_mode: host
    privileged: true
    environment:
      - TZ=Europe/Madrid
    volumes:
      - /run/udev:/run/udev:ro
      - /docker/zigbee2mqtt:/app/data
 Z2M configuration.yaml:

homeassistant:
  legacy_entity_attributes: true
permit_join: true
mqtt:
  server: mqtt://192.168.137.112:1883
  keepalive: 60
  password: passwordmuycomplidcado
  reject_unauthorized: true
  user: mqtt
  version: 5
  include_device_information: true
serial:
# Location of SLZB-06P7
  port: tcp://192.168.137.142:6638
  baudrate: 115200
  adapter: zstack
#serial:
#  port: /dev/ttyUSB0
frontend: true
advanced:
  channel: 13
  transmit_power: 20
  network_key:
-numbers
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: info
device_options:
  legacy: false

Telnet test:
user@server:/docker/zigbee2mqtt$ telnet 192.168.137.142 6638
Trying 192.168.137.142...
Connected to 192.168.137.142.
Escape character is '^]'.
Connection closed by foreign host.

### What did you expect to happen?

I expected to happen that it starts with the new configuration:

serial:
  port: tcp://192.168.137.142:6638
  baudrate: 115200
  adapter: zstack

### How to reproduce it (minimal and precise)

unplug the device on /dev/ttyUSB0. After plug and configure SLZB-06P7:
- I updated the firmware
- I updated the docker-compose file
- I updated the Z2M configuration.yaml file
And redeploy the docker container. Don't work.

### Zigbee2MQTT version

1.39.0 commit: 0326926

### Adapter firmware version

20240716	v2.3.6

### Adapter

SLZB-06P7

### Setup

Docker Ubuntu 22.04.04

### Debug log

    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:101:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:129:29)
    at Zigbee.start (/app/lib/zigbee.ts:63:27)
    at Controller.start (/app/lib/controller.ts:139:27)
    at start (/app/index.js:154:5)
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
info: 	z2m: Logging to console, file (filename: log.log)
info: 	z2m: Starting Zigbee2MQTT version 1.39.0 (commit #0326926)
info: 	z2m: Starting zigbee-herdsman (0.50.1)
info: 	zh:zstack:znp: Opening TCP socket with 192.168.137.142:6638
info: 	zh:zstack:znp: Socket connected
info: 	zh:zstack:znp: Socket ready
info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
info: 	zh:zstack:znp: Port closed
info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
error: 	z2m: Error while starting zigbee-herdsman
error: 	z2m: Failed to start zigbee
error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error: 	z2m: Exiting...
error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)

commander55, 2024-07-23:
Same hardware, same error, spontaneously stopped working without any changes to the system:

[2024-07-23 20:05:14] info: 	z2m: Starting Zigbee2MQTT version 1.39.0 (commit #unknown)
[2024-07-23 20:05:14] info: 	z2m: Starting zigbee-herdsman (0.50.1)
[2024-07-23 20:05:16] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-07-23 20:05:16] info: 	zh:zstack:znp: Serialport opened
[2024-07-23 20:05:16] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-07-23 20:05:17] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-07-23 20:05:36] error: 	z2m: Error while starting zigbee-herdsman
[2024-07-23 20:05:36] error: 	z2m: Failed to start zigbee
[2024-07-23 20:05:36] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-07-23 20:05:36] error: 	z2m: Exiting...
[2024-07-23 20:05:36] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)

djcrawleravp, 2024-07-31:
Just want to report the same thing on a SLZB-06 on ethernet and rolled back to 20240710, seems like the 20240716 is having some issues

Volkimic, 2024-08-01:
Hello. Since I was tired of trying so many things, I decided to start my Zigbee network from scratch. 30 devices, it took me a while but now everything works.
All those YouTube videos that tell you to change the Sonoff Dongle for this one and that's it, everything works, should teach you how to do it.
It didn't work for me no matter how Texas the chips were.
Maybe I'm too much of a novice (irony).
Thanks for everything.
Best regards

maciekdnd, 2024-08-02:
Check, it may be related: #23355.
Try this (change path to yours): 
`rm opt/zigbee2mqtt/data/coordinator_backup.json
systemctl restart zigbee2mqtt.service`

closed 2024-12-28 by github-actions[bot]