Koenkk/zigbee2mqtt #23402: [TS011F_2_gang_power] unable to turn on or off sockets remotely

hanifsuandy, 2024-07-23:
### What happened?

https://vi.aliexpress.com/item/1005006518819987.html?spm=a2g0o.order_list.order_list_main.27.6b9c1802XQByqA&gatewayAdapt=glo2vnm

I bought a few 2 gang power sockets a while back and had them installed and paired. Initially, they were identified as single gang plugs and I could only remotely turn on and off 1 of the sockets. I found that there was an OTA update so I got them updated, however, the issue remained the same. 

Upon further research, it seemed that this device was not yet supported but was in the process of being added. Shortly after (lucky me), it was added so I updated Z2M and the device turned up properly. However, I am not able to turn on or off either sockets remotely and was met with `z2m: No converter available for 'state_l1' ("ON")` error.

Has it got to do with me performing OTA update while it was detected as a single gang plug? If so, is it possible to somehow reset the OTA update?

[image]

[image]

[image]

### What did you expect to happen?

Expected to be able to turn on and off both sockets.

### Zigbee2MQTT version

1.39.0 commit: 03269267

### Adapter firmware version

20230507

### Adapter

SLZB-06

### Setup

Proxmox VM

daffyonlsd, 2024-08-06:
Same problem,  it added correctly out of box. 
But power controls do not work.

hanifsuandy, 2024-08-06:
That would mean that it has nothing to do with my pairing it prior to it being supported and performing the OTA update.

daffyonlsd, 2024-08-08:
I presume there's a slight difference between model, our model is the universal plug type.

hanifsuandy, 2024-11-02:
Hoping someone has an idea about errors relating to ```
z2m: No converter available for 'state_l1' ("ON")
``` so that I can do my own troubleshooting. I have had no experience with external converters and I've tried meddling with it but have had no success so far. Thanks in advance!

beginner2047, 2024-11-10:
I have the same issue. It seems that it is due to the payload for device is changed. 

`{
    "current": 0,
    "energy": 6,
    "linkquality": 144,
    "power": 0,
    "state_left": "OFF",
    "state_right": "ON",
    "voltage": 229,
    "state_l1": null,
    "state_l2": null
}`

it use state_left and state_right. I am trying to study how to add the coverter "state_left" and "state_left" or remap the "state_l1" to "state_left" and "state_right".

beginner2047, 2024-11-10:
I have followed the issue https://github.com/Koenkk/zigbee2mqtt/issues/22981 to create the external converters

I have added the following code in configuration.yaml in zigbee2mqtt

```
external_converters: 
  - TS011F.js
```
Here is the file content of TS011F.js

```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const {deviceEndpoints, identify, onOff, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
  zigbeeModel: ['TS011F'],
  model: 'TS011F', // Update this with the real model of the device (written on the device itself or product page)
  vendor: '_TZ3000_bep7ccew', // Update this with the real vendor of the device (written on the device itself or product page)
  description: '2 Gang Socket with power monitoring and USB', // Description of the device, copy from vendor site.
  extend: [deviceEndpoints({"endpoints":{"left":1,"right":2}}), onOff({"powerOnBehavior":false,"endpointNames":["left","right"]}), identify(), electricityMeter()],
          meta: {multiEndpointSkip: ['current', 'voltage', 'power', 'energy']},
  };

module.exports = definition;

```
The external converter is loaded
[2024-11-10 11:44:37] info: 	z2m: Loaded external converter 'TS011F.js'

But I still can't control it. May I know if any incorrect in my external converter file?

beginner2047, 2024-11-10:
With reference to the file src/devices/tuya.ts (line 3169 - 3180)

```
    {
        fingerprint: tuya.fingerprint('TS011F', ['_TZ3000_bep7ccew']),
        model: 'TS011F_2_gang_power',
        vendor: 'Tuya',
        description: '2 gang socket with power monitoring and USB',
        extend: [
            deviceEndpoints({endpoints: {left: 1, right: 2}, multiEndpointSkip: ['current', 'voltage', 'power', 'energy']}),
            onOff({powerOnBehavior: false, endpointNames: **['l1', 'l2']}),**
            identify(),
            electricityMeter(),
        ],
    },
```

It seems that the endpointName defined l1 and l2. However, the device is only left and right for the endpointName. In issue#22981, the endpointName is also left and right. Can we change it to left and right?

beginner2047, 2024-11-10:
I have used mosquitto_pub command, it can change the power socket on/off

Client null received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/Socket', ... (148 bytes))
{"current":0,"energy":7,"linkquality":140,"power":0,"state":"ON","state_l1":null,"state_l2":null,"state_left":"ON","state_right":"ON","voltage":225}
Client null received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/Socket', ... (148 bytes))
{"current":0,"energy":7,"linkquality":140,"power":0,"state":"ON","state_l1":null,"state_l2":null,"state_left":"ON","state_right":"ON","voltage":232}
Client null received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/Socket/right/set', ... (3 bytes))
OFF
Client null received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/Socket', ... (149 bytes))
{"current":0,"energy":7,"linkquality":140,"power":0,"state":"ON","state_l1":null,"state_l2":null,"state_left":"ON","state_right":"OFF","voltage":232}
Client null received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/Socket', ... (149 bytes))
{"current":0,"energy":7,"linkquality":144,"power":0,"state":"ON","state_l1":null,"state_l2":null,"state_left":"ON","state_right":"OFF","voltage":232}
Client null received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/Socket', ... (150 bytes))
{"current":0,"energy":7,"linkquality":140,"power":0,"state":"ON","state_l1":null,"state_l2":null,"state_left":"OFF","state_right":"OFF","voltage":232}

but the problem is that both left and right will be changed too. At this moment, I created MQTT switch to remote control both sockets.

hanifsuandy, 2024-11-27:
Sorry, finally found time to do a bit of troubleshooting.

> const {} = require('zigbee-herdsman-converters/lib/modernExtend');
> // Add the lines below
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = [...]

I tried creating the same converter file as @beginner2047 did above with the exception of the name of the model and named it `TS011F_2_gang_power`. I restarted Z2M and re-paired the device but I couldn't do get it to work either. 

Then I saw something about `manuSpecificTuya` in the debug logs and I remember seeing it in the Z2M documentation (https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html) so I modified the code to this:
```js
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const {deviceEndpoints, identify, onOff, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
  // Since a lot of Tuya devices use the same modelID, but use different datapoints
  // it's necessary to provide a fingerprint instead of a zigbeeModel
  fingerprint: [
      {
          // The model ID from: Device with modelID 'TS0601' is not supported
          // You may need to add \u0000 at the end of the name in some cases
          modelID: 'TS011F_2_gang_power',
          // The manufacturer name from: Device with modelID 'TS0601' is not supported.
          manufacturerName: '_TZ3000_bep7ccew',
      },
  ],
  model: 'TS011F_2_gang_power', // Update this with the real model of the device (written on the device itself or product page)
  vendor: 'Tuya', // Update this with the real vendor of the device (written on the device itself or product page)
  description: '2 Gang Socket with power monitoring and USB', // Description of the device, copy from vendor site.
  fromZigbee: [tuya.fz.datapoints],
  toZigbee: [tuya.tz.datapoints],
  onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
  configure: tuya.configureMagicPacket,
  extend: [deviceEndpoints({"endpoints":{"left":1,"right":2}}), onOff({"powerOnBehavior":false,"endpointNames":["left","right"]}), identify(), electricityMeter()],
          meta: {multiEndpointSkip: ['current', 'voltage', 'power', 'energy']},
  };

module.exports = definition;
```

However, I still get very similar logs to before I modified the codes. I don't know where to go from here.
```bash
root@zigbee2mqtt:/opt/zigbee2mqtt/data# cat log/2024-11-27.12-59-25/log.log | grep -i converter | grep -i ts011f
[2024-11-27 12:59:25] info:     z2m: Loaded external converter 'TS011F_2_gang_power.js'
[2024-11-27 13:00:17] debug:    z2m: No converter available for 'TS011F_2_gang_power' with cluster '57344' and type 'raw' and data '{"data":[8,113,10,1,208,72,2,0,6],"type":"Buffer"}'
[2024-11-27 13:00:17] debug:    z2m: No converter available for 'TS011F_2_gang_power' with cluster '57344' and type 'raw' and data '{"data":[8,113,10,1,208,72,2,0,6],"type":"Buffer"}'
[2024-11-27 13:00:17] debug:    z2m: No converter available for 'TS011F_2_gang_power' with cluster '57344' and type 'raw' and data '{"data":[8,113,10,1,208,72,2,0,6],"type":"Buffer"}'
[2024-11-27 13:00:17] debug:    z2m: No converter available for 'TS011F_2_gang_power' with cluster '57344' and type 'raw' and data '{"data":[8,113,10,1,208,72,2,0,6],"type":"Buffer"}'
[2024-11-27 13:00:17] debug:    z2m: No converter available for 'TS011F_2_gang_power' with cluster '57344' and type 'raw' and data '{"data":[8,113,10,1,208,72,2,0,6],"type":"Buffer"}'
[2024-11-27 13:00:18] debug:    z2m: No converter available for 'TS011F_2_gang_power' with cluster '57344' and type 'raw' and data '{"data":[8,114,10,2,208,72,2,0,10],"type":"Buffer"}'
[2024-11-27 13:00:18] debug:    z2m: No converter available for 'TS011F_2_gang_power' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53248":0,"53249":0,"53250":2619665,"53251":123,"53252":11944,"53253":1374,"powerOnBehavior":2,"switchType":0}'
[2024-11-27 13:00:18] debug:    z2m: No converter available for 'TS011F_2_gang_power' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"powerOnBehavior":2}'
```

hanifsuandy, 2024-11-28:
https://github.com/Koenkk/zigbee2mqtt/issues/23402#issuecomment-2466849945

I can also confirm that I have the same issue as @beginner2047 with the mosquitto_pub command.

hanifsuandy, 2024-11-28:
> It seems that the endpointName defined l1 and l2. However, the device is only left and right for the endpointName. In issue#22981, the endpointName is also left and right. Can we change it to left and right?

@beginner2047 was right. I manually changed the endpointNames to "left" and "right" from "l1" and "l2" respectively in the /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/tuya.js file in my Z2M container.
```js
{
        fingerprint: tuya.fingerprint('TS011F', ['_TZ3000_bep7ccew']),
        model: 'TS011F_2_gang_power',
        vendor: 'Tuya',
        description: '2 gang socket with power monitoring and USB',
        extend: [ 
            (0, modernExtend_1.deviceEndpoints)({ endpoints: { left: 1, right: 2 }, multiEndpointSkip: ['current', 'voltage', 'power', 'energy'] }),
            (0, modernExtend_1.onOff)({ powerOnBehavior: false, endpointNames: ['left', 'right'] }),
            (0, modernExtend_1.identify)(),
            (0, modernExtend_1.electricityMeter)(),
        ],
    },
```

The states now correctly show OFF/ON toggles in the Z2M frontend and will trigger the sockets.
[image]

However, when I turn either one of them on/off, both will be triggered simultaneously.
[image]

sleepypig2003, 2024-12-21:
I had the same problem.  I fixed mine using the home assistant terminal this way:

## get into the container containing zigbee2mqtt
```
docker exec -it `docker ps | grep zigbee2mqtt | cut -f 1 -d " "` /bin/bash
```

## Edit the expanded javascript file
```
vi node_modules/zigbee-herdsman-converters/devices/tuya.js
```

## Under fingerprint: tuya.fingerprint('TS011F', ['_TZ3000_bep7ccew']),
## 
## I changed this line:

```
            (0, modernExtend_1.onOff)({ powerOnBehavior: false, endpointNames: ['l1', 'l2'] }),
```

## to:

```
            (0, modernExtend_1.onOff)({ powerOnBehavior: false, endpointNames: ['left', 'right'] }),
```
			
## exit the container
```
exit
```

## restart the docker container
```
docker container restart `docker ps | grep zigbee2mqtt | cut -f 1 -d " "`
```

The on off can now be controlled.

@Koenkk it seems like this commit is the cause of the problem:

https://github.com/Koenkk/zigbee-herdsman-converters/commit/13ff5c5bc1324a0150d837c768f8cf81f099ab3f#diff-0157bb612c63e92c50f93116e1f50bf8e263030ed26d86caa28479159c2a9f4aR2593

I don't know if there are other devices with the name 'l1' and 'l2' endpoints, but mine definitely used 'left' and 'right':

```
info 2024-12-22 06:37:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1384b927520e5', payload '{"current":0,"energy":0,"linkquality":255,"power":0,"state_l1":null,"state_l2":null,"state_left":"OFF","state_right":"OFF","voltage":248}'
info 2024-12-22 06:37:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1384b927520e5', payload '{"current":0,"energy":0,"linkquality":255,"power":0,"state_l1":null,"state_l2":null,"state_left":"OFF","state_right":"OFF","voltage":248}'
```

Koenkk, 2024-12-22:
Fixed the issue, thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. Note that the dev branch currently contains **breaking changes**, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

hanifsuandy, 2024-12-23:
@sleepypig2003 after the fix, are you able to control the switches individually? In my case, every time I turn one on or off, the other one will follow.

sleepypig2003, 2024-12-23:
Yes and no.  I was able to individually control left and right for the socket I am testing nearest my zigbee coordinator.  On the swtich where it works, turning the switch on/off would send only *one* log message in the z2m logs:

`info 2024-12-24 06:27:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1384b927aaaaa', payload '{"current":0,"energy":0,"linkquality":255,"power":0,"state_left":"OFF","state_right":"ON","voltage":249}'
`
But, I have several other sockets installed on the wall further from the coordinator, so I'm guessing these go through a router, and those seem to always turn on and off together like you said.  When I turn on one switch through the exposed z2m device menu, I get *three* log messages at once one after the other.  Here's an example when I pressed the left "on" in Z2M: the first few logs seem like the correct state, but the last one shows both turned on.

```
info 2024-12-24 06:31:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138f05c8340ed', payload '{"current":0,"energy":0,"linkquality":140,"power":0,"state_left":"ON","state_right":"OFF","voltage":243}'
info 2024-12-24 06:31:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138f05c8340ed', payload '{"current":0,"energy":0,"linkquality":144,"power":0,"state_left":"ON","state_right":"OFF","voltage":243}'
info 2024-12-24 06:31:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138f05c8340ed', payload '{"current":0,"energy":0,"linkquality":144,"power":0,"state_left":"**ON**","state_right":"**ON**","voltage":243}'
```

Koenkk, 2024-12-24:
@hanifsuandy pushed a fix to allow separate control, let me know if it works, after updating you need to re-configure the device via the frontend (yellow refresh button)

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. Note that the dev branch currently contains **breaking changes**, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

hanifsuandy, 2024-12-25:
Thanks @Koenkk! I can control the sockets separately now. 

However, I have another issue. I have 3 of these devices and only 1 of them has an Energy reading while the other 2 have `Null` Energy readings. Have yet to troubleshoot for these. Maybe I'll open a different issue if this persists in the near future. Right now I'm just happy I can control them properly. Besides, it takes me too long to do troubleshooting as I'm not very well-versed in this.

beginner2047, 2024-12-25:
Thank you for your help. May I know how to apply the new fix? 

I have clicked the yellow button (reconfigure) and click "OK" for the confirmation prompt (Are you sure?). But there is no change and only have the following error

`z2m: Failed to configure 'Socket', attempt 10 (Error: ZCL command 0xa4c1385beb6448a1/1 haElectricalMeasurement.read(["acPowerDivisor","acPowerMultiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:446:29) at result.configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:1956:37) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:206:21) at Configure.configure (/app/lib/extension/configure.ts:130:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:57:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))`

sleepypig2003, 2024-12-25:
To apply the changes temporarily I used the [same steps](https://github.com/Koenkk/zigbee2mqtt/issues/23402#issuecomment-2558267754) as I did above, but edit it so that these two lines are changed.

Change these lines:
```
        extend: [
            tuyaMagicPacket(),                          # <--- add this line
            deviceEndpoints({endpoints: {left: 1, right: 2}, multiEndpointSkip: ['current', 'voltage', 'power', 'energy']}),
            onOff({powerOnBehavior: false, endpointNames: ['left', 'right']}),     # <--- make sure this is left and right
            identify(),
            electricityMeter(),                            # <--- delete this line if you don't want the electricity meter errors
```

Restart the container as above, then click reconfigure on the device with the yellow button.

Note: editing this in the running container does not survive a reboot, you'd have to edit the docker image for that.  If the container is restarted with a different ID, then you have to re-edit the changes in the running container.

I tried this on mine and it works.
The duplicate messages still appears when I click on/off for far-away devices, but the on/off state is now always ends up correct.

It looks like mine also doesn't get the energy meter readings and give this error on trying to configure it:

```
error 2024-12-22 06:37:24z2m: Failed to configure '0xa4c1384b92755555', attempt 1 (Error: ZCL command 0xa4c1384b927520e5/1 haElectricalMeasurement.read(["acPowerDivisor","acPowerMultiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:446:29) at result.configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:1925:37) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:205:21) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
```

So I also removed the  electricityMeter(), line as above and the error message above disappears.  Unsure if I should also remove "multiEndpointSkip: ['current', 'voltage', 'power', 'energy']})" as well.
 
But overall, now my devices are usable.

Thanks Koenkk!

beginner2047, 2024-12-31:
Thanks @sleepypig2003 and @Koenkk 

I have updated the tuya.js on HAOS (i.e. /var/lib/docker) directly because I don't want update it again after reboot. 

I also keep electricityMeter because I would like to keep the 'current', 'voltage', 'power', 'energy'. If removed, it will not show those value although they are total value from left and right. 

I am not sure if the power monitor can be split into left and right too.

beginner2047, 2025-01-02:
i have another minor issue after updated. I found that the right will be turn on and left will be turn off after z2m restarted. I don't know how to keep the existing status after power outage?

dokicro, 2025-03-01:
@sleepypig2003 have you found an solution for "But, I have several other sockets installed on the wall further from the coordinator, so I'm guessing these go through a router, and those seem to always turn on and off together like you said."

@Koenkk this still seems to be an issue ...

dokicro, 2025-03-01:
I am not sure why and how but if I create a group and then control the switches from the group, then it works, can somebody please test this, in endpoint choose l2 or l1

[image]

closed 2025-05-09 by github-actions[bot]