Koenkk/zigbee2mqtt #23403: [TS0041] LoraTap SS6100ZB - Battery percentage

arths31, 2024-07-23:
### What happened?

Hello,
I'm trying to integrate the LoraTap SS6100ZB wireless button.
The device is detected as a Tuya TS0041.
All good except battery percentage expose is reporting 1%.

Here is the generated external definition,
```
const {deviceEndpoints, battery, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0041'],
    model: 'TS0041',
    vendor: '_TZ3000_vn88ezar',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3,"4":4}}), battery(), onOff({"powerOnBehavior":false,"endpointNames":["1","2","2","3","3","4","4"]})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

I've tried using an external converter to use battery voltage instead to determine percentage but to no avail.
Got some of definitions items from tuya.ts of zigbee-herdsman-converters.

```
const {} = require('zigbee-herdsman-converters/lib/modernextend');
const fz = require( 'zigbee-herdsman-converters/converters/fromZigbee');
const tz = require ('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require( 'zigbee-herdsman-converters/lib/exposes');
const reporting = require( 'zigbee-herdsman-converters/lib/reporting');
const ota = require( 'zigbee-herdsman-converters/lib/ota');
const utils = require( 'zigbee-herdsman-converters/lib/utils');
const globalStore = require( 'zigbee-herdsman-converters/lib/store');
const tuya = require ('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
fingerprint: [{modelID: 'TS0041', manufacturerName: '_TZ3000_vn88ezar' }],
model: 'TS0041', 
vendor: '_TZ3000_vn88ezar',
description: 'LORATAP SS6100ZB', 
meta: {battery: {voltageToPercentage: '3V_1500_2800'}},
exposes: [e.action(['single', 'double', 'hold']), e.battery(), e.battery_voltage()], 
fromZigbee: [tuya.fz.on_off_action, fz.battery],
toZigbee: [], 
configure: tuya.configureMagicPacket 
}

module exports = definition;
```

I see things changing if I change the definition in my converter but battery stays the same.
What am I doing wrong ?

Thanks 😃

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

### Setup

HA on Raspberry Pi

CodeForLove83, 2024-07-24:
I don't have this device, but I faced recently a similar issue with another device that I'm about to make a PR.

Would you be able to share the debug log?
I recommend first checking the log for entries related to `batteryVoltage` or `voltage`. I assume they will be there because you mentioned 1%, but it's worth confirming.

arths31, 2024-07-28:
Here is what i have
`z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/remote_alarme', payload '{"action":"double","battery":1,"last_seen":"2024-07-28T10:55:30+02:00","linkquality":232,"voltage":2500}'`

closed 2024-12-28 by github-actions[bot]