Koenkk/zigbee2mqtt #23406: Issue with IKEA E14 Bulbs LED2101G4 and LED1949C5

monotonus, 2024-07-24:
### What happened?

Home Assistant user. To ensure a light is really off, I usually switch a light off, wait some seconds and then switch the light off again without further checks if it's still on or not. Background: I had some bulbs in the past which where reported as off, but where still on.

With both bulbs I now observed, that if you switch them off two times, the light get's on again on a minimal brigthness level and it's still reported as off. In fact it's toggling the bulb every time I use the light.turn_off service. 
This is an issue where I use this as a sleeping light for my daughter. Until some time it should be dark.

---

I already found a workaround: If I put the buld into a z2m group and do the same as described above, the bulb stays off.

### What did you expect to happen?

Buld should remain off, like several others in my setup, like an E27 IKEA LED2003G10.

### How to reproduce it (minimal and precise)

Use light.turn_off two times after another on the same bulb.

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

20230507

### Adapter

zStack3x0 / SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

docker container

### Debug log

`Info 2024-07-24 08:42:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/nachttisch_kinderzimmer_group', payload '{"brightness":254,"color_mode":"color_temp","color_temp":370,"state":"ON"}'
Info 2024-07-24 08:42:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/nachttisch_kinderzimmer', payload '{"brightness":254,"color_mode":"color_temp","color_options":null,"color_temp":370,"level_config":{"on_level":"previous"},"state":"ON","update":{"installed_version":16842784,"latest_version":16842784,"state":"idle"},"update_available":false}'
Info 2024-07-24 08:42:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/nachttisch_kinderzimmer_group', payload '{"brightness":254,"color_mode":"color_temp","color_temp":370,"state":"OFF"}'
Info 2024-07-24 08:42:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/nachttisch_kinderzimmer', payload '{"brightness":254,"color_mode":"color_temp","color_options":null,"color_temp":370,"level_config":{"on_level":"previous"},"state":"OFF","update":{"installed_version":16842784,"latest_version":16842784,"state":"idle"},"update_available":false}'
Info 2024-07-24 08:42:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/nachttisch_kinderzimmer', payload '{"brightness":254,"color_mode":"color_temp","color_options":null,"color_temp":370,"level_config":{"on_level":"previous"},"state":"OFF","update":{"installed_version":16842784,"latest_version":16842784,"state":"idle"},"update_available":false}'`

smeen89, 2024-07-29:
I've got a similar issue. I'm also on Home Assistant. After firmware update of LED2101G4 to 1.1.20 it sometimes stays on at a very low brightness after a lights.turn_off comand has been sent from Home Assistant. The light reports as off in both HA and Z2MQTT.

Tried to remove the bulb and re-pair it but it did not make any difference.

Zigbee2MQTT version: 1.39.0-1
Adapter: zStack3x0 Sonoff Zigbee 3.0 USB Dongle Plus
Adapter firmware version: 20221226
Setup: VM

Appoxus, 2024-07-30:
Same issue with the Ikea jetstroem 30x30 panels. If they are and I issue an Alexa turn off all lights command they get turned on to minimal brightness. Home assistant and Z2M report them as off.

The Jetstroem 100x40 panels are behaving normally.

Koenkk, 2024-08-01:
This can be fixed by removing the `transition` for this device.

Denis19xx, 2024-08-01:
> This can be fixed by removing the `transition` for this device.

This solution doesn't work for me.
The problem remains the same as before.

[image]

Appoxus, 2024-08-01:
Same here, setting the transition to 0 doesn't help.

monotonus, 2024-08-02:
Transition was also 0 at this bulb, but had this behaviour. You can also use the OFF button within z2m to test.

I had a similar bulb without this behaviour and there the transition setting was empty. So I deleted also the "0" for this bulb and this works also.

So please test this also with an empty setting if this works for you.

Koenkk, 2024-08-03:
It should not be set to `0` but to nothing, stop z2m, remove it from your `configuration.yaml` and set it.

Appoxus, 2024-08-03:
That did the trick, thanks @Koenkk & @monotonus.
So there's no way to have a transition with those lights without causing this issue?
Because it's kinda like BAM when they're turning on.

Koenkk, 2024-08-03:
I think using it for turning ON is fine, turning OFF seems to be the problem.

Appoxus, 2024-08-03:
Yeah, I meant if you turn them on without transition it's kinda unpleasant because they turn on instantly.

closed 2024-12-28 by github-actions[bot]