Koenkk/zigbee2mqtt #23410: [New device support]: add variant to TS0601_smart_air_house_keeper

herrtschaefer, 2024-07-24:
### Link

https://www.zigbee2mqtt.io/devices/TS0601_smart_air_house_keeper.html

### Database entry

Don’t know how to find that.

### Comments

I tried to use the existing zigbee herdsman converter with the added fingerprint but no luck, but the. Again I have not done anything like that before. 

### External definition

```shell
The auto generated definition.

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_dwcarsat',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-07-25:
Does it work when stopping z2m, changing all occurrences of `_TZE204_dwcarsat` to `_TZE200_dwcarsat` in the `database.db` and start z2m?

herrtschaefer, 2024-07-26:
How can I view and edit the db?

Koenkk, 2024-07-27:
With any text editor

linuxianer99, 2024-08-04:
Hey! 
Had the same device with "manufName":"_TZE204_dwcarsat". Changing the manufName to _TZE200_dwcarsat in database.db works well!
So looks like it just a new fingerprint of the same device :-)

Koenkk, 2024-08-05:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-08-05 by Koenkk