Koenkk/zigbee2mqtt #23413: NULL attribute due to double Z2M published payloads

CodeForLove83, 2024-07-24:
### What happened?

Action attribute in Expose tab is an empty string.

[image]

It seems that because of the duplicate Z2M payloads, the last value is being pulled from the action attribute in the payload, which according to the log is an empty string. I cannot figure out how to fix this. 

A similar [issue](https://github.com/Koenkk/zigbee2mqtt/issues/17048) has been opened in the past but it is in stale state. 

### What did you expect to happen?

No duplicate payloads and action correctly set.

### How to reproduce it (minimal and precise)

Test with [067646](https://www.zigbee2mqtt.io/devices/067646.html#legrand-067646).

### Zigbee2MQTT version

1.39.0-1 

### Adapter firmware version

0x26530900

### Adapter

ConBee3

### Setup

Supervised container HA

### Debug log

```
[2024-07-24 21:14:24] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 101010
[2024-07-24 21:14:24] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2024-07-24 21:14:24] debug:    zh:deconz:driver: query aps data indication
[2024-07-24 21:14:25] debug:    zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 80
[2024-07-24 21:14:25] debug:    zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 80 srcAddr: 0xdd9 destAddr: 0x0 profile id: 0x104 cluster id: 0xf lqi: 199
[2024-07-24 21:14:25] debug:    zh:deconz:frameparser: response payload: 24,225,10,0,240,33,0,0
[2024-07-24 21:14:25] debug:    zh:controller: Received payload: clusterID=15, address=3545, groupID=null, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=199, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":225,"commandIdentifier":10},"payload":[{"attrId":61440,"dataType":33,"attrData":0}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-07-24 21:14:25] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-07-24 21:14:25] debug:    z2m: Received Zigbee message from 'Generale Frangisole Taverna', type 'attributeReport', cluster 'genBinaryInput', data '{"61440":0}' from endpoint 1 with groupID null
[2024-07-24 21:14:25] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Generale Frangisole Taverna', payload '{"action":"stopped","battery":65,"linkquality":199,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":71}'
[2024-07-24 21:14:25] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Generale Frangisole Taverna', payload '{"action":"","battery":65,"linkquality":199,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":71}'
[2024-07-24 21:14:25] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Generale Frangisole Taverna/action', payload 'stopped'
[2024-07-24 21:14:25] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 101010
[2024-07-24 21:14:25] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2024-07-24 21:14:25] debug:    zh:deconz:driver: query aps data indication
[2024-07-24 21:14:25] debug:    zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 81
[2024-07-24 21:14:25] debug:    zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 81 srcAddr: 0xdd9 destAddr: 0x0 profile id: 0x104 cluster id: 0x102 lqi: 200
[2024-07-24 21:14:25] debug:    zh:deconz:frameparser: response payload: 17,226,1
[2024-07-24 21:14:25] debug:    zh:controller: Received payload: clusterID=258, address=3545, groupID=null, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=200, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":226,"commandIdentifier":1},"payload":{},"command":{"ID":1,"parameters":[],"name":"downClose"}}
[2024-07-24 21:14:25] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-07-24 21:14:25] debug:    z2m: Received Zigbee message from 'Generale Frangisole Taverna', type 'commandDownClose', cluster 'closuresWindowCovering', data '{}' from endpoint 1 with groupID null
[2024-07-24 21:14:25] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Generale Frangisole Taverna', payload '{"action":"close_1","battery":65,"linkquality":200,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":71}'
[2024-07-24 21:14:25] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Generale Frangisole Taverna', payload '{"action":"","battery":65,"linkquality":200,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":71}'
[2024-07-24 21:14:25] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Generale Frangisole Taverna/action', payload 'close_1'
```

CodeForLove83, 2024-07-25:
Can you confirm that's the reason?

I guess that's the reason ([link](https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty)):
_Why is the action property always empty?
When the Home Assistant integration is enabled (homeassistant: true in your configuration.yaml) the action property of your e.g. buttons will almost always be empty. Whenever an action is published e.g. {"action": "single"} it will be immediately followed up by a {"action": ""}. This is to trigger a state change in the Home Assistant action sensor (so that it can be used in e.g. automations)._

Koenkk, 2024-07-27:
It's expected indeed when you have the home assistant integration enbaled.

closed 2024-07-27 by Koenkk