Koenkk/zigbee2mqtt #23414: Tuya TS110E_2gang_2way Error

papa-aus-rastatt, 2024-07-24:
### What happened?

Device  show Error with using.

z2m crashes after > 1 hour. 

### What did you expect to happen?

change Light

### How to reproduce it (minimal and precise)

wait. DO NOTHING. OR CHANGE LIGHT!

### Zigbee2MQTT version

1.39.0-1

### Adapter firmware version

14.1.0.2 (3537d18-zbbrdgpro)

### Adapter

https://www.zigbee2mqtt.io/devices/TS110E_2gang_2.html#tuya-ts110e_2gang_2

### Setup

add-on in Home Assistant

### Debug log

2024-07-24 22:15:32z2m: Publish 'set' 'min_brightness' to 'EG Küche Licht' failed: 'Error: 'min_brightness' is not a number, got string ()'
Error 2024-07-24 22:15:34z2m: Publish 'set' 'max_brightness' to 'EG Küche Licht' failed: 'Error: 'max_brightness' is not a number, got string ()'
Error 2024-07-24 22:15:40z2m: Publish 'set' 'min_brightness' to 'EG Küche Licht' failed: 'Error: 'min_brightness' is not a number, got string ()'
Error 2024-07-24 22:15:42z2m: Publish 'set' 'max_brightness' to 'EG Küche Licht' failed: 'Error: 'max_brightness' is not a number, got string ()'
Error 2024-07-24 22:15:45z2m: Publish 'set' 'min_brightness' to 'EG Küche Licht' failed: 'Error: ZCL command 0xa4c13807fbb4dfe0/1 genLevelCtrl.write({"64515":{"value":209,"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error 2024-07-24 22:15:58z2m: Publish 'set' 'max_brightness' to 'EG Küche Licht' failed: 'Error: ZCL command 0xa4c13807fbb4dfe0/1 genLevelCtrl.write({"64516":{"value":1000,"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error 2024-07-24 22:16:10z2m: Publish 'set' 'min_brightness' to 'EG Küche Licht' failed: 'Error: ZCL command 0xa4c13807fbb4dfe0/2 genLevelCtrl.write({"64515":{"value":205,"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error 2024-07-24 22:16:22z2m: Publish 'set' 'max_brightness' to 'EG Küche Licht' failed: 'Error: ZCL command 0xa4c13807fbb4dfe0/2 genLevelCtrl.write({"64516":{"value":992,"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error 2024-07-24 22:16:34z2m: Publish 'set' 'max_brightness' to 'EG Küche Licht' failed: 'Error: ZCL command 0xa4c13807fbb4dfe0/2 genLevelCtrl.write({"64516":{"value":1000,"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error 2024-07-24 22:16:46z2m: Publish 'set' 'power_on_behavior' to 'EG Küche Licht' failed: 'Error: ZCL command 0xa4c13807fbb4dfe0/1 genOnOff.write({"moesStartUpOnOff":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error 2024-07-24 22:16:56z2m: Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Bind 0xa4c13807fbb4dfe0/1 genOnOff from '0x00124b00257bb192/1' failed (AREQ - ZDO - bindRsp after 10000ms)'
Error 2024-07-24 22:19:15z2m: Publish 'set' 'power_on_behavior' to 'EG Küche Licht' failed: 'Error: ZCL command 0xa4c13807fbb4dfe0/1 genOnOff.write({"moesStartUpOnOff":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error 2024-07-24 22:19:26z2m: Publish 'set' 'max_brightness' to 'EG Küche Licht' failed: 'Error: ZCL command 0xa4c13807fbb4dfe0/1 genLevelCtrl.write({"64516":{"value":1000,"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error 2024-07-24 22:19:38z2m: Publish 'set' 'max_brightness' to 'EG Küche Licht' failed: 'Error: ZCL command 0xa4c13807fbb4dfe0/2 genLevelCtrl.write({"64516":{"value":1000,"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error 2024-07-24 22:19:51z2m: Publish 'set' 'min_brightness' to 'EG Küche Licht' failed: 'Error: ZCL command 0xa4c13807fbb4dfe0/2 genLevelCtrl.write({"64515":{"value":205,"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error 2024-07-24 22:20:03z2m: Publish 'set' 'min_brightness' to 'EG Küche Licht' failed: 'Error: ZCL command 0xa4c13807fbb4dfe0/1 genLevelCtrl.write({"64515":{"value":205,"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error 2024-07-24 22:20:15z2m: Publish 'set' 'power_on_behavior' to 'EG Küche Licht' failed: 'Error: ZCL command 0xa4c13807fbb4dfe0/1 genOnOff.write({"moesStartUpOnOff":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Error 2024-07-24 22:20:27z2m: Publish 'get' 'power_on_behavior' to 'EG Küche Licht' failed: 'Error: ZCL command 0xa4c13807fbb4dfe0/1 genOnOff.read(["moesStartUpOnOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'

[2024-07-24 22:19:51] error: 	z2m: Publish 'set' 'min_brightness' to 'EG Küche Licht' failed: 'Error: ZCL command 0xa4c13807fbb4dfe0/2 genLevelCtrl.write({"64515":{"value":205,"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
[2024-07-24 22:20:03] error: 	z2m: Publish 'set' 'min_brightness' to 'EG Küche Licht' failed: 'Error: ZCL command 0xa4c13807fbb4dfe0/1 genLevelCtrl.write({"64515":{"value":205,"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
[... cut ...]
[2024-07-24 22:20:27] error: 	z2m: Publish 'get' 'power_on_behavior' to 'EG Küche Licht' failed: 'Error: ZCL command 0xa4c13807fbb4dfe0/1 genOnOff.read(["moesStartUpOnOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
[2024-07-24 22:27:16] error: 	z2m: Adapter disconnected, stopping

CodeForLove83, 2024-07-24:
Could you check LQI? 
Can you try connecting it straight to the coordinator?

papa-aus-rastatt, 2024-07-24:
Linkquality
Signalstärke
69lqi

no, only wireless.

CodeForLove83, 2024-07-24:
The LQI appears to be low. Could you move closer to the coordinator and verify if it works?

papa-aus-rastatt, 2024-07-24:
that is built into the wall, and I have other devices with much worse signal display that have no problem. 

It looks more like an incompatibility to me, switching on and off works, adjusting brightness does. Because after a current path it can be controlled again.
I have just logged it off and on again. The minimum and maximum brightness settings are already giving errors.

Ev15, 2024-08-12:
Same here:

z2m: Publish 'set' 'state' to 'Eettafel' failed: 'Error: ZCL command 0xa4c138222ea1681f/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'

After re-pairing the problem returns i a matter of hours. Tried to pair it with a nearby router (hue bulb) but that did not solve it..

closed 2024-12-28 by github-actions[bot]