Koenkk/zigbee2mqtt #23415: Ikea E2001/E2002 Reporting Incorrect Commands for Button Press

mark1foley, 2024-07-24:
### What happened?

When doing a top or bottom button press the commands issued toggle between "brightness_move_up" and "brightness_stop" (for the top button) and "brightness_move_down" and "brightness_stop" (for the bottom button).   Note that I am talking about a button press here (i.e. press and release quickly).  The hold and release behaviour is issuing the "brightness_move_up"/"brightness_move_down" commands on hold and "brightness_stop" on release.

### What did you expect to happen?

When doing a press on the top button the "on" command should be issued and "off" for the bottom button (as per documentation https://community.home-assistant.io/t/zigbee2mqtt-ikea-remote-styrbar-type-e2001-e2002/451893).
This previously worked fine.  However, I have done both zigbee2mqtt upgrade and a firmware update to the device recently and one of these may have caused the issue

### How to reproduce it (minimal and precise)

Press the top or bottom buttons on an E2001

### Zigbee2MQTT version

 1.38.0

### Adapter firmware version

26780700

### Adapter

ConBee2

### Setup

Home Assistant 2024.7.1 on Home Assistant Blue with Mosquitto version 1.5.7 on separate raspberry pi

### Debug log

[Zigbee2mqtt Log.txt](https://github.com/user-attachments/files/16368094/Zigbee2mqtt.Log.txt)

Bkrickl, 2024-08-01:
Look at this issue. There may be a firmware update soon available (hopefully).

https://github.com/Koenkk/zigbee2mqtt/issues/23204

mark1foley, 2024-08-18:
Hi,

I looked at that issue before logging mine.  My reading of that issue is that the remotes are no longer communicating at all Z2MQTT, but maybe I misunderstood.  My issue is different in that the remote is still communicating but the button presses are no longer being interpreted correctly.  Specifically, when pressing the top and bottom buttons Z2MQTT is sending brightness up and down commands rather than the on and off commands that it did previously.  Also, the documentation (https://community.home-assistant.io/t/zigbee2mqtt-ikea-remote-styrbar-type-e2001-e2002/451893) states that pressing these buttons should result in the on and off commands being sent.

Mark

#23204<https://github.com/Koenkk/zigbee2mqtt/issues/23204>

Bkrickl, 2024-08-19:
I think your problem is mentioned here: https://github.com/Koenkk/zigbee2mqtt/issues/23204#issuecomment-2221321344

Update is now available for the switch. After the installation everything should work again.

closed 2024-12-28 by github-actions[bot]