Koenkk/zigbee2mqtt #23416: GZCGQ01LM battery not showing value

oneseventhree, 2024-07-25:
### What happened?

I have a GZCGQ01LM and it used to show all values on my Sonoff adapter but now I have a SLZB-06M (ember) on Firmware version 20240510.

I've replaced the battery to a brand new one and waited over 24 hours twice but still I get:
[image]

### What did you expect to happen?

 Battery is a number value

### Zigbee2MQTT version

1.39.0 commit: 03269267

### Adapter firmware version

20240510

### Adapter

SLZB-06M - Ember

### Setup

plain in proxmox

oneseventhree, 2024-08-03:
Screenshot of 1.39.1 commit: e132316a

[image]

oneseventhree, 2024-12-20:
This is still an issue

adynis, 2025-01-25:
I confirm, I have the same issue ,

[image]

oneseventhree, 2025-01-25:
I have been using “cheaper” batteries from
Amazon so I switched to a a Duracell battery and it still didn’t work. Out of frustration I deleted the device and re-pairing and it worked with both the cheaper battery and the newer battery!

I recommend deleting the device then re-pairing:
    
[image]

closed 2025-01-25 by oneseventhree

adynis, 2025-01-26:
cool :) it worked also for me 👍 Thanks