Koenkk/zigbee2mqtt #23439: TS0601_dimmer_5 switch type attribute not working

samvel-galstyan, 2024-07-27:
### What happened?

When I am changing the attribute "Switch Type" (Momentary, Toggle, State) nothing happens.

### What did you expect to happen?

I want to set switch mode to toggle.

### How to reproduce it (minimal and precise)

add Moes dimmer (zm-105-M) to Z2M and go to settings. Change the switch type to toggle, state or momentary. Then try to use dumb(rocker) switch. It will act as momentary (push) switch.

### Zigbee2MQTT version

1.39.0-1

### Adapter firmware version

1.42

### Adapter

ZBDongle-E with ezsp

### Setup

Add on on HAOS

samvel-galstyan, 2024-08-11:
@Koenkk can you please help me?

lichtzeichner, 2024-12-08:
I have two of these dimmers and I also stumbled upon this issue. I checked the manual, this module is **not** capable of changing the switch type. It expects to be connected to a "rocker switch" (I think in zigbee terms this is a "momentary switch").

See also the manual: https://drive.google.com/file/d/1QHva-X5YcHzhNy_2Jrg2I1MmQKdUKSBU/view

Can we turn this into a bugreport to update the device database to remove the setting? Or should I file a new issue?

closed 2025-02-15 by github-actions[bot]

Pierre-Lavoine, 2026-02-19:
I'm sorry @DJm00n  but this is not clear, since the wiring diagram clearly show a rocker switch and they also mention it, in option b for reseting the module. ¯\_(ツ)_/¯

[image]

[image]