Koenkk/zigbee2mqtt #23442: [New device support]: SMLIGHT SLZB-06 (without M)

Zuendapp, 2024-07-28:
### Link

https://smlight.tech/product/slzb-06/

### Database entry

{"id":37,"type":"Router","ieeeAddr":"0x00124b002f8f17e7","nwkAddr":15117,"manufId":0,"manufName":"SMLIGHT","powerSource":"Mains (single phase)","modelId":"SLZB-06","epList":[8,242],"endpoints":{"8":{"profId":260,"epId":8,"devId":255,"inClusterList":[0,3],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"modelId":"SLZB-06","manufacturerName":"SMLIGHT","powerSource":1,"zclVersion":3,"hwVersion":1,"swBuildId":"20240716"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"hwVersion":1,"swBuildId":"20240716","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1722151170334}

### Comments

In this case used as Router not as Conductor!

### External definition

```shell
const {identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['SLZB-06'],
    model: 'SLZB-06',
    vendor: 'SMLIGHT',
    description: 'Automatically generated definition',
    extend: [identify()],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-07-28:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `SLZB-06.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2024-07-28 by Koenkk

Zuendapp, 2024-07-29:
Hi @Koenkk 

thank you very much!
The fork is created and the picture uploaded.

BR

gam-st, 2024-08-13:
> Hi @Koenkk
> 
> thank you very much! The fork is created and the picture uploaded.
> 
> BR

Hello Zuendapp, please send the ticket to smlight ticketing system with the link to this Thread, smlight will send you a gift, our pleasure that you took care about this naming issue in z2m.