Koenkk/zigbee2mqtt #23446: HA Supervisor logs

Anto79-ops, 2024-07-29:
### What happened?

hello,

Noticed these logs in HA Supervisor (using the Z2M addon on HAOS, SMlite SLB-06):

```
2024-07-28 23:28:38.998 WARNING (MainThread) [supervisor.addons.options] Unknown option 'disable_led' for Zigbee2MQTT (45df7312_zigbee2mqtt)
2024-07-28 23:28:38.999 WARNING (MainThread) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
```

### Zigbee2MQTT version

1.39.0-1

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

Add-on HA, HAOS

Anto79-ops, 2024-07-29:
addon config:

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://192.168.1.234:1883
  user: [redacted]
  password:[redacted]
serial:
  port: tcp://slzb-06.local:6638
  baudrate: 115200
  adapter: zstack
  disable_led: false
advanced:
  transmit_power: 20
  channel: 20
```

Anto79-ops, 2024-07-29:
it seems if move these two options from the addon-config to the Z2M configuration yaml file, and the supervisor logs stop complaining.

```
  disable_led: false
advanced:
  transmit_power: 20
  channel: 20
  ```

Anto79-ops, 2024-07-30:
after 24 hrs, I'm not seeing the logs being reported in Supervisor anymore.

there may be some bug in how the addon handles what the user puts in the Z2M addon configuration vs what in the Z2M configuration.yaml file

serlinGi, 2024-11-02:
same here.

Log supervisor:
"2024-11-02 15:55:47.916 WARNING (MainThread) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)"

without option advanced in log, No Warning.

HA O.S.
Core 2024.10.4
Supervisor 2024.10.3
Operating System 13.2
Frontend 20241002.4
Zigbee2mqtt:  1.41.0-1
adapter: zstack   
Coordinator: SLZB-06P10 in LAN with PoE

My only option "advanced" in conf:
advanced:
  trasmit_power: 15

closed 2025-01-02 by Anto79-ops