Koenkk/zigbee2mqtt #23450: Awox model [33955] - problem with detection, firmware and behaviour

drupry, 2024-07-29:
### What happened?

I have 6 of Eglo Fueva-Z lights [900102](https://www.eglo.com/uk/recessed-light-fueva-z-900102.html) witch seems to be the same as already supported awox [33955](https://www.zigbee2mqtt.io/devices/33955.html) but 2 of those lights are detecting itself as Awox [33957](https://www.zigbee2mqtt.io/devices/33957.html) witch have slightly deferent white temperature range. Its happening since beginning. It's not working correctly with adaptive lightning integration.
Now with last fw update (through Awox homeControl app), all my 6 lights should have latest 2.7.5 fw from Awox app but z2m sees the FW version as 2.7.2_1348 for 33955 lights and 0122052017 for 33957 lights. I don't know if its changes anything but its strange, before update, all lights reporting it'self as 2.7.2 in Z2M even they had 2.7.3 fw.
But the biggest problem i have with them is that they are slow, showing very low LQI and often do not react at all because of timeout. I have quite solid zigbee network with 50+ devices but only those lights have this problem. Is there any solution other than "do not buy Awox/Eglo lights" ?  What is causing those timeout problems ??

### What did you expect to happen?

I want all of my lights reporting itself as Awox [33955](https://www.zigbee2mqtt.io/devices/33955.html) because that's the right color temperature range (2700-6500 K).
I want to know if its there a solution how to make them work reliable.

### How to reproduce it (minimal and precise)

I tried re-pair them several times different ways. it's always report itself wrong because it's showing different zb model. It's possible that 2 of 6 lights bought in the same store at once have different hw ?
<details>
 <summary> 33957 </summary>
 {"id":30,"type":"Router","ieeeAddr":"0xa4c138092fe2aa66","nwkAddr":65099,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":268,"inClusterList":[65360,65361,4],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":2,"hwVersion":60,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1722258075916}
</details>
and 

<details>
 <summary> 33955 </summary>
`{"id":8,"type":"Router","ieeeAddr":"0xa4c138ef573f0c45","nwkAddr":36933,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"EGLO_ZM_TW","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"EGLO_ZM_TW","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":1,"hwVersion":60,"swBuildId":"2.3.13_800"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":268,"inClusterList":[65360,65361,4],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":60,"swBuildId":"2.3.13_800","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1721399387497}
`
</details>

and this is log with my timeout problem. it have 3 problems - on/off, brightness lvl and temperature lvl. 
` z2m: Publish 'set' 'state' to 'koupelna stropní nad vanou' failed: 'Error: ZCL command 0xa4c138846224c888/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 52610 - 1 - 33 - 6 - 11 after 10000ms)'`

` z2m: Publish 'set' 'color_temp' to 'koupelna stropní nad umyvadlem' failed: 'Error: ZCL command 0xa4c138092fe2aa66/1 lightingColorCtrl.moveToColorTemp({"colortemp":161,"transtime":450}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 65099 - 1 - 30 - 768 - 11 after 10000ms)'`

`z2m: Publish 'set' 'brightness' to 'koupelna stropní nad vanou' failed: 'Error: ZCL command 0xa4c138846224c888/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 52610 - 1 - 15 - 8 - 11 after 10000ms)'`

### Zigbee2MQTT version

 1.39.0

### Adapter firmware version

20230507

### Adapter

zStack3x0

### Setup

Z2M on Intel NUC with ZB dongle-P stick.

drupry, 2024-09-05:
**For anyone dealing with this:**
After some more problems with my network, I switched to ember coordinator - Sonoff dongle-E with 7.4.3. firmware and all my Awox lights report itself right - as Awox 33955 at first try so I think this is some coordinator fw bug.  The problem with timeout error still occur, only with different description so I assume this is problem related with Awox hw ? Anyway the Eglo/Awox lights are nice but I can't recommend them to be used in Z2M.

drupry, 2024-09-05:
**For anyone dealing with this:**
After some more problems with my network, I switched to ember coordinator - Sonoff dongle-E with 7.4.3. firmware and all my Awox lights report itself right - as Awox 33955 at first try and even reports the right fw version (2.7.5) so I think this is some Zstack coordinator fw bug.  
The problem with timeout error still occur, only with different description so I assume this is problem related with Awox hw ? 
Anyway the Eglo/Awox lights are nice but I can't recommend them to be used in Z2M.

closed 2024-12-29 by github-actions[bot]

filipcoja, 2025-04-06:
Also have the same problem with those lights, sadly it looks like a HW issue since the LQI is really bad, cant even turn on my 2.4ghz radio on my AP because they will stall the lights